arendst/Tasmota #5672: Alexa Wemo Emu doesn´t work with 6.5.0.8 and core 2.5.0

nightfly0815, 2019-04-18:
Hi I have a problem withe the wemo emu and Amazon Alexa. After I restarted my Router I got new IP adresse for my Sonoff Basic. After that Alexa App couldn´t finde my device. So I flashed the newest dev build with core 2.5.0 and it doesn´t help. I saw that there are new sections in the alexa app to find and add a new device. I tried different types but Alexa couldn't find the device. When I set the sonoff device to hue bridge then Alexa find the device as a light. Is this a new problem that Amazon changed anything with the wemo emu? Because when I choose wemo device at the alexa app then the app told me to use the wemo skill and the wemo mobile app to configure the wemo device.

ascillato2, 2019-04-18:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-04-18:
Alexa does work with core 2.5.0. and latest Arduino feature/stage. I use it...
So without completly filled out template no help is possible since it is your personal setup that has issues

nightfly0815, 2019-04-18:
Ok that is funny, because I used my self compiled firmwares since months with 2.5.0 and the worked well. (And I only thing that I changed was the core 2.5.0 from 2.3.0, when I compile the firmware. Nothing else!)
What I mean is, that I connected the sonoff basic with 2.3.0 core a year ago to Alexa. And few months ago I changed to 2.5.0 and all worked well, with Alexa. Just the new teaching won´t work with Alexa since I got a new IP Adresse as I wrote bevor. 
But I will try it with the actual dev and compile it with 2.3.0 and look if it works with the wemo emu.

regards

toto79, 2019-04-18:
> ...(And I only thing that I changed was the core 2.5.0 from 2.3.0, when I compile the firmware. Nothing else!)

http://thehackbox.org/tasmota/020300/ and http://thehackbox.org/tasmota/020500/ ...here you'll find the nightly builds

nightfly0815, 2019-04-19:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
**Alexa Wemo Emu doesn´t work with 6.5.0.8/9 and core 2.3.0/2.5.0. It´s not possible to add a new device in the Alexa app.**

I tried booth nightly builds form the links [http://thehackbox.org/tasmota/020300/](url) and [ http://thehackbox.org/tasmota/020500/ ](url) with the same result. Today I tried my old Sonoff SV with version 6.4.0.1 and core 2.3.0. It worked without any problems at Alexa APP. I just choose add device and the last position "other". And Alexa found the wemo device. After that I updated my device to the actual nightly from the link with core 2.5.0 and it still works in Alexa with wemo emu. As I described at the first posting when I try to add a new device in the the alexa app with the actual firmware (core 2.3.0 and core 2.5.0) it´s not possible to find the device with the wemo emu. When I changed it to Hue bridge then alexa found the device. But alexa set the devices automatic as a light.
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : _____
- [x] Tasmota binary firmware version number used : _6.5.0.9-2_5_0/3.0.0-dev(c0f7b44)/ (pre-compiled and self-compiled )
- [x] Development IDE - Compiler / Upload tools used : VS Code
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

14:25:06 CMD: status 0
14:25:06 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Küche"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:25:06 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff-4m.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:03","StartupUTC":"2019-04-19T13:14:03","Sleep":50,"CfgHolder":4617,"BootCount":33,"SaveCount":108,"SaveAddress":"3F9000"}}
14:25:06 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0.9(sonoff)","BuildDateTime":"2019.04.19 14:52:07","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
14:25:06 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Openwrt_sonoff","Openwrt_sonoff"],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008001","280500000100000000000000000000000000","00000020"]}}
14:25:06 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":560,"Free":2508,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","0FDAE394","140783A0","23B617CE","01003BC0"]}}
14:25:06 RSL: STATUS5 = {"StatusNET":{"Hostname":"Küche","IPAddress":"192.168.0.159","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"195.34.133.21","Mac":"2C:3A:E8:4F:7A:12","Webserver":2,"WifiConfig":4}}
14:25:06 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Apr 19 13:25:06 2019","Local":"Fri Apr 19 14:25:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:51","Sunset":"19:47"}}
14:25:06 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-04-19T14:25:06"}}
14:25:06 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-04-19T14:25:06","Uptime":"0T00:11:03","Vcc":3.486,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Openwrt_sonoff","BSSId":"F8:1A:67:4D:76:11","Channel":13,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
14:26:18 CMD: weblog 4
14:26:18 RSL: RESULT = {"WebLog":4}
14:26:19 CFG: in Flash gespeichert am 3F8, zählen 109, Bytes 3584
14:26:27 WIF: Prüfe Verbindung...
14:26:27 WIF: verbunden
14:26:27 DNS: MDNS.update
14:26:33 UDP: Packet (199)
14:26:33 UDP: Packet (199)
14:26:33 UDP: Packet (199)
14:26:33 UDP: Packet (199)
14:26:33 UDP: Packet (199)
14:26:33 UDP: Packet (199)
14:26:33 UDP: Packet (199)
14:26:33 UDP: Packet (199)
14:26:42 UDP: Packet (172)
14:26:43 UDP: Packet (172)
14:26:45 UDP: Packet (172)

```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-30:
Now i have the same behaviour with Belkin/Wemo Emulation. It seems there is a change.
Please use Hue Emulation until it is fixed (if possible).

Jason2866, 2019-05-19:
Tried a old firmware (6.1.1b core 2.3.0) that i had by hand. With this version Alexa discovers the device when searched via miscellaneous (original in german Alexa "Sonstiges").
@arendst Something broke Belkin discovery since 6.4.x. What do you think about removing Belkin Emulation since it doesnt work anymore and it can completly replaced by HueEmulation? 
Saves code space and just one Emulation needs to be supported.

s-hadinger, 2019-05-19:
+1
This would free some space, whole emulation is currently ~18kb and Hue protocol supports much more features than Belkin's.

ascillato2, 2019-05-19:
Closing this issue as a poll has been opened for this.

Thanks everyone for sharing your ideas

closed 2019-05-19 by ascillato2