arendst/Tasmota #5673: Problem with Serial HW on WEMOS with PZEM-004T

michelebossa, 2019-04-19:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
I have a problem with Serial Hardware connection between WEMOS and PZEM-004T actually i'm using the Software Serial but somethings some reboot appening it is not a big problem but i have tried to use Serial Hardware connection to have more stability but if i switch there is no information from PZEM.

My worked configuration is:
[image]

[image]

if i try to switch TX and RX into configuration Module and the cable between PZEM and WEMOS into console i see there is Serial Hardware on but there isn't information from PZEM

### REQUESTED INFORMATION
There is some other step to do to use Serial HW connection?

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : WEMOS D1
- [X] Tasmota binary firmware version number used : 6.5.0-2_4_2/pre-compiled 
- [X] Development IDE - Compiler / Upload tools used : Platform-IO
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
09:22:35 MQT: sonoff_energia/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff_Energia"],"Topic":"sonoff_energia","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:22:35 MQT: sonoff_energia/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/020402/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T19:00:00","StartupUTC":"2019-04-17T13:22:35","Sleep":50,"CfgHolder":4617,"BootCount":160,"SaveCount":371,"SaveAddress":"F9000"}}
09:22:35 MQT: sonoff_energia/stat/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-03-19T12:47:29","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
09:22:35 MQT: sonoff_energia/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BOSSA_GALLO","iPhone"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A8009","280500000100000000000000000000000000","00000200"]}}
09:22:35 MQT: sonoff_energia/stat/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
09:22:35 MQT: sonoff_energia/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff_energia-5712","IPAddress":"192.168.1.114","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:24:B6:50","Webserver":2,"WifiConfig":4}}
09:22:35 MQT: sonoff_energia/stat/STATUS6 = {"StatusMQT":{"MqttHost":"XXXXXX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_24B650","MqttUser":"XXXXXX","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:22:35 MQT: sonoff_energia/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Apr 19 08:22:35 2019","Local":"Fri Apr 19 09:22:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:51","Sunset":"19:47"}}
09:22:35 MQT: sonoff_energia/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:22:35 MQT: sonoff_energia/stat/STATUS10 = {"StatusSNS":{"Time":"2019-04-19T09:22:35","ENERGY":{"TotalStartTime":"2019-03-25T19:25:56","Total":203.713,"Yesterday":8.378,"Today":1.891,"Power":297,"ApparentPower":366,"ReactivePower":214,"Factor":0.81,"Voltage":221,"Current":1.660}}}
09:22:35 MQT: sonoff_energia/stat/STATUS11 = {"StatusSTS":{"Time":"2019-04-19T09:22:35","Uptime":"1T19:00:00","Vcc":2.980,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"XXXXXXXXX","BSSId":"F0:B0:14:5A:1B:82","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [X] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
00:00:00 CFG: Loaded from flash at F6, Count 374
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff_Energia Version 6.5.0(sonoff)-2_4_2
00:00:00 SNS: Hardware Serial
```

### TO REPRODUCE

### EXPECTED BEHAVIOUR

### SCREENSHOTS
Correct configuration
[image]

[image]

If i try to use Serial HW there is no data.
[image]

[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2019-04-19:
I'm not sure I understand what is the problem exactly? Is your Wemos rebooting? If that is the case it is most likely a bad voltage regulator on the D1 mini. Connect PZEM to a separate power supply.

RX and TX have to be switched and your first example shows everything working. TX to TX is just sending to the D1's  serial out pin and data won't show since D1 isn't receiving any. Check our wiki, there are three examples of a PZEM004 setup with various devices

michelebossa, 2019-04-19:
The problem it is at moment i'm using the software serial but if i'm try to use the HW serial there is no data from PZEM (Voltage,Current.. display 0). How can i use the HW Serial? I have tried to swap RX/TX into configuration and the cable on PZEM in all possible combination without success. Thanks

Jason2866, 2019-04-19:
Try without level shifter as described here https://github.com/arendst/Sonoff-Tasmota/wiki/PZEM004T,-Wemos-D1-Mini-and-a-1602-I2C-display

wongnam, 2019-04-20:
@michelebossa No need level shifter by adding 1KΩ as below picture. My energy monitor with PZEM004T is running fine with this type of modification.

[image]

this is from wiki:

> Wiring Diagram:

> Check images below for more information about the 1kOhm resistor needed to shift the voltage to 5V from 3V3 for the PZEM-004T serial connection.
> ALTERNATIVE: No modification required on the PZEM-004T. Rather than using the 3.3V output from the Wemos D1 Mini to power the PZEM-004T serial interface, use the same 5V buck converter output (also used to supply power to the display and the Wemos D1 Mini) as VDD on the PZEM-004T serial interface. This eliminates the need for the resistor and the need to modify the PZEM-004T in any way at all.

[image]

ricargr, 2019-04-20:
I have the same problem with the level converted. @michelebossa, did you solve your restart problems? I think the alternative feeding PZEM-004t directly with 5V is not good (wemos d1 mini has no 5v tolerance)

Adding other question, did you notice than using level coverter you get very high reactive power values? I did some probes with a tri-phase pzem-004t proyect and I think I get not good values.

Thanks
[image]

Jason2866, 2019-04-20:
@ricargr  Level converting has no way input to measured values this is impossible because values are transfered digital encoded via serial...

ricargr, 2019-04-20:
Thanks @Jason2866
Could be the problem I use GPIO D5 and D0 instead TX and RX?

Jason2866, 2019-04-20:
Yes since hardware serial is only supported at Gpio 1 and 3

ascillato2, 2019-04-23:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-23 by ascillato2

martin072, 2020-07-20:
@michelebossa 

> @michelebossa You can use 3.3V directly for PZEM004T, No need level shifter by adding 1KΩ as below picture. My energy monitor with PZEM004T is running fine with this type of modification.
> 
> [image]
> 
> This is from wiki:
> 
> > Wiring Diagram:
> 
> > Check images below for more information about the 1kOhm resistor needed to shift the voltage to 5V from 3V3 for the PZEM-004T serial connection.
> > ALTERNATIVE: No modification required on the PZEM-004T. Rather than using the 3.3V output from the Wemos D1 Mini to power the PZEM-004T serial interface, use the same 5V buck converter output (also used to supply power to the display and the Wemos D1 Mini) as VDD on the PZEM-004T serial interface. This eliminates the need for the resistor and the need to modify the PZEM-004T in any way at all.
> 
> [image]

Hi @michelebossa,

I am a bit confused. I am making a similar project with a Wemos D1 Mini and Tasmota and the PZEM-004T, but not sure how to proceed with the serial connection. Some Internet sites say the ESP8266 is 5v tolerant on the I/O pins so they connect to 5v and tx/rx direct to the esp, others use level shifters. You mention a 1K resistor solution, assuming you provide the PZEM unit with 5v? Or als alternative use 3.3v directly to power the PZEM and no use of a resistor. Is this correct?  i.e. does the PZEM TTL part run on 3.3v?

Thanks

michelebossa, 2020-07-20:
Hi @martin072,
I have used the 1K resistor solution and power the PZEM with 5v.
Thanks

martin072, 2020-07-20:
Thanks,

Do you know if it also works with 3.3v? that would be a bit easier for me (as I have a custom pcb with already a 3.3v, GND and I/O header present).

michelebossa, 2020-07-20:
No :(

CarlosGS, 2020-08-08:
Works with 3.3V see https://github.com/arendst/Tasmota/issues/3626#issuecomment-670952578
The 1k resistor mod looks like this in new SMD power meters: https://github.com/arendst/Tasmota/issues/6050#issuecomment-510130142