arendst/Tasmota #5674: Extracting "Temperature" from feed for use in io.adafruit

Toepolev, 2019-04-19:
Hello
New here as member. I have been searching for a long time now how to solve this problem. 
Sonoff loaded with Tasmota as basic. Connected a DS18B20 and get a readout in the console. It sends a feed in this format: MQT: Toepolev/feeds/SENSOR = {"Time":"2019-04-19T13:08:32","DS18B20":{"Temperature":3.5},"TempUnit":"C"}.
So far so good. But I'm trying to use teh mqtt feed with io.adafruit. Here the feed should enter with only the value of the temperature, beeing 3.5 in this case. 
How do I change te feed from the above to a feed with only the value "temperature"?

Jason2866, 2019-04-19:
Sorry wrong place here to ask. This is Tasmota GitHub. You can post issues related to Tasmota.
See https://github.com/arendst/Sonoff-Tasmota/blob/master/CONTRIBUTING.md

closed 2019-04-19 by ascillato2

Toepolev, 2019-04-19:
Hello
I don't understand. I posted a question/issue I have concerning a Tasmota flashed Sonoff. Is there no place where People support each other in an open source setting?
Thx.

ascillato2, 2019-04-19:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

meingraham, 2019-04-19:
Perhaps a rule?

Rule1 ON DS18B20#Temperature DO Publish <yourTopic> %value% ENDON

Having said that, could the destination not parse the original JSON payload?