arendst/Tasmota #5677: #define D_START_UPGRADE is unnecessary

ngalfas, 2019-04-19:
This is more like memory improvement...

I really think that #define D_START_UPGRADE is unnecessary string.
I am currently reviewing the greek translation and i made a change on this define.

I noticed this..
[image]

I didn't go to check the code but i bet this is the strings used on this page.  I really do not see any reason to have different labels on these two buttons.

arendst, 2019-04-19:
Nice one.

If you searched the code you would find a reason for it. 

The string D_START_UPGRADE could be replaced by D_START + D_UPGRADE.

The code using D_START + D_UPGRADE is also being used by D_START + D_RESTORE.

So it might be an option to keep D_START_UPGRADE and add D_START_RESTORE. The code would be larger to handle this. Codespace was/is an issue.

For now we keep it this way.

ngalfas, 2019-04-19:
I see.  It's a code housekeeping issue then :)

closed 2019-04-19 by ascillato2