arendst/Tasmota #5678: PCF8594 Conection but no readings

SciTii, 2019-04-19:
Hi
I have board called ARD7063 with microcontroler PCF8594 that works on I2C. It is Board that allows you to have more analog pinouts same as ADS 1115. If I'm not wrong there is no official info about supporting This board by tasmota but it works on I2C so it should be connecting with Tasmota Right? What is more after sending i2cscan command it says " RSL: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x48"}" So there is a communications between Nodemcu and this board, but sadly there is no readings

If it is not a problem please add support for this.

Jason2866, 2019-04-19:
PCF8594 is not supported. It needs a driver. Since ADS 1115 is supported there will probably no driver written from the developer(s). Your are invited to write and make a PR.

ascillato2, 2019-04-19:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-19 by ascillato2

lalo-uy, 2019-04-24:
PCF8594 is not a microprocesor, is an eeprom chip.

Jason2866, 2019-04-24:
You are right but the author asked for support for PCF 8591. He typed wrong IC typ....
The text clearly describes a ADC chip. The given answer is correct.

SciTii, 2019-04-27:
Thanks for all Answers. Have a nice day!