arendst/Tasmota #5680: Randomly shutdown

Fabio7586, 2019-04-20:
### BUG DESCRIPTION
My sonoff t1 us 3 gang shutdown 2 of three lights randomly.
Initially i have thinked that it was ghost switching, so i have soldered in parallel a 12 nf capacitor and i have shutdown hassio with mqtt server. All works greatly until i reconnect home assistant. Lights 2 and three shutdown randomly. i cannot understand why.. please help me

this is a switch (they are all same): 

 - platform: mqtt
    name: "Luci Divano"
    state_topic: "stat/Sonoff_Luci_Soggiorno/POWER2"
    command_topic: "cmnd/Sonoff_Luci_Soggiorno/POWER2"
    availability_topic: "tele/Sonoff_Luci_Soggiorno/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"

### REQUESTED INFORMATION
Can someone help to understand what i wrong?

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Sonoff T1 3CH
- [X] Tasmota binary firmware version number used : Sonoff-Tasmota 6.4.1.19/ (uploaded with arduino ide)
- [X] Development IDE - Compiler / Upload tools used : ____ / uploaded with arduino ide
- [X] Provide the output of command ``status 0`` : provided at bottom

CONSOLE OUTPUT HERE:
21:59:05 MQT: cmnd/Sonoff_Luci_Soggiorno/POWER = 
21:59:05 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER1":"ON"}
21:59:05 MQT: stat/Sonoff_Luci_Soggiorno/POWER1 = ON
21:59:05 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER2":"OFF"}
21:59:05 MQT: stat/Sonoff_Luci_Soggiorno/POWER2 = OFF
21:59:05 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER3":"OFF"}
21:59:05 MQT: stat/Sonoff_Luci_Soggiorno/POWER3 = OFF
21:59:16 MQT: Attempting connection...
21:59:16 MQT: Connect failed to 192.168.2.11:1883, rc -2. Retry in 10 sec
21:59:30 MQT: Attempting connection...
21:59:31 MQT: Connected
21:59:31 MQT: tele/Sonoff_Luci_Soggiorno/LWT = Online (retained)
21:59:31 MQT: cmnd/Sonoff_Luci_Soggiorno/POWER = 
21:59:31 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER1":"ON"}
21:59:31 MQT: stat/Sonoff_Luci_Soggiorno/POWER1 = ON
21:59:31 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER2":"OFF"}
21:59:31 MQT: stat/Sonoff_Luci_Soggiorno/POWER2 = OFF
21:59:31 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER3":"OFF"}
21:59:31 MQT: stat/Sonoff_Luci_Soggiorno/POWER3 = OFF
21:59:45 MQT: Attempting connection...
21:59:46 MQT: Connected
21:59:46 MQT: tele/Sonoff_Luci_Soggiorno/LWT = Online (retained)
21:59:46 MQT: cmnd/Sonoff_Luci_Soggiorno/POWER = 
21:59:46 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER1":"ON"}
21:59:46 MQT: stat/Sonoff_Luci_Soggiorno/POWER1 = ON
21:59:46 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER2":"OFF"}
21:59:46 MQT: stat/Sonoff_Luci_Soggiorno/POWER2 = OFF
21:59:46 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER3":"OFF"}
21:59:46 MQT: stat/Sonoff_Luci_Soggiorno/POWER3 = OFF
22:00:19 MQT: Attempting connection...
22:00:20 MQT: Connected
22:00:20 MQT: tele/Sonoff_Luci_Soggiorno/LWT = Online (retained)
22:00:20 MQT: cmnd/Sonoff_Luci_Soggiorno/POWER = 
22:00:20 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER1":"ON"}
22:00:20 MQT: stat/Sonoff_Luci_Soggiorno/POWER1 = ON
22:00:20 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER2":"OFF"}
22:00:20 MQT: stat/Sonoff_Luci_Soggiorno/POWER2 = OFF
22:00:20 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER3":"OFF"}
22:00:20 MQT: stat/Sonoff_Luci_Soggiorno/POWER3 = OFF
22:00:52 MQT: Attempting connection...
22:00:52 MQT: Connect failed to 192.168.2.11:1883, rc -2. Retry in 10 sec
22:00:53 WIF: Connecting to AP1 The LAN Before Time in mode 11N as Sonoff_Luci_Soggiorno-2940...
22:00:56 WIF: Connected
22:01:07 MQT: Attempting connection...
22:01:08 MQT: Connected
22:01:08 MQT: tele/Sonoff_Luci_Soggiorno/LWT = Online (retained)
22:01:08 MQT: cmnd/Sonoff_Luci_Soggiorno/POWER = 
22:01:08 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER1":"ON"}
22:01:08 MQT: stat/Sonoff_Luci_Soggiorno/POWER1 = ON
22:01:08 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER2":"OFF"}
22:01:08 MQT: stat/Sonoff_Luci_Soggiorno/POWER2 = OFF
22:01:08 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER3":"OFF"}
22:01:08 MQT: stat/Sonoff_Luci_Soggiorno/POWER3 = OFF
22:02:00 MQT: tele/Sonoff_Luci_Soggiorno/UPTIME = {"Time":"2019-04-20T22:02:00","Uptime":"3T05:00:59"}
22:03:26 MQT: Attempting connection...
22:03:28 MQT: Connected
22:03:28 MQT: tele/Sonoff_Luci_Soggiorno/LWT = Online (retained)
22:03:28 MQT: cmnd/Sonoff_Luci_Soggiorno/POWER = 
22:03:28 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER1":"ON"}
22:03:28 MQT: stat/Sonoff_Luci_Soggiorno/POWER1 = ON
22:03:28 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER2":"OFF"}
22:03:28 MQT: stat/Sonoff_Luci_Soggiorno/POWER2 = OFF
22:03:29 MQT: stat/Sonoff_Luci_Soggiorno/RESULT = {"POWER3":"OFF"}
22:03:29 MQT: stat/Sonoff_Luci_Soggiorno/POWER3 = OFF
22:04:11 MQT: tele/Sonoff_Luci_Soggiorno/STATE = {"Time":"2019-04-20T22:04:11","Uptime":"3T05:03:10","Vcc":3.543,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"The LAN Before Time","BSSId":"C0:C1:C0:A0:0B:CB","Channel":6,"RSSI":60,"LinkCount":106,"Downtime":"0T00:06:18"}}
22:09:11 MQT: tele/Sonoff_Luci_Soggiorno/STATE = {"Time":"2019-04-20T22:09:11","Uptime":"3T05:08:10","Vcc":3.543,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"The LAN Before Time","BSSId":"C0:C1:C0:A0:0B:CB","Channel":6,"RSSI":66,"LinkCount":106,"Downtime":"0T00:06:18"}}

Status 0:

22:18:04 CMD: status 0
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS = {"Status":{"Module":30,"FriendlyName":["Luci_Soggiorno","Luci_Soggiorno_Divano","Luci_Soggiorno_TV"],"Topic":"Sonoff_Luci_Soggiorno","ButtonTopic":"0","Power":7,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"3T05:17:03","StartupUTC":"2019-04-17T16:01:01","Sleep":50,"CfgHolder":4617,"BootCount":41,"SaveCount":926,"SaveAddress":"F7000"}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS2 = {"StatusFWR":{"Version":"6.4.1.19(sonoff)","BuildDateTime":"2019-03-29T12:56:31","Boot":7,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["The LAN Before Time",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":288,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000783A0","23B617CE","00003BC0"]}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS5 = {"StatusNET":{"Hostname":"Sonoff_Luci_Soggiorno-2940","IPAddress":"192.168.2.101","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:97:2B:7C","Webserver":2,"WifiConfig":4}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.11","MqttPort":1883,"MqttClientMask":"Sonoff_Luci_Soggiorno","MqttClient":"Sonoff_Luci_Soggiorno","MqttUser":"root","MqttType":1,"MqttCount":28,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 20 21:18:04 2019","Local":"Sat Apr 20 22:18:04 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:49","Sunset":"19:48"}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS10 = {"StatusSNS":{"Time":"2019-04-20T22:18:04"}}
22:18:04 MQT: stat/Sonoff_Luci_Soggiorno/STATUS11 = {"StatusSTS":{"Time":"2019-04-20T22:18:04","Uptime":"3T05:17:03","Vcc":3.523,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"The LAN Before Time","BSSId":"C0:C1:C0:A0:0B:CB","Channel":6,"RSSI":76,"LinkCount":106,"Downtime":"0T00:06:18"}}}

ascillato, 2019-04-20:
Hi, you are using SDK 3.0.0 and an old version of Tasmota. That SDK is known to have bugs. Tasmota does not use that anymore.

Please, update to latest Tasmota and using the precompiled bins.

Fabio7586, 2019-04-21:
Where do you see the sdk version? then, what sdk? i have downloaded in first days of march the Tasmota 6.4.1.19 then i have taken arduino solution and i have uploaded it in sonoff. I have tried to update sonoff with ota, but after a while that i click on start upgrade, it shutdown two of three lights and then return to main menu without upgrading to 6.5.

ascillato, 2019-04-21:
In your status 0 it shows your SDK version.

Please, download the latest version from http://thehackbox.org/tasmota/sonoff.bin and upload it by the Tasmota web menu. If it say that there isn't enough space, please download the minimal version and then do the update to the full version

Fabio7586, 2019-04-21:
Upload Failed

Upload buffer miscompare

this is result message, any ideas?

Thnak you for your support

ascillato, 2019-04-21:
Let's talk in the Tasmota support chat https://discord.gg/Ks2Kzd4

ascillato, 2019-04-21:
I'm going to sleep now. Let's see that in the chat. If you don't want to use the chat, we can continue here.

Try to use the minimal firmware first and after that use the full version

Fabio7586, 2019-04-21:
Me too ;) it's 3 of AM. We can continue here,  because i cannot answer immediately every time as now ;)

Thank you for your support!!

blakadder, 2019-04-21:
Did you clear retained messages stored in mqtt broker persistence database. https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration#side-effects-with-using-mqtt-messages

Fabio7586, 2019-04-21:
> Did you clear retained messages stored in mqtt broker persistence database. https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration#side-effects-with-using-mqtt-messages

Ok i have done what you say, now i have to wait for eventually wrong behavior. I will inform you in a days if i solve.

> I'm going to sleep now. Let's see that in the chat tomorrow. If you don't want to use the chat, we can continue here.
> 
> Try to use the minimal firmware first and after that use the full version

I have tried to update with minimal firmware but it say:
`20:43:33 UPL: File sonoff-minimal.bin ...
20:43:44 MQT: Attempting connection...
20:43:45 MQT: Connected
20:43:45 MQT: tele/Sonoff_Luci_Soggiorno/LWT = Online (retained)
20:43:45 MQT: cmnd/Sonoff_Luci_Soggiorno/POWER = 
20:43:45 MQT: stat/Sonoff_Luci_Soggiorno/STATUS = {"Status":{"Module":30,"FriendlyName":["Luci_Soggiorno","Luci_Soggiorno_Divano","Luci_Soggiorno_TV"],"Topic":"Sonoff_Luci_Soggiorno","ButtonTopic":"0","Power":1,"PowerOnState":2,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:43:45 MQT: stat/Sonoff_Luci_Soggiorno/STATUS2 = {"StatusFWR":{"Version":"6.4.1.19(sonoff)","BuildDateTime":"2019-03-29T12:56:31","Boot":7,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
20:43:45 MQT: stat/Sonoff_Luci_Soggiorno/STATUS5 = {"StatusNET":{"Hostname":"Sonoff_Luci_Soggiorno-2940","IPAddress":"192.168.2.101","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:97:2B:7C","Webserver":2,"WifiConfig":4}}
20:43:45 MQT: stat/Sonoff_Luci_Soggiorno/STATUS11 = {"StatusSTS":{"Time":"2019-04-21T20:43:45","Uptime":"0T19:36:40","Vcc":3.523,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":739,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"The LAN Before Time","BSSId":"C0:C1:C0:A0:0B:CB","Channel":6,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:05"}}}
20:43:45 MQT: stat/Sonoff_Luci_Soggiorno/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T19:36:40","StartupUTC":"2019-04-21T00:07:05","Sleep":50,"CfgHolder":4617,"BootCount":42,"SaveCount":950,"SaveAddress":"FB000"}}
20:43:45 MQT: stat/Sonoff_Luci_Soggiorno/STATUS8 = {"StatusSNS":{"Time":"2019-04-21T20:43:45"}}
20:43:46 MQT: stat/Sonoff_Luci_Soggiorno/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["The LAN Before Time",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
20:43:48 MQT: stat/Sonoff_Luci_Soggiorno/STATUS8 = {"StatusSNS":{"Time":"2019-04-21T20:43:48"}}`

ascillato, 2019-04-21:
>I have tried to update with minimal firmware but it say
>Upload Failed
>
>Upload buffer miscompare

Ok. So seems that the memory map you use for compiling is not the recommended one used in Tasmota precompiled bins.

So, for updating you need to erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins with esptool.py also.

Jason2866, 2019-04-22:
Yes, different (not recommended) memory model used.
`ProgramSize":587,"Free":288,"Heap":20,"ProgramFlashSize":1024` SPIFFS used!
As @ascillato wrote only way is erasing flash and upload via esptool

Fabio7586, 2019-04-22:
And if i would compile with arduino ide and i'd upload result  .bin? So i can imagine that i can upload with same mamory mapping or not?

ascillato, 2019-04-22:
You can upload with the map you want but the settings that work with Tasmota are the recommended in the wiki. 

So, for solving your issue, you need to update, erasing all flash with esptool.py as explained in the wiki and flash again with the precompiled bins with esptool.py also.

Jason2866, 2019-04-22:
You cant upload via OTA because you have not enough free flash space.
So the only possibility is to flash via serial. Doing this you can use recommended settings.
This gives you the possibility in future to flash via OTA.

Fabio7586, 2019-04-22:
Yes, i understand that upload precompiled bin can solve issue, but Im trying to think a workaround(Im not at home ;) ) To compile with arduino ide, i have used settings described in the wiki because i dont have knowledge to make it myself ;)

Jason2866, 2019-04-22:
You cant solve it via OTA. No way!

ascillato2, 2019-04-23:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-23 by ascillato2

Fabio7586, 2019-04-25:
Sorry for delay, i solved with @blakadder solution. Thank you to all for support!!

HobbytronicsPK, 2019-05-13:
Have a look on to [Ghost / Random Switching on Sonoff Devices (MQTT Retained)](https://blog.hobbytronics.pk/sonoff-tasmota-tips-tricks/#Ghost-Random-Switching)