arendst/Tasmota #5681: ADS1115 not working with rules

doughecka, 2019-04-21:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
I have a single ADS1115 attached to Sonoff 4ch pro r2. I'm able to see the analog values perfectly fine in web/MQTT. However, if I try to use this sensor in a rule, it seems like it's not being detected. 
data:
05:47:55 MQT: PoolController/tele/SENSOR = {"Time":"2019-04-21T05:47:55","ADS1115":[{"A0":1959,"A1":-1,"A2":3165,"A3":3238}]}
rule:
rule1 on tele-ADS1115#A0 do publish PoolController/tele/SENSOR/pumppressure %value% endon

I verified the same rule works if I substitute a OneWire temp sensor. 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Sonoff 4ch Pro R2
- [X] Tasmota binary firmware version number used : 6.5.0(release-sensors) precompiled 
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [X] Provide the output of command ``status 0`` :
05:56:56 CMD: status 0
05:56:56 SRC: WebConsole from 192.168.124.10
05:56:56 RSL: Received Topic /status, Data Size 1, Data 0
05:56:56 RSL: Group 0, Index 1, Command STATUS, Data 0
05:56:56 MQT: PoolController/stat/STATUS = {"Status":{"Module":23,"FriendlyName":["Pool pump Sonoff","PoolSonoff2","PoolSonoff3","PoolSonoff4"],"Topic":"PoolController","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:56:56 MQT: PoolController/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T00:35:47","StartupUTC":"2019-04-21T04:21:09","Sleep":50,"CfgHolder":4617,"BootCount":25,"SaveCount":165,"SaveAddress":"F7000"}}
05:56:56 MQT: PoolController/stat/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sensors)","BuildDateTime":"2019-03-19T12:26:22","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
05:56:56 MQT: PoolController/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["heckaiot",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A8009","280500000100000000000000000000000000","000002C0"]}}
05:56:56 MQT: PoolController/stat/STATUS4 = {"StatusMEM":{"ProgramSize":544,"Free":456,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDEE394","0007A3A4","B7FFBFCC","005ABBC0"]}}
05:56:56 MQT: PoolController/stat/STATUS5 = {"StatusNET":{"Hostname":"PoolController-2268","IPAddress":"192.168.124.157","Gateway":"192.168.124.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.124.1","Mac":"84:0D:8E:48:E8:DC","Webserver":2,"WifiConfig":4}}
05:56:56 MQT: PoolController/stat/STATUS6 = {"StatusMQT":{"MqttHost":"hassio.tadhome.heckanet.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_48E8DC","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
05:56:56 MQT: PoolController/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 21 04:56:56 2019","Local":"Sun Apr 21 05:56:56 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:48","Sunset":"19:50"}}
05:56:56 MQT: PoolController/stat/STATUS10 = {"StatusSNS":{"Time":"2019-04-21T05:56:56","ADS1115":[{"A0":2050,"A1":-1,"A2":3293,"A3":3230}]}}
05:56:56 MQT: PoolController/stat/STATUS11 = {"StatusSTS":{"Time":"2019-04-21T05:56:56","Uptime":"0T00:35:47","SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"heckaiot","BSSId":"FE:EC:DA:A2:2F:79","Channel":6,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}}
05:56:56 WIF: Checking connection...
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
Attach ADS1115
rule1 on tele-ADS1115#A0 do publish PoolController/tele/SENSOR/pumppressure %value% endon

### EXPECTED BEHAVIOUR
rule should be triggered when sensor is polled

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2019-04-21:
Enable rule1 using `Rule1 1`
Disable one-shot using `Rule1 4`

Provide output of `Rule1` to see what's really setup.
Provide output of `Weblog 4` to see what's happen when teleperiod is over.

doughecka, 2019-04-21:
16:52:10 CMD: Rule1 1
16:52:10 SRC: WebConsole from 192.168.120.1
16:52:10 RSL: Received Topic /Rule1, Data Size 1, Data 1
16:52:10 RSL: Group 0, Index 1, Command RULE, Data 1
16:52:10 MQT: PoolController/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":428,"Rules":"on tele-ADS1115#A0 do publish PoolController/tele/SENSOR/pumppressure %value% endon"}
16:52:13 MQT: PoolController/tele/HASS_STATE = {"Version":"6.5.0(release-sensors)","BuildDateTime":"2019-03-19T12:26:22","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff 4CH Pro","RestartReason":"Software/System restart","Uptime":"0T10:43:31","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":26,"SaveCount":172,"IPAddress":"192.168.124.157","RSSI":"80","LoadAvg":37}
16:52:27 MQT: PoolController/tele/STATE = {"Time":"2019-04-21T16:52:27","Uptime":"0T10:43:45","SleepMode":"Dynamic","Sleep":50,"LoadAvg":37,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"heckaiot","BSSId":"92:2A:A8:57:F5:36","Channel":1,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}
16:52:27 MQT: PoolController/tele/SENSOR = {"Time":"2019-04-21T16:52:27","DS18B20-1":{"Id":"01143D2B9AAA","Temperature":20.4},"DS18B20-2":{"Id":"01143D67B2AA","Temperature":21.2},"ADS1115":[{"A0":1906,"A1":-1,"A2":0,"A3":-1}],"TempUnit":"C"}
16:52:28 WIF: Checking connection...
16:52:28 WIF: Connected
16:52:34 CMD: Rule1 4
16:52:34 SRC: WebConsole from 192.168.120.1
16:52:34 RSL: Received Topic /Rule1, Data Size 1, Data 4
16:52:34 RSL: Group 0, Index 1, Command RULE, Data 4
16:52:34 MQT: PoolController/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":428,"Rules":"on tele-ADS1115#A0 do publish PoolController/tele/SENSOR/pumppressure %value% endon"}
16:52:43 MQT: PoolController/tele/HASS_STATE = {"Version":"6.5.0(release-sensors)","BuildDateTime":"2019-03-19T12:26:22","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff 4CH Pro","RestartReason":"Software/System restart","Uptime":"0T10:44:01","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":26,"SaveCount":172,"IPAddress":"192.168.124.157","RSSI":"80","LoadAvg":35}
16:52:48 WIF: Checking connection...
16:52:48 WIF: Connected
16:52:55 CMD: Rule1
16:52:55 SRC: WebConsole from 192.168.120.1
16:52:55 RSL: Received Topic /Rule1, Data Size 0, Data 
16:52:55 RSL: Group 0, Index 1, Command RULE, Data 
16:52:55 MQT: PoolController/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":428,"Rules":"on tele-ADS1115#A0 do publish PoolController/tele/SENSOR/pumppressure %value% endon"}
16:52:57 MQT: PoolController/tele/STATE = {"Time":"2019-04-21T16:52:57","Uptime":"0T10:44:15","SleepMode":"Dynamic","Sleep":50,"LoadAvg":37,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"heckaiot","BSSId":"92:2A:A8:57:F5:36","Channel":1,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}
16:52:57 MQT: PoolController/tele/SENSOR = {"Time":"2019-04-21T16:52:57","DS18B20-1":{"Id":"01143D2B9AAA","Temperature":20.4},"DS18B20-2":{"Id":"01143D67B2AA","Temperature":21.2},"ADS1115":[{"A0":1828,"A1":-1,"A2":0,"A3":1}],"TempUnit":"C"}
16:53:08 WIF: Checking connection...
16:53:08 WIF: Connected
16:53:14 MQT: PoolController/tele/HASS_STATE = {"Version":"6.5.0(release-sensors)","BuildDateTime":"2019-03-19T12:26:22","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff 4CH Pro","RestartReason":"Software/System restart","Uptime":"0T10:44:32","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":26,"SaveCount":172,"IPAddress":"192.168.124.157","RSSI":"82","LoadAvg":35}
16:53:28 MQT: PoolController/tele/STATE = {"Time":"2019-04-21T16:53:28","Uptime":"0T10:44:46","SleepMode":"Dynamic","Sleep":50,"LoadAvg":37,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"heckaiot","BSSId":"92:2A:A8:57:F5:36","Channel":1,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}
16:53:28 MQT: PoolController/tele/SENSOR = {"Time":"2019-04-21T16:53:28","DS18B20-1":{"Id":"01143D2B9AAA","Temperature":20.4},"DS18B20-2":{"Id":"01143D67B2AA","Temperature":21.2},"ADS1115":[{"A0":1864,"A1":0,"A2":0,"A3":0}],"TempUnit":"C"}
16:53:58 MQT: PoolController/tele/SENSOR = {"Time":"2019-04-21T16:53:58","DS18B20-1":{"Id":"01143D2B9AAA","Temperature":20.4},"DS18B20-2":{"Id":"01143D67B2AA","Temperature":21.2},"ADS1115":[{"A0":1875,"A1":-1,"A2":0,"A3":-1}],"TempUnit":"C"}
16:54:09 WIF: Checking connection...
16:54:09 WIF: Connected
16:54:14 MQT: PoolController/tele/HASS_STATE = {"Version":"6.5.0(release-sensors)","BuildDateTime":"2019-03-19T12:26:22","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff 4CH Pro","RestartReason":"Software/System restart","Uptime":"0T10:45:32","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":26,"SaveCount":172,"IPAddress":"192.168.124.157","RSSI":"82","LoadAvg":19}
16:54:28 MQT: PoolController/tele/STATE = {"Time":"2019-04-21T16:54:28","Uptime":"0T10:45:46","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"heckaiot","BSSId":"92:2A:A8:57:F5:36","Channel":1,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}
16:54:28 MQT: PoolController/tele/SENSOR = {"Time":"2019-04-21T16:54:28","DS18B20-1":{"Id":"01143D2B9AAA","Temperature":20.4},"DS18B20-2":{"Id":"01143D67B2AA","Temperature":21.2},"ADS1115":[{"A0":1954,"A1":0,"A2":0,"A3":0}],"TempUnit":"C"}
16:54:29 WIF: Checking connection...
[... cut ...]
16:55:29 WIF: Checking connection...
16:55:29 WIF: Connected
16:55:45 MQT: PoolController/tele/HASS_STATE = {"Version":"6.5.0(release-sensors)","BuildDateTime":"2019-03-19T12:26:22","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff 4CH Pro","RestartReason":"Software/System restart","Uptime":"0T10:47:03","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":26,"SaveCount":172,"IPAddress":"192.168.124.157","RSSI":"80","LoadAvg":20}
16:55:49 WIF: Checking connection...
16:55:49 WIF: Connected

MadDoct, 2019-04-21:
Probably due to the fact that ads1115 uses "[" in its json report...

doughecka, 2019-04-21:
Hmm, is this something I can fix via a rule or is this a code level fix?

MadDoct, 2019-04-21:
It's a code level fix, I think... I submitted a pull request, but can't test it as I don't have an ads1115... Let the gurus see if it is Ok... ;)

doughecka, 2019-04-22:
Awesome... I can give it a test if I can obtain some firmware, or if I can
figure out how to compile the pull request myself. :)

curzon01, 2019-04-22:
@MadDoct - thanks, the needless json array [] might be the issue.

@doughecka 
Either you wait if the pull request will be merged by the project owner and you can than download it from as  [developer binary](http://thehackbox.org/tasmota/)

or you make your changes by yourself (see MadDoct [pull request change details](https://github.com/arendst/Sonoff-Tasmota/pull/5684/commits/393953d89456943ae3b10cea7cb62be30bf67f1d) about the changed two lines) and compile your own (see [Flashing Tool Guides](https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisites#flashing-tool) _If you want to modify the code or default settings you can use_)

ascillato2, 2019-04-23:
Closing as the fix is already in the PR #5684 

Thanks for reporting and fixing it :+1:

closed 2019-04-23 by ascillato2

qingz2004, 2019-09-10:
Sorry for commenting on the old closed issue.
I'm experiencing new issues on ADS1115 with rules with today's development version 6.6.0.11 core 2.3.0.
I found the trigger ADS1115#A0 does not work, while ADS1115#A0<10 works.
Is it a bug or expected behavior?
Thanks!

qingz2004, 2019-09-10:
@meingraham You are right. Not just ADS1115, the kind of trigger for all the sensors stop work with this version.

PR #5684 (2019-04-25): Change tele message of ADS1115 to allow it to work correctly with rules
Change tele message of ADS1115 to allow it to work correctly with rules, by removing "[", as reported by issue #5681

## Description:

**Related issue (if applicable):** fixes #5681

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched (Also remember to update _changelog.ino_ file)
  - [x] Only one feature/fix was added per PR.
  - [ ] The code change is tested and works.
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass* [... cut]