arendst/Tasmota #5683: Add a touch icon

duenni, 2019-04-21:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
On an Apple device you open the Tasmota webui in Safari, touch the "share" icon and select "Add to homescreen". As a result a link is added to your homescreen which looks like any other app. But there is no "app" logo/icon, so the icon is blank.

**Describe the solution you'd like**
Would it be possible to add a touch icon `<link rel="apple-touch-icon" href="some_icon.png"/>` to the web ui, so when you add the webaddress to your homescreen on an Apple device you would see this icon instead of a white/blank app icon.

**Describe alternatives you've considered**
Live with a blank icon. 

**Additional context**
[image]

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2019-04-21:
independently that's an enhancement for specific brand the icon (here 'some_icon.png') must be stored somewhere as source even it will be copied when the link was generated on your homescreen. Storing it in ESP will cost space of the rare memory, using a link somewhere in the internet will be invalid in some future.

duenni, 2019-04-21:
Well, not that I expect anything fancy. Maybe a black background with white letters "TSM" or whatever. Quite sure something like this can be packed in <1kB?

Jason2866, 2019-04-21:
You know 1kb is a lot (for this device)? Will not happen....

ascillato2, 2019-04-23:
Thanks for sharing your ideas. Sorry, that is not going to be added soon due the remaining RAM and flash space are reserved for sensors and features support. 

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-23 by ascillato2