arendst/Tasmota #5685: Sonoff RF Bridge occasionally unresponsive

stepanov1975, 2019-04-22:
Once per a couple of days web interface stops responding and received RF signals not sent over MQTT. In the same time device sometimes totally unresponsive, but sometimes MQTT telemetry still going on and device answer pings. This is especially annoying because in this case, I don't know how to detect this automatically to cycle power. Maybe it is a hardware problem but is it any way to investigate what happens?

ascillato2, 2019-04-22:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-04-22:
Please provide to Status 0 output, log files with loglevel 4

stepanov1975, 2019-04-22:
```
- [ ] Device used (i.e. Sonoff Basic) : Sonoff RF bridge
- [ ] Tasmota binary firmware version number used : 6.5.0 pre-compiled
- [ ] Development IDE - Compiler / Upload tools used : esptool
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
12:41:45 CMD: status 0
12:41:45 SRC: WebConsole from 192.168.0.66
12:41:45 RSL: Received Topic /status, Data Size 1, Data 0
12:41:45 RSL: Group 0, Index 1, Command STATUS, Data 0
12:41:45 MQT: stat/sonoffrfbridge/STATUS = {"Status":{"Module":25,"FriendlyName":["SonoffBridge"],"Topic":"sonoffrfbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T02:24:47","StartupUTC":"2019-04-22T09:16:58","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":51,"SaveAddress":"F8000"}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":4,"LogHost":"192.168.0.7","LogPort":514,"SSId":["alex_wireless",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","280500000100000000000000000000000000","00000000"]}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS5 = {"StatusNET":{"Hostname":"sonoffrfbridge-7270","IPAddress":"192.168.0.10","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.3","Mac":"DC:4F:22:F8:1C:66","Webserver":2,"WifiConfig":4}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.7","MqttPort":1883,"MqttClientMask":"sonoffrfbridge","MqttClient":"sonoffrfbridge","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 22 11:41:45 2019","Local":"Mon Apr 22 12:41:45 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:46","Sunset":"19:51"}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS10 = {"StatusSNS":{"Time":"2019-04-22T12:41:45"}}
12:41:45 MQT: stat/sonoffrfbridge/STATUS11 = {"StatusSTS":{"Time":"2019-04-22T12:41:45","Uptime":"0T02:24:47","Vcc":3.212,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"alex_wireless","BSSId":"CC:2D:21:02:BE:A1","Channel":3,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:06"}}}
12:41:50 WIF: Checking connection...

Jason2866, 2019-04-22:
Please post log (loglevel 4) where rf bridge reacts unresponsive.
Status 0 looks okay. RSSI could be better but it should worky with 58...

stepanov1975, 2019-04-22:
I can trigger Status 0 via MQTT, right? I will try to catch this.

talkleasy, 2019-04-23:
My Sonoff Bridge becomes unresponsive every few minutes using 6.5 but  is fine using 6.3.

ascillato2, 2019-04-27:
@stepanov1975 

Hi, any news on this?

stepanov1975, 2019-04-27:
> @stepanov1975
> 
> Hi, any news on this?

It takes 2-5 days to happen and in past days I added a couple of new light switches so shut down the electricity. Still works fine for now (3d)

ascillato, 2019-04-27:
Ok, great to know that it is working for you now. Please, close this issue and reopen it if necessary. Thanka

closed 2019-04-27 by ascillato2

stepanov1975, 2019-05-12:
ok it happened
web interface unresponsive

May 12 06:59:29 ha hass[1104]: 2019-05-12 06:59:29 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on cmnd/sonoffrfbridge/status: 0
May 12 06:59:29 sonoffrfbridge-7270 ESP-SRC: MQTT
May 12 06:59:29 sonoffrfbridge-7270 ESP-RSL: Received Topic cmnd/sonoffrfbridge/status, Data Size 1, Data 0
May 12 06:59:29 sonoffrfbridge-7270 ESP-RSL: Group 0, Index 1, Command STATUS, Data 0
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS = {"Status":{"Module":25,"FriendlyName":["SonoffBridge"],"Topic":"sonoffrfbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"5T19:36:49","StartupUTC":"2019-05-06T08:22:37","Sleep":50,"CfgHolder":4617,"BootCount":24,"SaveCount":68,"SaveAddress":"F7000"}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":4,"LogHost":"192.168.0.7","LogPort":514,"SSId":["alex_wireless",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","280500000100000000000000000000000000","00000000"]}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS5 = {"StatusNET":{"Hostname":"sonoffrfbridge-7270","IPAddress":"192.168.0.10","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.3","Mac":"DC:4F:22:F8:1C:66","Webserver":2,"WifiConfig":4}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.7","MqttPort":1883,"MqttClientMask":"sonoffrfbridge","MqttClient":"sonoffrfbridge","MqttUser":"mqttuser","MqttCount":17,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS7 = {"StatusTIM":{"UTC":"Sun May 12 03:59:26 2019","Local":"Sun May 12 06:59:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+03:00","Sunrise":"07:12","Sunset":"22:20"}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS10 = {"StatusSNS":{"Time":"2019-05-12T06:59:26"}}
May 12 06:59:29 sonoffrfbridge-7270 ESP-MQT: stat/sonoffrfbridge/STATUS11 = {"StatusSTS":{"Time":"2019-05-12T06:59:26","Uptime":"5T19:36:49","Vcc":3.212,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"alex_wireless","BSSId":"CC:2D:21:02:BE:A1","Channel":2,"RSSI":62,"LinkCount":10,"Downtime":"0T00:02:03"}}}

meingraham, 2019-05-12:
Please try updating your firmware to http://thehackbox.org/tasmota/020300/sonoff.bin so that the latest fixes are deployed on your device.

stepanov1975, 2019-05-12:
> Please try updating your firmware to http://thehackbox.org/tasmota/020300/sonoff.bin so that the latest fixes are deployed on your device.

But I have 6.5.0 firmware version. Isn't it is the latest version?

meingraham, 2019-05-12:
No

stepanov1975, 2019-05-12:
> No

Ok, I will update. But how can I know which version is the latest in the future?

meingraham, 2019-05-12:
The latest is always available at the link I sent. The latest codebase is compiled on an hourly basis.

This is the latest version codebase: https://github.com/arendst/Sonoff-Tasmota/tree/development

https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/_changelog.ino
https://github.com/arendst/Sonoff-Tasmota/commits/development

stepanov1975, 2019-05-12:
I see. I meant latest stable version ;)

meingraham, 2019-05-12:
Those are stable. THere is a separate branch for experimental.

stepanov1975, 2019-05-12:
Ok. This didn't go well. After a couple of attempts to update to this version, the bridge is dead now.
[sonoffrfbridge.zip](https://github.com/arendst/Sonoff-Tasmota/files/3170349/sonoffrfbridge.zip)
I will try to update via serial later

meingraham, 2019-05-12:
But, will it go unresponsive again?

We haven't seen users report issues similar to yours. The fact that both versions are exhibiting problems (yes, one more drastic than the other), could be highlighting a problem with the device. It could just be that the latest version is just exacerbated by the device's problem.

lobocobra, 2019-10-26:
Same issue here with version 6.6.0
Once a week sonoff RF Bridge does not recognize any RF message. I can still access the web interface and after a reboot it works again. 

Portisch RF Firmware is loaded and after a reboot everything works perfectly. After some days, the problem restarts. 

Status 0
`00:27:05 MQT: stat/y_SonoffRFbridge/STATUS = {"Status":{"Module":25,"FriendlyName":["y_SonoffRFbridge"],"Topic":"y_SonoffRFbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":4,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:59:42","StartupUTC":"2019-10-26T21:27:23","Sleep":50,"CfgHolder":4617,"BootCount":41,"SaveCount":61,"SaveAddress":"F8000"}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domaincats",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","280500000100060000005A64000000000000","00000000"]}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS5 = {"StatusNET":{"Hostname":"y_SonoffRFbridge-7339","IPAddress":"192.168.20.62","Gateway":"192.168.20.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.2","Mac":"DC:4F:22:F8:1C:AB","Webserver":2,"WifiConfig":5}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.20.90","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F81CAB","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 26 22:27:05 2019","Local":"Sun Oct 27 00:27:05 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:01","Sunset":"18:18"}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS10 = {"StatusSNS":{"Time":"2019-10-27T00:27:05"}}
00:27:05 MQT: stat/y_SonoffRFbridge/STATUS11 = {"StatusSTS":{"Time":"2019-10-27T00:27:05","Uptime":"0T00:59:42","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"domaincats","BSSId":"AC:22:05:85:AD:D3","Channel":6,"RSSI":52,"LinkCount":1,"Downtime":"0T00:00:07"}}}`

I will now install the latest code and then report if it re-occurs again. My suspicion is that it has to do with the SleepMode.

ascillato, 2019-10-26:
I have 2 sonoff bridge working since 2 years and I never had this issue. The mcu for rf is very reliable. Please, try with a better power supply. I have them with a standard 5v 1A power source. The sonoff bridge is pretty reliable.
I have them with standard settings (sleep and others as default) with mqtt and some rules. 
The sleep parameter only affects the esp8266. The rf mcu is not affected with that command. 

Please, try updating Tasmota, updating portish firmware and using a better power source.

atv2016, 2021-12-11:
@lobocobra did this fix your problem?