arendst/Tasmota #5689: script language as an alternative to rules

gemu2015, 2019-04-23:
i always disliked the syntax of rules. therefore i developed a script language.
see initial doku below. it is now default on my fork. it is work in progress and not yet  ready for pr but works quite well already. you may give it a try =>https://github.com/gemu2015/Sonoff-Tasmota

Script Editor for Tasmota

As an alternative to rules. (uses about the same amount of flash and ram)
In submenu of setup => edit script

Up to 50 variables (40 numeric and 10 strings)
Freely definable variable names
Nested if,the,else up to a level of 8

Math operators  +,-,*,/,%
Left right evaluation with optional brackets
all numbers are float
e.g. temp=hum*(100/37.5)+temp-(timer*hum%10)
Comparison operators ==,!=,>,>=,<,>=

Comments start with ;

Sections defined:

>DEF
define and init variables here, no other code allowed
>BOOT
executed on BOOT time
>TELE
executed on teleperiod time, get tele vars only in this section
>TIME
executed every second
>EVENT
executed e.g. on power change

special variables:

upsecs = seconds since start
uptime = minutes since start
time = minutes since midnight
sunrise = sunrise minutes since midnight
sunset = sunset minutes since midnight

tstamp = timestamp (local date and time)

powerx = power state

gtmp = global temperature
ghum = global humidity
(gprs = global pressure) not yet

boot = true on BOOT
tinit = true on time init
tset = true on time set
mqttc = true on mqtt connect
mqttd = true on mqtt disconnect
wific = true on wifi connect
wifid = true on wifi disconnect

stack = stack size
heap = heap size

tasmota cmds start with => 

special cmds:
=>print prints to web console
=>dprecx sets decimal precision to x (0-9)

example script

>DEV
p:mintmp=10  (p:means permanent) 
p:maxtmp=30
hello="hello world"
string="xxx"
hum=0
temp=0
timer=0
dimmer=0
sw=0
oldsw=0

>BOOT
=>print BOOT executed
=>print %hello%
=>mp3track 1

>TELE
hum=BME280#Humidity
temp=BME280#Temperature
sw=Switch1
if sw!=oldsw
then oldsw=sw
=>power1 %sw%
endif

if temp>30
and hum>70
then =>print damn hot!
endif

>TIME
; every second but not completely reliable time here
; use upsecs and uptime for reliable timers
timer+=1
if timer>=5
then =>print timer=%timer%
timer=0
endif

dimmer+=1
if dimmer>100
then dimmer=0
endif
=>dimmer %dimmer%

=>dprec0
=>displaytext [c1l1f1s2p20] dimmer=%dimmer%

if (upsecs%5)==0
then
=>print %upsecs%  (every 5 seconds)
endif

=>print %upsecs% %uptime% %time% %sunrise% %sunset% %tstamp%

if time>sunset
and time<sunrise
then
 ; night time
 if power1==0
 then
  =>power1 1
 endif
else
 ; day time
 if power1>0
 then
  =>power1 0
 endif
endif

if mqqtc>0
then =>print mqqt connected
endif

if boot>0
then =>displaytext [z]
endif

>EVENT
=>print event executed!

joba-1, 2019-04-23:
looks great! Much more intuitive (to me)

blakadder, 2019-04-23:
Nicely thought out. Looks very good!

Jason2866, 2019-04-23:
Cool, will try!

curzon01, 2019-04-24:
Looks nice and very usable.

Proposal: Sometimes I'm still fighing with each single byte when using rules. 
Words like `then` and `endif` cost valuable script space. Would it be possible using `{ }` for blocks in general independently if the leading keyword is `if` or `else`?

gemu2015, 2019-04-25:
@curzon01 
i don't see what you mean. you can put any code lines between then and endif. its like on endon but much more flexible.

ok, implementation is now nearly complete. (string catenation and string compare are now also supported)
needs a lot of testing now before pr

meingraham, 2019-04-25:
@gemu2015 ,

I believe this is what Norbert was trying to say.

```
if (condition) then
  action(s)
else
  alternate action(s)
endif
```
vs.

```
if (condition) {
  action(s)
}
else {
  alternate action(s)
}
```

eliminates "then" and "endif" saving 7 bytes of code space for each use of an if clause.

gemu2015, 2019-04-25:
ok, is implemented use either if, then,else,endif or

if (condition) {
  action(s)
}
else {
  alternate action(s)
}

gemu2015, 2019-04-27:
updated to current tasmota 
here is a wemos binary for testing: https://www.dropbox.com/s/kq1ckldzzft5nrl/wemos-d1-4m.bin?dl=0

+ updated doku now supporting subroutines

[scripter.md.zip](https://github.com/arendst/Sonoff-Tasmota/files/3123599/scripter.md.zip)

arendst, 2019-05-15:
@gemu It looks promising to me too!

In your fork I noticed it replaces rules but does it still use the same Settings area memory allocations? If so it would be great if you provided a PR for implementation in Tasmota. Pls keep it as short as possible so without SML and other changes related to your fork as I like it to be as modular as possible.

gemu2015, 2019-05-17:
@arendst 
yes it replaces rules and uses the same memory in settings.
i am constantly expanding the capabilities and optimizing the code but will provide a pr in the next week or so 
i would have preferred having a few beta testers to cope with more possible use cases then i have.

remark:
one prerequisite is that sensor names may not contain any math operators. i had to replace the '-' from sht3 drivers with '_'. (only digits, letters and underscore should be allowed in sensor names) 
don't know if any other sensors are affected too.

arendst, 2019-05-17:
Thx.

As there were/are some issues with the dash (-) I will investigate the impact of a change to underscore (_) solving these issues too.

arendst, 2019-05-17:
Just released 6.5.0.11 with a user option to change the minus to underscore. 

Now enforcing underscore.

Jason2866, 2019-05-17:
Mhh, will generate many issues questions (breaks backwards compatibility) for nearly all Home Automation sytems. What do you think only to swap (via ifdef...) when script ruels are activated via #define?

arendst, 2019-05-17:
Agree partly as it will just delay the issues coming up.

As the define is user configurable they can always change it back to "-".

Jason2866, 2019-05-17:
Yes, no problem for users who compile there own version. Our experience in Discord chat is that most users do use the provided prebuild ones. If the install the new version on a fresh device no standard setup described for the mainstream home automation solutions does work.
On the other side i do agree sometimes cuts have to been made to get rid of old stuff.
Your project, your decision!  We can handle the upcoming questions in Discord 😀

arendst, 2019-05-17:
I appreciate your work.

Let's decide that if there too many related issues we'll change the default back (reluctantly).

Jason2866, 2019-05-17:
Yes, let see what happens.

gemu2015, 2019-05-17:
or have a settable user option to change this behavior? setoption64 ?

effelle, 2019-05-17:
As discussed with @Jason2866 I'll do a series of tests to see the impact on memory compared with rules.
But yes, I like this new syntax. Thanks @gemu2015!

arendst, 2019-05-17:
SetOption64 it will be.

Default 0 keeps "-", 1 will be "_". No other choices allowed.

Why didn't I come up with this ...

jziolkowski, 2019-05-17:
So by using setoption for keeping the old behavior we change one type of issues ("hey, sensors no longer work in my home automation") to another ("hey, I can't use new rule scripts with my sensor")

arendst, 2019-05-17:
At least now they can change it on the fly ...

closed 2019-05-18 by gemu2015

davidgurr, 2019-07-11:
@gemu2015 Can Scripts be used in conjunction with the IrReceived functionality that's implemented via Rules for LED strips? See https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller for details.

I'm looking at using Scripts for a different reason, but need to replicate the Rules functionality for IrReceived. The Console shows the following on receiving the IR commands when using Rules:

```
17:23:39 MQT: tasmota/kitchen/lights/sonoff-ledutil/tele/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB04F"}}
17:23:39 RUL: IRRECEIVED#DATA=0X00FFB04F performs "WAKEUP"
17:23:39 MQT: tasmota/kitchen/lights/sonoff-ledutil/stat/RESULT = {"POWER":"ON"}
17:23:39 MQT: tasmota/kitchen/lights/sonoff-ledutil/stat/POWER = ON
17:23:39 MQT: tasmota/kitchen/lights/sonoff-ledutil/stat/RESULT = {"Wakeup":"Started"}
17:23:44 MQT: tasmota/kitchen/lights/sonoff-ledutil/tele/WAKEUP = {"Wakeup":"Done"}

```

gemu2015, 2019-07-11:
this should be no problem.

in D section define a string e.g.
irstr=""

in E section
irstr=IrReceived#Data
if irstr=="0x00FFB04F"
then
=>print received %irstr% 
endif

gemu2015, 2019-07-12:
since i own a magic home with ir receiver i made a script to show how it works
i have 2 magic home devices that should be synchronized so i send the commands also to a seconds magic home via websend

; define (expand string section to 25 chars)
/>D 25
istr=""
ws="websend [192.168.178.84]"

; event section
/>E
; get ir data
istr=IrReceived#Data

; on
if istr=="0x00F7C03F"
then
=>wakeup
=>%ws% wakeup
endif

; off
if istr=="0x00F740BF"
then
=>power1 0
=>%ws% power1 0
endif

;white
if istr=="0x00F7E01F"
then
=>color 000000ff
=>%ws% color 000000ff
endif

;red
if istr=="0x00F720DF"
then
=>color ff000000
=>%ws% color ff000000
endif

;green
if istr=="0x00F7A05F"
then
=>color 00ff0000
=>%ws% color 00ff0000
endif

;blue
if istr=="0x00F7609F"
then
=>color 0000ff00
=>%ws% color 0000ff00
endif

istr=""

meingraham, 2019-08-04:
@gemu2015 

The documentation mentions that the conditionals may not be enclosed in brackets. Can logic be enclosed in parentheses? from the example in the documentation

```
if a==b  
and x==y  
or k==i {  
  => do this  
} else {  
  => do that  
}  
```

can I specify (extra parentheses for clarity)?

`(a==b) and ( (x==y) or (k==i) )`

gemu2015, 2019-08-05:
@meingraham 
the current implementation is strictly sequential line by line, no support for bracketed expressions.
this would require a substantial rewrite of the expression evaluation and probably force the use of brackets in all cases

gemu2015, 2019-08-05:
ups was much easier then expected (half an hour recoding)
needs more testing but coming soon

spakecdk, 2019-10-21:
Would it be possible for a variable to read a mqtt payload?

gemu2015, 2019-10-21:
MQTT messages can be decoded in section >T and >E

e.g 

v1=ENERGY#Total

https://github.com/arendst/Sonoff-Tasmota/wiki/Scripting-Language

giessener, 2019-11-27:
When I use code like this:

```
>D
>b
if bt[1]==1
then
=> print Hello world
endif
```

Then the code is executed correct, but the relais is toggled too.

```
10:26:18 APP: Knopf1 Mehrfachdruck 1
10:26:18  Hello world
10:26:18 SRC: Button
10:26:18 DSP: Power 1
10:26:18 MQT: stat/EG-ELTERN-Sensor/RESULT = {"POWER1":"OFF"}
10:26:18 MQT: stat/EG-ELTERN-Sensor/POWER1 = OFF
10:26:19 CFG: in Flash gespeichert am 3F8, zählen 257, Bytes 4096
```

When I use Rules, the Rule is executed INSTEAD of the Relais. 
Is the mentioned this way? Or is this a bug?

ascillato, 2019-11-27:
That behaviour is intended.

giessener, 2019-11-27:
Can I disassociate the button and the relais in script language like it is possilble with rules? I can't find a way to use the buttons only in my script without toggling the relais.

bkirkman, 2020-01-03:
I see the answer to the question by @spakecdk , but I'm still having trouble trying to figure out how to set a variable from an MQTT message that originates from somewhere other than the local Tasmota device running the script.

For my specific application, I want to be able to change a thermostat setpoint in the Tasmota script from another MQTT device/message. With Rules, this could be done by sending a message to cmnd/mqttTopic/mem<x> or cmnd/mqttTopic/var<x>. By reading the scripting documentation, it might seem that this could be accomplished in some fashion by using the subscribe command (which I have compiled into the firmware) and listening in the >E section of the code.

Unfortunately, I've tried dozens of different combinations and have been unsuccessful. Any pointers from anyone on how to accomplish this in the scripting engine? Thanks!

giessener, 2020-01-04:
You can send a mqtt message  like cmnd/tasmota/Script >Temp=22 to change the var Temp in your Script. This works for me.
I also managed to disassociate the Buttons from the Relais by setting the buttontopic to something useless.

ascillato, 2020-01-04:
If you use buttontopic, you have to be aware that when mqtt or wifi got disconnected, your button will start controlling relay1

bkirkman, 2020-01-04:
Thank you both for the response. I was eventually able to write to the variable using the subscribe method. But the the way that @giessener showed was much more straight forward and worked perfectly. Plus I don't have to work about making sure Tasmota is subscribed to a topic. Thanks for the help!

giessener, 2020-01-04:
@ascillato I know that, the buttontopic is a workaround for me. Do you have a better method to control things in the script without toggle the relay?

ascillato, 2020-01-04:
@giessener I don't know. May be @gemu2015 knows

gemu2015, 2020-01-04:
one option would be to omit the tasmota setup and handle iOs yourself.

\>D
pin0=0
\>B
; set GPIO zero as input with pullup
spinm(0 2)
; set gpio 13 as relay output
spinm(13 1)
\>F
pin0=pin[0]
if chg[pin0]>0
and pin0==0
then
print pin changed 
print toggle relay
if pin[13]>0
then spin(13 0)
else spin(13 1)
endif
endif

or you only define the relays in tasmota and the button in scripter

\>D
pin0=0

\>B
; set GPIO zero as input with pullup
spinm(0 2)
; relay 1 defined in tasmota

\>F
pin0=pin[0]
; if pin changed
if chg[pin0]>0
; and pressed
and pin0==0
then
; toggle power
if pwr[0]==0
then ->power0 1
else ->power0 0
endif
endif

paolomorabito, 2020-01-26:
Is there a way to compare strings? E.g.

myText = "Hello World"
if myText.contains("World") then .... else ....

maxela, 2020-03-02:
I'm unsure about a feature of scripts described in the docu called "dynamic or self modifying code":
>```=(svar)``` executes a routine whose name is passed as a string in a variable

This didn't work for me. Can you please explain it in more detail? 

Example:
```>D
svar=""

>B
svar="sub1"

>S
=(svar)

#sub1
=>print sub1 was called
#``` 

An I using it wrong?

martinvana, 2020-07-06:
I love the scripting - much more logical flow (at least for people with programming background). I have one project up and running where "rules" approach was failing due to an inacceptable lag. Scripting rules! ;-)
Now I reached a point when I may need help:
Compiled with (among other) ```#define USE_SCRIPT_STATUS```
Yet I didn't figure out how to trigger the >U section.
All I am trying to do is setting a script variable from a JSON. Didn't find a complete example or good definition of syntax anywhere.
I am doing my best to send a JSON message either through Tasmota Console or through "Publish a Packet" in Home Assistant Developer Tools panel (MQTT)
`cmnd/PIR_entrance/STATE {"TwilightRange":44}`
(Not sure whether there should be "=" before the left curly bracket, tried all combinations.
Also the formatting with "hash" like the one from an example below isn't clear to me.
`rssi=Wifi#RSSI`
What would be corresponding JSON packet to match?
However, this is not my imminent problem since the code in the >U section doesn't even get triggered. When I send a JSON message, the teleperiod ">T" section get executed, though.
Simplified script is below. I guess I am missing something fundamental?
(The `>P >b >S` sections are not exactly relevant - except for fixing my ego that the code actually can do anything meaningful ;-) )
Thanks - Martin

```
>D	; define all vars here
; PIR_Light attempt 20200705_2111
: Just (desperate) testing how to set variable out from JSON payload
t_minute=59
ssid=""
tm_twlt=25

>B ; Boot section
print =========== >B section  =============
dp0
=>print Power1=%pwr[1]%

>T
print =========== >T section  =============
=>Publish tele/%topic%/LIGHT {"Script": {"TwilightRange":%tm_twlt%}}

; Read the status JSON payload
>U
print =========== >U section JSON payload  =============
rssi=Wifi#RSSI
print RSSI= %rssi%
tm_twlt=Time#TwilightRange
print >U %tm_twlt%

>P
=>print Power1=%pwr[1]%
>b
if bt[1]==0 {  
 print =========== >b section  =============
 print TwilightRange %tm_twlt%
}

>S
if t_minute<=0 {
 print Minute Passed
 t_minute=59
} else {
 t_minute-=1
}
#
```

Jason2866, 2020-07-06:
@martinvana  Probably it is better to ask directly for help in gemu2015 github
https://github.com/gemu2015/Sonoff-Tasmota

gemu2015, 2020-07-06:
@martinvana 
ok doku is completely misleading according >U. >U is just for receiving JSON payloads coming from status cmds  send by the script itself or the console. e.g 
->status 11
generates a JSON from which you may get 
rssi=Wifi#RSSI

the notation of JSON access is equivalent to rules  the # char separates elements of the JSON payload
in contrast to rules however the variables are case sensitiv. 

the >T section is called from sensors (TelePeriod JSON messages) 

the >E section ist called from results (RESULT  JSON messages)

if you want to set a variable in a script you may simply set it by sending
e.g. 
script >tm_twlt=30  (also from console)

martinvana, 2020-07-06:
Yay! Thanks @gemu2015 !
the ```status 11``` typed in console sure triggers the >U. Need to learn how to parse with the "#" but that is secondary. Just less of a "mystery".
Console entry ```script >tm_twlt=30``` does set the variable. I can keep writing and debugging parameterized script :-)
The remaining point I am missing: How to achieve the same from JSON payload?
What should be the MQTT prefix and topic? And how to format the payload?

Once I get through this "learning experience" - where is the best place to share my example(s) to augment the documentation?
Thanks again!

mattiassjogren, 2020-07-06:
Funny... I have just started to tinker with scripts and got stuck at the same spot as you as I want to send data to my tasmota device... Just a couple of hours after you.

I have not managed to send a correct MQTT message using home assistant dev tools but with a stand alone MQTT client I can set a variable using the following...

Topic: cmnd/tasmota_BExxxx/Script
Payload: >tm_twlt=30

Replace tasmota_BExxxx with whatever you have as topic in your MQTT configuration.

martinvana, 2020-07-06:
@mattiassjogren Thank you! 
Many details and I never got all of the combinations right
topic: cmnd/#/Script (where # substitutes the correct topic)
and payload without the curly braces formatted like you suggested was the answer.
Thanks!
```cmnd/PIR_entrance/Script >tm_twlt=33```

saidlm, 2020-08-09:
Is there any possibility to receive and decode MQTT messages generated by Domoticz? I am trying to connect BECA (Tuya) thermostat to domoticz via MQTT. I have success with reporting to domoticz but I am not able to find way how to set thermostat values from Domoticz.
Well I know there is a way to do it via NodeRed but it looks to me too complicated and I am looking for another "direct" way instead.

Thanks.

giovcris, 2021-07-23:
Sorry, how can I set the timezone in the script? And in what section? 
I have the same doubt to release commands as "Setoption X" and others that usually I send by console
Thanks for the answer

freifunkerffm, 2022-03-27:
I struggle with the command "swb" which I interpreted as binary.
Because I need to send OP-Codes in HEX like 80 83 87 (Decimal 128 131 131) via serial to control a device.
Unfortunately it looks like that "swb" is clinging somehow to the ASCII Code.
Is there a way to send hexadecimal values via serial triggered by script?
Before you ask, if I send this OP-Codes via "CuteCom" Terminal Program on my PC in HEX it woks so it is definitely Tasmota / Script related.
Thanks for an hint.

gemu2015, 2022-03-27:
swb() writes a single byte to the serial port
so your sequence would be sent by

swb(0x80)
swb(0x83)
swb(0x87)

freifunkerffm, 2022-03-31:
Just did exactly this and just got 00 00 00 in "CuteCom" on the monitoring Serial USB-Device, so somehow this is not working as expected.
As an example my current script running on: Tasmota Version 2022.01.3(tasmota)-2_7_4_9(2022-01-25T09:33:35)
>D
RCL=0

>B
so(3 1 115200)
delay(50)
=>print Serial setup done so(3 1 115200)

>S
if upd[RCL]>0
then
=>print Transmit received
swb(0x80)
delay(50)
swb(0x83)
delay(50)
swb(0x87)
delay(50)
endif

Output in CuteCom:
00000000 00 00 00

gemu2015, 2022-03-31:
swb() is now fixed

freifunkerffm, 2022-04-01:
Good Morning,
I have sourced Tasmota again from GitHub and compiled again a version that is supporting script this morning as the status has changed to "merged".
Unfortunately the HEX-output has now changed from "00 00 00" to something different but also all the same: 
00000000 c0 4e c0 4e c0 4e                                 	.N.N.N

Can you may have a closer look?
Thanks

gbananachewicz, 2022-05-30:
I have Wemos D1 mini and can't read values (only in script! of DS18b20 on  D0 GPIO16 and analog value of A0 connected to Capactive soil moisture sensor v 1.2

SENSOR = {"Time":"2022-05-29T14:57:55","ANALOG":{"pH0":13.55},"DS18B20":{"Id":"0000000C2CDF","Temperature":22.6},"TempUnit":"C"}

 

value of analog in  Rules (not Scripting) is shown under ANALOG#pH0. Temperature not checked

Tasmota 11.1.0

How to display values of PH and TEMPERATURE in tasmota Script ?

gbananachewicz, 2022-05-30:
Works ! . Thanks