arendst/Tasmota #5690: Fatal exception (10k per day!) Wemos D1 with PZEM016

xCapeITA, 2019-04-23:
I have bound a Wemos D1 mini to a PZEM016 and configured the tasmota as in the guide, but D1 keeps restarting like every minute , with 10.000 reboots per day.
Reason for reboot is a fatal exception.
I tried more than one D1 and more than one PZEM with no luck. Without GPIO configuration D1 has no issue
Thanks

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Wemos D1mini
- [X] Tasmota binary firmware version number used : 6.50 precompiled
- [X] Provide the output of command ``status 0`` :

00:00:00 Project sonoff Sonoff Version 6.5.0(release-sensors)-2_3_0
00:00:00 WIF: Connecting to AP1 FASTWEB_DGX in mode 11N as pzem016-7967...
00:00:03 RSL: tele/pzem016/SENSOR = {"Time":"1970-01-01T00:00:03","ENERGY":{"TotalStartTime":"2019-04-20T15:02:23","Total":5313.981,"Yesterday":0.029,"Today":5313.952,"Power":605,"ApparentPower":666,"ReactivePower":278,"Factor":0.91,"Frequency":50,"Voltage":234,"Current":2.841}}
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on pzem016-7967 with IP address 192.168.1.138
08:43:42 MQT: Attempting connection...
08:43:44 MQT: Connected
08:43:44 MQT: tele/pzem016/LWT = Online (retained)
08:43:44 MQT: cmnd/pzem016/POWER = 
08:43:44 MQT: tele/pzem016/INFO1 = {"Module":"Generic","Version":"6.5.0(release-sensors)","FallbackTopic":"cmnd/DVES_32DF1F_fb/","GroupTopic":"sonoffs"}
08:43:44 MQT: tele/pzem016/INFO2 = {"WebServerMode":"Admin","Hostname":"pzem016-7967","IPAddress":"192.168.1.138"}
08:43:44 MQT: tele/pzem016/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40238690 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
08:43:52 MQT: tele/pzem016/STATE = {"Time":"2019-04-23T08:43:52","Uptime":"0T00:00:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"Wifi":{"AP":1,"SSId":"FASTWEB_DGX","BSSId":"A0:64:8F:DF:88:E9","Channel":1,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:07"}}
08:43:52 MQT: tele/pzem016/SENSOR = {"Time":"2019-04-23T08:43:52","ANALOG":{"A0":11},"ENERGY":{"TotalStartTime":"2019-04-20T15:02:23","Total":5313.984,"Yesterday":0.029,"Today":5313.955,"Period":3,"Power":572,"ApparentPower":636,"ReactivePower":277,"Factor":0.90,"Frequency":50,"Voltage":235,"Current":2.707}}
08:44:06 CMD: STATUS 0
08:44:06 MQT: stat/pzem016/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"pzem016","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:44:06 MQT: stat/pzem016/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Exception","Uptime":"0T00:00:32","StartupUTC":"2019-04-23T07:43:34","Sleep":50,"CfgHolder":4617,"BootCount":10377,"SaveCount":10411,"SaveAddress":"FB000"}}
08:44:06 MQT: stat/pzem016/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sensors)","BuildDateTime":"2019-03-19T12:26:22","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
08:44:06 MQT: stat/pzem016/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FASTWEB_DGX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
08:44:06 MQT: stat/pzem016/STATUS4 = {"StatusMEM":{"ProgramSize":544,"Free":456,"Heap":12,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16400E","FlashMode":3,"Features":["00000809","0FDEE394","0007A3A4","B7FFBFCC","005ABBC0"]}}
08:44:06 MQT: stat/pzem016/STATUS5 = {"StatusNET":{"Hostname":"pzem016-7967","IPAddress":"192.168.1.138","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"3C:71:BF:32:DF:1F","Webserver":2,"WifiConfig":4}}
08:44:06 MQT: stat/pzem016/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.119","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32DF1F","MqttUser":"brokers","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:44:06 MQT: stat/pzem016/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 23 07:44:06 2019","Local":"Tue Apr 23 08:44:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:44","Sunset":"19:53"}}
08:44:06 MQT: stat/pzem016/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:44:06 MQT: stat/pzem016/STATUS10 = {"StatusSNS":{"Time":"2019-04-23T08:44:06","ANALOG":{"A0":12},"ENERGY":{"TotalStartTime":"2019-04-20T15:02:23","Total":5313.986,"Yesterday":0.029,"Today":5313.957,"Power":572,"ApparentPower":634,"ReactivePower":275,"Factor":0.90,"Frequency":50,"Voltage":235,"Current":2.703}}}
08:44:06 MQT: stat/pzem016/STATUS11 = {"StatusSTS":{"Time":"2019-04-23T08:44:06","Uptime":"0T00:00:32","SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"Wifi":{"AP":1,"SSId":"FASTWEB_DGX","BSSId":"A0:64:8F:DF:88:E9","Channel":1,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:07"}}}

- [ ] Provide the output of console when you experience your issue if apply :

Jason2866, 2019-04-23:
Fatal Exception 0 is most of the time a power issue. WemosMini often do have bad Vregs...

ascillato2, 2019-04-24:
Closing this issue as it is **duplicated**. Sorry.

https://github.com/arendst/Sonoff-Tasmota/issues/5673
https://github.com/arendst/Sonoff-Tasmota/issues/5554
https://github.com/arendst/Sonoff-Tasmota/issues/4995
And others

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-24 by ascillato2