arendst/Tasmota #5692: NFC PN532 Freeze

pcpaulo, 2019-04-23:
I am using tasmota version 6.5.0 self compiled with sonoff Basic.

I have installed a RFC card PN532, after reboot he works fine but after few hours ( normally next day) the RFC card are frozen all others funcionallity works fine.

Any direction for this issue?

Jason2866, 2019-04-23:
Try it to run via a wemos mini. Since you have not provided the infos asked in issue template no advice can ne given or error hunting is possible.

ascillato2, 2019-04-24:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-24 by ascillato2