arendst/Tasmota #5697: Two PZEM-004T on single ESP8266

s89511, 2019-04-24:
Hello, I'm using an ESP8266 NodeMCU with two connected PZEM-004t with a custom firmware I wrote by myself. I would like to use Tasmota and I would like to know if it would be possible to configure two PZEM-004t.

Thanks

Jason2866, 2019-04-24:
Already answered... Not supported.

ascillato2, 2019-04-24:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-24 by ascillato2