arendst/Tasmota #5698: Add HTTP Post & Headers to WebSend

hongtat, 2019-04-25:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Allow WebSend to use HTTP POST and set HTTP headers (e.g. content-type application/json)

**Describe alternatives you've considered**
None

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2019-04-25:
What is the use case for such a feature?

Jason2866, 2019-04-25:
See https://github.com/arendst/Sonoff-Tasmota/pull/5273#issuecomment-465088242

hongtat, 2019-04-25:
For RF & IR bridge, I'd like to send the received codes to another server/hub to process.

I'm using Rules and WebSend, but it's rather slow as the remote server hangs (~5sec) on HTTP GET (without content-length header) before responding. This also resulted in "{"WebSend":"Connect failed"}"

The remote sever has no problem with HTTP POST and I can customize the content.

Jason2866, 2019-04-25:
The problem is the remote server ;-)
Dont expect that this feature will be implemented from a developer of Tasmota. 
If you want to see this feature you have to do a PR. If code size increase is small there is a chance that PR will be merged.

ascillato2, 2019-04-25:
Thanks for sharing your ideas. 

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-25 by ascillato2

lynema, 2020-08-13:
Depending on what the use case is, this should be re-opened. 

This was something that had to be worked around due to most media player APIs taking a POST to executed a command and not a GET.   

Examples: 
https://github.com/open-dynaMIX/simple-mpv-webui -- webui to remotely control mpv -- needs to have a POST to play music
https://wiki.videolan.org/VLC_HTTP_requests/ -- vlc web api needed basic auth without a username which didn't seem to work with websend.  Being able to send an auth header would help here.
https://github.com/hyperblast/beefweb -- remote control for foobar2000 -- needs to have POST to play music

In a couple of these examples, just being able to send a POST request  with a URL would've worked since the parameters can be pulled from the request URL.

iquix, 2020-12-29:
@hongtat is a author of the "TASMOTA Connect" smartapp in Smartthings platform.
Since Smartthings Hub requires POST method to receive data properly, it is crucial to have POST support in TASMOTA to work properly with Smartthings Hub. If you use GET method to send data to Smartthings Hub, the communication hangs for a while.

Please support POST method in WebSend.

Jason2866, 2020-12-29:
> See [#5273 (comment)](https://github.com/arendst/Tasmota/pull/5273#issuecomment-465088242)

is still valid. Tasmota is designed to use mqtt.

iquix, 2020-12-30:
It's not what this thread is about. 

What we're talking is HTTP client(websend), not HTTP server.

Websend works as HTTP client, which only supports GET method. What we need is POST method as a HTTP client.

iquix, 2020-12-30:
There are a lot of new features that are added to TASMOTA. I don't think we have to limit the possibility of TASMOTA. I think Websend POST could be much more useful than other new features.

s-hadinger, 2020-12-30:
I'm not sure POST without TLS makes real sense, unless you have a strong use case. 

But TLS for HTTPS as a client is not on the roadmap.

Jason2866, 2020-12-30:
@iquix  The `m` in Tasmota stands for mqtt. Theo and the known developers contributing to the code will not do. If a PR is provided it will be considered. Always keeping the code size increase in mind. 
I disagree new features are always nice to have in Tasmota.