arendst/Tasmota #5699: Use WIFI_LIGHT_SLEEP also when Dynamic sleep is used

sislakd, 2019-04-25:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Why there is used only WIFI_MODEM_SLEEP mode when Dynamic sleep is used? Here is snipped from the code in master:
```
void WiFiSetSleepMode(void)
{
#if defined(ARDUINO_ESP8266_RELEASE_2_4_1) || defined(ARDUINO_ESP8266_RELEASE_2_4_2)
#else  // Enabled in 2.3.0, 2.4.0 and stage
  if (sleep && Settings.flag3.sleep_normal) {
    WiFi.setSleepMode(WIFI_LIGHT_SLEEP);  // Allow light sleep during idle times
  } else {
    WiFi.setSleepMode(WIFI_MODEM_SLEEP);  // Disable sleep (Esp8288/Arduino core and sdk default)
  }
#endif
}
```
I'm using latest version of this project in combination with Core 2.3.0 as others are not working stable.

**Describe the solution you'd like**
Use WIFI_LIGHT_SLEEP even for Dynamic sleep. I've did local modification, and it working stable. There are no WiFi or MQTT drops at all after several days.

**Describe alternatives you've considered**
The only alternative I found is to use normal sleep but then run of various regular activities within Tasmota is irregular especially if normal sleep with 50 ms is used. Thus, I don't see any reason why we cannot use WIFI_LIGHT_SLEEP with Dynamic sleep mode (which is default now).

Jason2866, 2019-04-25:
Because using both is not necessary.
See wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Dynamic-Sleep
Using normal sleep has impact to device responsive. It doesnt has positive impact to lower down power usage if used with dynamic sleep

sislakd, 2019-04-25:
With WIFI_LIGHT_SLEEP it seems that CPU is really paused and it brings additional decrease of power usage even Dynamic 50 ms is used. With Dynamic 50 ms in WIFI_LIGHT_SLEEP brings decrease of current down to 1-2mA while sleeping.

Jason2866, 2019-04-25:
The current implementation is tested and is working very well on all cores.
There will be no special version which has only support for one core.
There are more drawbacks then advantages.
It is Open Source. Feel free to Fork and implement the way you want.

sislakd, 2019-04-25:
For me, the current version is not working for any core above 2.3.0.  Frequent disconnects of both WiFi and MQTT unless changed to WIFI_NONE_SLEEP with Sleep 0. Which has way high power consumption and ESP8266 chip is quite hot. Do you know which WiFi router is used by those who don't see WiFi/MQTT disconnects on newer cores or staging core using Dynamic 50ms sleep? I've tested ASUS, TP-LINK, Cisco with more than 20 ESP8266 connected and there are frequent disconnects causing noticeable latency in actions. 

On the other hand, in the code there are a lot of preprocessor conditions for various core versions. Thus, I was wondering why we are not using optimal combination for core 2.3.0 which is distributed from you and used by many devices which can save a lot of energy in total.

Jason2866, 2019-04-25:
Sleep 0 in arduino core will fail with bad designed devices (power supply and thermal design)
Sonoff Basic is such a device. 
This was the reason for developing dynamic sleep after a long bug hunting to find the reason for not working Tasmota on some devices....
Wifi disconnecting is a error behaviour of devices which doesnt have a sufficent power supply.

Wifi disconnects with core 2.4.2, 2.5.0 and Stage is a different issue. NOT related to dynamic sleep function.

With dynamic sleep you save energy in total as nearly as with wifi sleep in core 2.3.0
AND it works with ALL cores AND devices doesnt lag as with wifi sleep.

There will be no change in implementation.  

P.S. I use Routers and APs from LANCOM and Bintec Elmeg and have no reconnect with any core.
Known to work well are APs and routers from mikrotik.

sislakd, 2019-04-25:
Ok, to summarize it if I understand it properly:

- cores 2.4.2, 2.5.0 and Stage have no WiFi disconnects only for some AP vendors like LANCOM, Bintec Elmeg and mikrotik regardless sleep configuration; in other words they are unstable if different AP is used
- dynamic sleep 50ms with with WIFI_MODEM_SLEEP for Stage core gives similar power saving like 0ms sleep with WIFI_LIGHT_SLEEP in 2.3.0
- there is no additional benefit for core 2.3.0 to use combination of dynamic sleep 50ms and  WIFI_LIGHT_SLEEP

My observation is slightly different:
- core 2.3.0, 1 ms normal sleep works well
- core 2.3.0 50 ms normal sleep cause disconnects
- core 2.3.0 50 ms dynamic sleep works well consuming even less energy than with 1ms normal sleep as CPU is stopped (current consumption drops down to 1-2mA) for quite a long time without those 1ms peaks when CPU is running loop scanning through all drivers and sensors to wait for the next 50 ms interval where more things is done usually

Jason2866, 2019-04-25:
> Ok, to summarize it if I understand it properly:
>
> cores 2.4.2, 2.5.0 and Stage have no WiFi disconnects only for some AP vendors like LANCOM, Bintec Elmeg and mikrotik regardless sleep [...]

- No, i just have devices from LANCOM and Bintec and cant speak for other brands. 
  There are user feedback others brands work too.
- Dynamic sleep saves a lot of power because CPU is idling (no wifi sleep!).
   saves more energy than sleep 0 with wifi light sleep 
- yes

closed 2019-04-27 by ascillato2