arendst/Tasmota #5700: TSL2591

SargonofAssyria, 2019-04-25:
### BUG DESCRIPTION
The TSL2591 displays only 0.000 lx

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sensors
- [x] Tasmota binary firmware version number used : 6.5.0.9 / (self-compiled)
- [x ] Development IDE - Compiler / Upload tools used : VSC 1.33.1 / esptools
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
09:15:51 MQT: stat/SDS011/STATUS = {"Status":{"Module":0,"FriendlyName":["Buiten"],"Topic":"SDS011","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:15:51 MQT: stat/SDS011/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:35","StartupUTC":"2019-04-25T08:08:16","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":24,"SaveAddress":"3F4000"}}
09:15:51 MQT: stat/SDS011/STATUS2 = {"StatusFWR":{"Version":"6.5.0.9(sensors)","BuildDateTime":"2019-04-25T09:23:42","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
09:15:51 MQT: stat/SDS011/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"SysLog":1,"LogHost":"192.168.2.xx","LogPort":514,"SSId":["RhenenXX","RhenenYY"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
09:15:51 MQT: stat/SDS011/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":2480,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDEE394","0407A3A5","B7FFBFCC","01DABBC0"]}}
09:15:51 MQT: stat/SDS011/STATUS5 = {"StatusNET":{"Hostname":"SDS011-5761","IPAddress":"192.168.2.xx","Gateway":"192.168.2.xx","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.248","Mac":"80:7D:3A:6E:xx:xx","Webserver":2,"WifiConfig":4}}
09:15:51 MQT: stat/SDS011/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6EF681","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:15:51 MQT: stat/SDS011/STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 25 08:15:51 2019","Local":"Thu Apr 25 09:15:51 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:19","Sunset":"19:50"}}
09:15:51 MQT: stat/SDS011/STATUS10 = {"StatusSNS":{"Time":"2019-04-25T09:15:51","BME280":{"Temperature":25.9,"Humidity":38.0,"Pressure":1002.2},"TSL2561":{"Illuminance":0.000},"SDS0X1":{"PM2.5":1.3,"PM10":3.3},"PressureUnit":"hPa","TempUnit":"C"}}
09:15:51 MQT: stat/SDS011/STATUS11 = {"StatusSTS":{"Time":"2019-04-25T09:15:51","Uptime":"0T00:07:35","SleepMode":"Dynamic","Sleep":50,"LoadAvg":55,"Wifi":{"AP":1,"SSId":"RhenenXX","BSSId":"58:23:8C:EA:xx:xx","Channel":1,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:02"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
The TSL2591 does not display any output.
The SDS011 and BME280 work fine. So the I2C bus is working fine.
The EXCACT same hardware and wiring is working fine under ESPEasy-mega-20190413
So the TSL2591 is connected the right way and producing output under ESPEasy.
I would like change to Tasmota because it seems more stable.
ESPEasy is rebooting almost every day, even more the once a day.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-25:
Driver enabled in my_user_config.h?

SargonofAssyria, 2019-04-25:
Oeps, No typo, so I misread the devicetype. So back to ESPEasy again, or buy a TSL2561

joba-1, 2019-04-25:
Log says tsl2561, probably enabled the wrong driver?

Jason2866, 2019-04-25:
Please close issue.
THX

closed 2019-04-25 by SargonofAssyria