arendst/Tasmota #5701: Feature request, LED bar/pointer via MQTT

myozone, 2019-04-25:
Simple request of MQTT message turns on Pixel bar or pointer. I.E. pixel length of 150 pixels and a MQTT mesage of say of 85 turns on 85 pixels with a specific colour

gemu2015, 2019-04-25:
i implemented this already a long time ago. its a led bar with ws2812 . you may try it on my fork
#define USE_LEDBAR

myozone, 2019-04-25:
Sorry I don't understand your answer could you explain more please /

gemu2015, 2019-04-25:
i have a fork of tasmota where i implemented a ledbar. you may test my fork at =>https://github.com/gemu2015/Sonoff-Tasmota. if there is more interest in the ledbar i can make a pr (but i already did with no response)

myozone, 2019-04-25:
Is there a simple way of getting your code onto my d1 mini or one of those Helight dongles - I was hoping to find a bin file :-)
All I'm trying to do is get my house power on the bargraph or pointer dot. I already have node-red storing power data

gemu2015, 2019-04-25:
here is a link to a wemos binary => https://www.dropbox.com/s/brbc8fhkp0uhain/wemos-d1.bin?dl=0
when flashed setup the number of pixels. You can use the LEDs as usual with dimmer etc.
only if you turn off the light you can set the ledbar with ledbar value. (default setup is -5000 to + 5000) in my case watts of house power in out with solar panels.

default is +-5000 units in 5 steps each<br>
ledbar xxx =\> show value<br>
ledbar rxxx =\> set range positiv values =\> positiv up, else negativ up<br>
ledbar sxx =\> set steps<br>
only in use when LED Power is off, else ignored<br>

myozone, 2019-04-26:
Gosh, Thank you very much ! Interesting to know someone else is doing a similar ideas to me.

blakadder, 2019-05-16:
Please close this issue if your problem is solved.

myozone, 2019-05-16:
Many thanks to gemu2015

closed 2019-05-16 by myozone