arendst/Tasmota #5703: Smarteefi Module 8

Shaileshverma123, 2019-04-25:
Hey Guys,
I got a device called Smarteefi module 8... i am trying to figure out the way to work with my new firmware... but i am not able to figure out input/output properly. Can anyone help me about the same...
Thank you

ascillato2, 2019-04-25:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-25 by ascillato2

dibyadip-das, 2020-10-25:
@Shaileshverma123 Did you try ESPHome? I am interested to look at the board, Can you share some pictures?