arendst/Tasmota #5705: Hi,

johnsethwatkins, 2019-04-25:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

closed 2019-04-25 by johnsethwatkins

ascillato, 2019-04-25:
> Hi,
> 
> In an effort to avoid a problem, rather than create one and then try to fix it I note that HA release .92 today (April 25th) includes;
> 
> MQTT - MQTT discovery will not longer implicitly set state_topic except for mqtt.alarm_control_panel, mqtt.binary_sensor and mqtt.sensor. (@emontnemery - #22998) (mqtt docs)
> 
> Is this likely to impact discovered (option19 enabled) nodes running tasmota please?
> 
> Thanks!

Tasmota is running ok with latest Hass.io

emontnemery, 2019-04-25:
Yes, this change should be compatible with Tasmota option 19.