arendst/Tasmota #5706: Sonoff Basic LED keeps flashing, despite working fine

Timmiej93, 2019-04-25:
### BUG DESCRIPTION
I've flashed an old Sonoff Basic I had lying around (so not the R2). Everything went fine and as expected, and it works just fine, but the LED keeps toggling every 1.5 seconds. I've tried messing with the LedState command, but nothing seems to affect the LED. Toggling the relay doesn't change anything either.

Using the precompiled 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (i.e. Sonoff Basic) : Sonoff Basic
- [ x] Tasmota binary firmware version number used : 6.5.0.8/ pre-compiled
- [ x] Development IDE - Compiler / Upload tools used : NodeMCU PyFlasher
- [ x] Provide the output of command ``status 0`` :
```

Status 0
```
16:59:42 CMD: status 0
16:59:42 SRC: WebConsole from 192.168.0.32
16:59:42 RSL: Received Topic /status, Data Size 1, Data 0
16:59:42 RSL: Group 0, Index 1, Command STATUS, Data 0
16:59:42 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:59:42 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:11","StartupUTC":"2019-04-25T15:55:31","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":9,"SaveAddress":"FB000"}}
16:59:42 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:59:42 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID HERE",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
16:59:42 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
16:59:42 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3532","IPAddress":"192.168.0.198","Gateway":"192.168.0.2","Subnetmask":"255.255.252.0","DNSServer":"192.168.0.254","Mac":"5C:CF:7F:72:CD:CC","Webserver":2,"WifiConfig":4}}
16:59:42 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_72CDCC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:59:42 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 25 15:59:42 2019","Local":"Thu Apr 25 16:59:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:40","Sunset":"19:55"}}
16:59:42 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-04-25T16:59:42"}}
16:59:42 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-04-25T16:59:42","Uptime":"0T00:04:11","Vcc":3.174,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SSID HERE","BSSId":"00:0C:F6:5E:A7:20","Channel":7,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

### TO REPRODUCE
Flash the 6.5.0 pre-compiled release and connect to WiFi.

### EXPECTED BEHAVIOUR
Instead of the LED toggling every 1.5 seconds, I expect it to work according to the LedState it's set to.

Jason2866, 2019-04-25:
See wiki commands...
Use `Backlog LedPower 0; SetOption31 1`

ascillato, 2019-04-25:
Led toggling means no MQTT configured. Please, disable it in the extra configuration menu

Timmiej93, 2019-04-25:
Well, that works, thank you. No need to be arrogant about it though. It's great info if you know what's causing the flashing, but I didn't know what was causing it. I remember from way back when that there was a table that had all the flash patterns for the LED and their meanings, but I couldn't find that either. 
Anyway, I know what it is now, I know how to fix it, so I'm good. Thanks again.

closed 2019-04-25 by Timmiej93

Jason2866, 2019-04-25:
No one was arrogant. Just explaining what causes this behaviour. 
What else we can do?

ascillato, 2019-04-25:
Please, do not take a short and direct answer as being arrogant. No one meant to do that.

Timmiej93, 2019-04-25:
We all know what the three dots mean, plain and simple. A direct answer would have been "This is caused by X, you can find the info on how to fix it here: ...". "See wiki commands..." is definitely arrogant, and isn't very helpful. Now we can argue about this all day, but we both know there was passive aggressiveness in that line. Since the issue is resolved, I'd say there's no reason to continue the conversation.

Also, ascillato, please understand I was not talking about you, you replied just before I posted my comment.

ascillato2, 2019-04-25:
@Timmiej93 

I understood that your complain was directed to Jason. Please, do not take a short message as passive agressiveness and as an attack to you. No one wanted that. We all want to make Tasmota better and useful for everyone that wants to make a Smart Home. Remember that Tasmota Community is global with people from all over the world, so communication is not 100% precise (when writing or reading) due we all speak different languages. This types of misunderstandings NEEDs to be tolerated and the focus should be only in the Technical stuff of TASMOTA.

Hope everyone can chill out and share their ideas for Tasmota.

Also, remember that the Chat is a really good place to make questions and that in issues, several problems like this one had already being addressed and solved.

blakadder, 2019-04-25:
@Timmiej93 

Definition
An ellipsis (plural: ellipses) is a punctuation mark consisting of three dots.

Use an ellipsis when omitting a word, phrase, line, paragraph, or more from a quoted passage. Ellipses save space or remove material that is less relevant. They are useful in getting right to the point without delay or distraction:

jziolkowski, 2019-04-25:
@Timmiej93 you're so full of yourself... I can't even.

Timmiej93, 2019-04-25:
@blakadder Sadly, there's a difference between a definition and how things are used. By that definition, the only use I can think off is the "ommitting a word" bit. Usually, when people end a sentence in three dots, they want to add something like "you dumbass" (See wiki commands, you dumbass) for example. It's rarely anything nice.

@jziolkowski I'm full of myself because I dare to speak up against the passive aggressiveness that may put off kids starting with programming / tinkering? Honestly, I don't get it. 
I do understand that it sucks if you spend a lot of time in one area, and see the same "dumb" question asked over and over again, especially if the answer is out there to be found. However, if those questions annoy you, you probably shouldn't answer, or just suck it up. It's never a reason to become toxic.
Of course, people can do as they with, but so can I. I speak up, where others won't. Also, by editing your comment, you clearly show a bias towards the other party's side. If you guys want to be like that among each other, fine by me, but don't get all upset if somebody mentions it.

jziolkowski, 2019-04-25:
You are seeing things man, seriously. Also, the edits are offensive now?

ascillato2, 2019-04-25:
Locking issue as it have gone agressive and offtopic.

Please, chill out