arendst/Tasmota #5709: Link on device individual LEDs to Relays

blakadder, 2019-04-26:
**Have you look for this feature in other issues and in the wiki?**
You could say so.

**Is your feature request related to a problem? Please describe.**
Influx of new devices with multiple relays and leds connected to a separate gpio. See [this](https://blakadder.github.io/templates/DS-102_3.html) or [this](https://blakadder.github.io/templates/kesen_KS-604S.html)

Ledmask helps in certain situations but doesn't cover 3 and 4 switch boards with 4 individual leds.

**Describe the solution you'd like**
Way to map leds to relays so that f.e. Led3 turns on when Relay3 is turned on. 
This would also require a separate mapping for the "status" LED which usually displays wifi and mqtt connectivity.

**Describe alternatives you've considered**
Right now the only way to do it is not to have LEDs or to create dummy relays instead of LEDs and link them with rules. Apart from writing to flash twice as much this makes most smart home integrations worse since you get 3 mostly useless switches.

**Additional context**
Best implementation would be leds responding to relays according to their number by default but that would possibly introduce breaking changes for some of the modules.

Other solution could be linking leds to relays with a command, lets name it `LedLink<x>` so you can do f.e. 
`LedLink3 2,3`
which would make a relation between led3 and relay2 and relay3.

Other way this could be done is to introduce a command to control led's individually same as relays and then create rules to link them to relays. Perhaps `LedPower<x>` so we can do f.e.
`on power1#state do ledpower1 %value% endon`

jziolkowski, 2019-04-27:
Additionaly, might be useful to add a command like `statusled` which defaults to led1 keeping reverse compatibility, and can be assigned to use extra leds on devices which provide them, like many wall switches. This way, leds can be mapped 1:1 by default, just like relays are. For most users that would work out of the box, and for advanced usages it can give much opportunities. 

I also +1 @blakadder's `ledpower` idea.

blakadder, 2019-04-27:
Technically it's @meingraham idea

jziolkowski, 2019-04-27:
Oh ok. We've discussed it so many times I've lost track.

digiblur, 2019-04-27:
ledpower<x> would be so nice!  https://github.com/arendst/Sonoff-Tasmota/pull/5612#issuecomment-482100968

jziolkowski, 2019-04-27:
I just came up with another example application, when I went to disable the lights in the kids room.
My wall switch has button3 configured to toggle the desk lamp via publish. Using ledpowerX, the desk lamp could reply back to the wall switch enabling the led.

blakadder, 2019-05-16:
Just adding here that we had  several inquiries on discord (ours and others) by users with similar triple wall switches on how to individually control leds and the "set led as relay" hack wasn't warmly received.

meingraham, 2019-05-16:
Outlet strips with LEDs for each receptacle. Light switches with multiple switches/buttons/LEDs. These are the common requests - can each LED follow the state of its associated relay. So, either a means to do so via an LedPower<x> in a rule or a means to link a configured LED to a relay (or both). Setting up virtual relays to control LEDs ends up with a messy/confusing UI and convoluted rule sets.

gemu2015, 2019-05-20:
with a script you can control gpios without any tasmota definition
so you can assign any gpio output (led) with any other action 

>D
; define led pins (0-16) e.g.
led1=12
led2=13

>B
; set gpio direction to out
spinm(led1 1)
spinm(led2 1)

>E
; now on power (relay) x, switch led  
if pow[1]>0
then spin(led1 0)
else spin(led1 1)
endif

if pow[2]>0
then spin(led2 0)
else spin(led2 1)
endif

blakadder, 2019-05-20:
That's a great development 
I'm still favoring a default behaviour of "relay2 is linked to led2 and relay4 is linked to led4" as a more user friendly no fuss approach, if you want something special then use scripting

arendst, 2019-05-21:
See changelog about three new GPIO defines.

What has changed:

1) For legacy devices nothing has changed
2) Future AriLux devices with RF/IR control need to use GPIO option 159 replacing option 59.
2) Up to four leds are related to first four power outputs when a LedLink GPIO is configured too
3) Direct led control using command LedPowerX 0/1 when a LedLink GPIO is configured too

Jason2866, 2019-05-21:
Like it and the strange activating Arilux RF receive is gone too 😀

closed 2019-05-21 by ascillato2

mnbrgr, 2019-06-20:
I was just looking for exactly this functionality but it does not work as expected for me.

What I want/expect:
I have a Sonoff RF. I want to add an external LED that indicates the state of the build in relays of the Sonoff.

What I did:
I flashed commit ddb81c7268f15b4c007df284d8c592684a8c5466 from the development branch. I configured the device as Sonoff RF (2). I set GPIO14 to LedLink (157). On the console I called LedState 1.

What I get/observe:
Sonoff build in LED is indicating the state of the build in relays. External LED connected to GPIO 14 is blinking once whenever I switch the state of the relays by using the sonoff build in button, or the a configured RF remote, and stays off otherwise. It does not react at all when switching the state of the relays via webinterface and also not when switching the relays by a external button connected via some other GPIO.

meingraham, 2019-06-20:
To my knowledge, the Sonoff RF only has one relay. The device has two LEDs:

GPIO12 is connected to the relay and hardwired to the red LED as well.
GPIO13 is connected to the green LED configured as LED1i

Why do you need the external LED on GPIO14?

The component configured as LEDLink will only display the state of the Wi-Fi connection and the MQTT connection.

In any case, please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

mnbrgr, 2019-06-21:
> Why do you need the external LED on GPIO14?

I intend to put the Sonoff into a box. This box will get the external LED and an external button.
The external LED shall indicate the status of the Sonoff relays.

> The component configured as LEDLink will only display the state of the Wi-Fi connection and the MQTT connection.

Yes, as long as the Sonoff LED (green one) is configured to show this behavior, right? As mentioned above, I called LedState 1 from the console, which configures the LED to show the power state (LED on when power on). I expect the linked LED to show then the same behavior.

> In any case, please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for [...]

Thanks for this advice. I was not aware of this option. Since this feature is still very new and not yet released in the master branch, I thought it might be good to comment here in the original request. I will check out the links you mentioned.

meingraham, 2019-06-21:
Use of Discord is even more relevant for the stable development releases. Your are misinterpreting the use of LedLink. If through the use of the chat it is determined that indeed there is a bug in the software, then a GitHub issue is the next step and the fix will be incorporated into the next **development** release.

My recommendation - use the [Sonoff RF template](https://blakadder.github.io/templates/sonoff_RF.html) instead of the built-in module type and change the GPIO linked to your respective LEDs and modify the behavior of the green LED and your external LED.

{"NAME":"Sonoff RF-Modified","GPIO":[17,255,255,255,255,0,0,0,21,158,56,0,0],"FLAG":0,"BASE":2}

If you have any further questions, please contacts us on the Discord chat.

Regards.

Mike