arendst/Tasmota #5711: Disable certain button controls

agoralive, 2019-04-26:
Which file do I need to edit to disable a couple of the button controls?

I would like to disable a function if the button is pushed 2, 3, 5 and 7 times.

This to "avoid" faulty presses and really learn to disble these, still having a "real" button to turn on/off my lamps

meingraham, 2019-04-26:
Search the Tasmota GitHub for "press". The very first code module in the results seems a likely candidate.

ascillato2, 2019-04-27:
You can use rules for that. No need to modify the code. See wiki at rules examples. Thanks.

ascillato2, 2019-04-27:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-27 by ascillato2

agoralive, 2019-04-29:
This is not closed because of Rules, I wan't button HARD CODED and disabled!!! No Rules, if I press a button for 40 seconds then there should only be the button's I wan't possible.

Or can the rules, be coded and STAY coded after a 40 sec reset?

meingraham, 2019-04-29:
@agoralive did you look at the module I suggested (https://github.com/arendst/Sonoff-Tasmota/issues/5711#issuecomment-487287427)? As the feature that you are requesting is very limited in the user base wanting it (i.e., 1 out of millions), I think it's going to be something you have to maintain in a code module customized by you and compiled for purpose.

agoralive, 2019-04-29:
@meingraham - Yes I did do a search like that, but I'm not much of a coder, and I do not see ex 

1 press: xxxxx

2 presses:   xxxxxx
 
3 presses: xxxx

Anywhere.

I know my request is quite unique that's why I'm not looking for a feature, but guidens to where the changes should be done.

meingraham, 2019-04-29:
You looked in `sonoff/support_button.ino`?

[image]

agoralive, 2019-04-30:
@meingraham - First at all: THANK YOU FOR YOUR TIME!!!!!!!!!!!!!!!!

I have seen/found this section, but this for me is like a foreign language. I thought that somewhere there would be a "commented out" section where it stated //3 press do the following.

Or something like that.... 

BUT... I did find this section:

```
if (!Settings.flag.button_single) {                    // Allow multi-press
          if (multiwindow[button_index]) {
            multiwindow[button_index]--;
          } else {
            if (!restart_flag && !holdbutton[button_index] && (multipress[button_index] > 0) && (multipress[button_index] < MAX_BUTTON_COMMANDS +3)) {
              bool single_press = false;
              if (multipress[button_index] < 3) {
```     

If I changed the above to:

```
if (!Settings.flag.button_single) {                    // Allow multi-press
//          if (multiwindow[button_index]) {
            multiwindow[button_index]--;
          } else {
            if (!restart_flag && !holdbutton[button_index] && (multipress[button_index] > 0) && (multipress[button_index] < MAX_BUTTON_COMMANDS +3)) {
              bool single_press = false;

Would that eliminate the possibility of pressing multiple times?

If that is possible, then how do I accept 3 presses to start the WiFiManager?
              if (multipress[button_index] < 3) {
```

meingraham, 2019-04-30:
I'm right there with you on deciphering this bit of code :disappointed: Perhaps @ascillato or @arendst could assist.

Good luck.

Mike