arendst/Tasmota #5713: Sonoff 4ch pro r2 button problem and auto turn on itself

itajackass, 2019-04-26:
I've a strange problem with my sonoff 4ch pro r2. 

I've 2 of this, i have flashed using Atom and tasmota 6.4.1

The first is done without problem. 

Second one i get error during flash. So i try another times and get it flashed ok.

But this one has strange problems:

1. Relay 1 button work if I press it to start, but doesn't work if I try to stop it (i need to switch to another relay, then turn off with this second relay to stop all)

2. After some time ALL are stopped, a random relay turn yourself on :(

How can I fix this? Need to reflash tasmota with same procedure?

Jason2866, 2019-04-27:
Have you erased the whole flash with esptool?
If not please do. After that flash latest Tasmota version. You use a old version.
See https://github.com/arendst/Sonoff-Tasmota/wiki/Flashing
and https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.5.0/sonoff.bin

itajackass, 2019-04-27:
Hi thank for the suggest.
Can i erase and flash in the same moment using NodeMCU PyFlasher? Or you suggest do it in 2 step with another method?

Jason2866, 2019-04-27:
You could use pyflasher with checked `erase flash`

blakadder, 2019-04-27:
Are the dip switches setup correctly?

itajackass, 2019-04-27:
Yes, i've set S6 to 0 for interlock mode.
Today i'll try to reflash it.

ascillato2, 2019-04-27:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-27 by ascillato2