arendst/Tasmota #5714: Upper and Lower limit for Tuya-Dimmer

andreas-nn, 2019-04-26:
Hello,
My Tuya-dimmer is working good. (An aliexpress-Clone)
It drives two "Triac Dimmable LED Driver 7-20x 1W"
But the useful range is from 12 to 35 percent.
Lower: the dimmable LED-driver do not start up.
Higher: there is no difference between 35 an 100%.
If I set to 100%, it takes me a long time to come to 35% or lower.

It would be good, if I could limit the range from i.e. 12..40%.

Thanks
Andreas

Jason2866, 2019-04-27:
You can implement this via rules. See wiki "Rules"
If you need help visit Tasmota Discord Chat

andreas-nn, 2019-04-27:
Thanks.
This is a right flexible way.

closed 2019-04-27 by andreas-nn

CJS0815, 2019-07-06:
Hi Andreas,

did you find a solution for your problem?
I've the same but not so much knowledge about rules and commands.

If you found something, please can you post it?