arendst/Tasmota #5715: OTA Update not working on sonoff basic

detxm80, 2019-04-26:
Hi, i've sonoff with tasmota 6.4.1 and i want to update to latest, but when i try to update via ota it reboots and fimware stay 6.4.1 and also manual update (basic minimal) i get this error:
Upload Failed

Upload buffer miscompare

can ypu help me?
thanks

Jason2866, 2019-04-27:
Have you done upgrade as in wiki described?
https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrading
If this doesnt work you have to upgrade via serial

detxm80, 2019-04-27:
yes, on the console i get this error: Non enough free space, why? i choose minimal, my language, .bin, nothing to do

Jason2866, 2019-04-27:
Sorry no, this error message appears in every case something is not normal in memory map.
You could try `reset 5` and flash via file upgrade minimal v.6.4.1 and try to upgrade to v.6.5 sonoff
If this doesnt work only option is to upgade via serial flash

ascillato2, 2019-04-27:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-27 by ascillato2