arendst/Tasmota #5717: MCP23017  THERE IS NO  state_topic: "stat/sonoff/POWER"

SciTii, 2019-04-27:
Hello

I'm Using tasmota with MCP230017 With all pint set to output and some relays. Eveything works great except One thing.
In Home Assistant I'm using This Config:

  - platform: mqtt
    name: "Sterownik1"
    command_topic: "cmnd/Sterownia/sensor29"
    state_topic: "stat/Sterownia/RESULT"
    qos: 1
    optimistic: false
    payload_on: "0,ON"
    payload_off: "0,OFF"

Work Great. BUT Tasmota Only Sends Info About "Results" Which contains very specyfic payload and it cannot works simply as Stat info which means that i cannot have any sync with HASS and Tasmota. When i connect relay to the esp8266 DIRECTLY  and then set it as relay is instantly send stat info after the relay state is changing. How can i Repair that?

The message after turning off the relay connected to MCP
stat/Sterownia/RESULT = {"S29cmnd_D0":{"COMMAND":"OFF","STATE":"OFF"}}

The message after turning off the relay connected to ESP and set as relay 
21:51:56 MQT: stat/Sterownia/RESULT = {"POWER":"OFF"}
21:51:56 MQT: stat/Sterownia/POWER = OFF

curzon01, 2019-04-27:
use SetOption4 1

SciTii, 2019-04-27:
> use SetOption4 1

Sadly, The only thing that changed is the name od the message but looks exactly the same
It is no longer Result. This is now Sensor. But there is this not just "OFF" or "ON payload like in normal Relays. The worst thing is that in hassio there is no possibility to set the template of the state topic inside config. NOW IT LOOKS LIKE THIS:

stat/Sterownia/SENSOR = {"S29cmnd_D0":{"COMMAND":"ON","STATE":"ON"}}

curzon01, 2019-04-27:
Use Rules, trigger on S29cmnd_D0#STATE and publish %value% to your topic you want

SciTii, 2019-04-27:
> trigger on S29cmnd_D0#STATE and publish %value%

I dont know to much about templating. I did something like this:
trigger S29cmnd_D0#ON do publish  tele/Sterownia/STATE ON endon

But is says:
stat/Sterownia/COMMAND = {"Command":"Unknown"}

blakadder, 2019-04-27:
This looks to me like a homeassistant templating problem that has nothing to do with tasmota.

SciTii, 2019-04-27:
But This is not a template inside home assistant. This is template inside Tasmota :] There is only A problem with tasmota that don't want to Publish state topic :]

blakadder, 2019-04-27:
this is the state topic `stat/Sterownia/SENSOR = {"S29cmnd_D0":{"COMMAND":"ON","STATE":"ON"}}` you just need to template HA to use it

SciTii, 2019-04-27:
Hmmm... Ok thanks. I have no idę nie to do it but I will try. Thanks A lot!

blakadder, 2019-04-27:
try value_template: "{{value_json['S29cmnd_D0'].STATE}}"

SciTii, 2019-04-27:
As anogher mqtt sensor or where? :p

blakadder, 2019-04-27:
buddy, in the config in the op... also change the payload message to ON and OFF

ascillato2, 2019-04-28:
>As anogher mqtt sensor or where?

You have two options. One is as a sensor in home assistant using templating and the other option is using a rule inside Tasmota to publish the mqtt topic you want for home assistant. You should choose just one method.

If you need further assistant with any of the solutions provided, please find us in the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-28 by ascillato2