arendst/Tasmota #5720: Vishay VEML7700 high accuracy ambient light sensor

subdancer, 2019-04-28:
hello, i just got a bunch of veml7700 high accuracy ambient light sensors.
i've been doing some tests on an arduino, they work great and are really accurate.
They talk i²c on adress 0x10.
I'd love to use them on one of my sonoffs. 
Here are datasheet and lib:

datasheet: https://www.vishay.com/docs/84286/veml7700.pdf
library: https://github.com/adafruit/Adafruit_VEML7700
 
Can we please get tasmota support for it? That would be awesome!

Jason2866, 2019-04-28:
If you write a driver for it ;-)  Since there a soooo many nice "things" out there...
Tasmota is a private non comercial project from 1 man, with help of some people.

subdancer, 2019-04-28:
i would if i could. the sensor is pretty new, so maybe it needs some more time to spread a little more. i'm hoping to find some ppl who are skilled in tasmotas codebase. the lib from adafruit works great on arduino but i haven't done anything on the esp8266 yet.

ascillato2, 2019-05-01:
Thanks for sharing your ideas.

We will wait for your PR. If you need help, please just ask here or in the chat. 

Adding the label requested feature (hold over) for future reference.

closed 2019-05-01 by ascillato2

subdancer, 2019-06-23:
I've got it working with basic functionality.

[image]

I printed a little enclosure for a wemos d1 mini, BME280, VEML7700 and 128x64 oled display
The sensor data is being fed into the diplay buffer and turned on via rule as soon as telemetry reports.

[image]

Things i still like to do:
Being able to set integration time and gain for the VEML7700 via webconsole.
Currently the integrationtime and gain are hardcoded to 25ms and 1/8 to achieve the 
maximum 120000 lux.
It would be nice if anyone can give me some pointers on howto do so.

Have a great day

joba-1, 2019-06-23:
cant help with the veml specifics, but I have implemented autogain in the
tsl2561 driver. Maybe you want to have a look?

Basically I have a table with integration time/gain combis, sorted by
sensitivity. As soon as values are close to the limits of one combi, I
switch to the next and repeat the measurement.

-- 
iphone so i typo

subdancer, 2019-06-23:
thx @joba-1 i will have a look when i get some time next week. thx for the pointers
this is what the datasheet says.:
[image]

i just tested direct sunlight and indoor at lowlight and i think the setting is sufficient as is. it still would be  a good idea to have autogain or switch function via webinterface if the sensor is used in only dark or bright environment to get a broader measure range.

iJaco, 2019-11-09:
hi  sundancer, on your post of 23 June you got the VEML7700 working, could you please explain your setup please?

subdancer, 2019-11-14:
it's all in my github.

ascillato2, 2020-04-23:
@subdancer Hi, can you help on this one? https://github.com/arendst/Tasmota/issues/8260

Thanks.

nsitt, 2020-04-24:
I already compiled / installed subdancer's code. However it breaks readings from another sensors I have attached (VELM6070, sun intensity). The sensors are on different adresses, I2cscan is showing the sensor but no values. sundancer's code is probably not clean. Troubleshooting code is still new to me, I am continuing and trying to implement a stable version and maybe one day I'll entere a PR.