arendst/Tasmota #5721: Switch - how to ignore one signal (in my case OFF)

witek1308, 2019-04-28:
Hi
I have the following setup: sonoff basic with Tasmota 6.5.0 and PIR (I don't know the model) connected as Switch1 to GPIO1 and controlling the LED light. Triggering works fine but I would like to be able to ignore OFF signal from the PIR. 
It seems PIR have some intelligence which I didn't expect from it and after sending ON signal when move detected, it sends OFF signal after 5s. I don't like this behaviour as light goes off too early. 
I tried to use pulseTime but it doesn't give expected result, actually some dodgy behaviour (or maybe it should go toghether with some other setting?) but this is kind of logical if PIR sends the OFF signal.
So expected behaviour would be:
- receive ON
- ignore any further OFF from switch
- send OFF from non-switch (MQTT or Toggle via HTTP)
Is it possible at all? I browsed the net but couldn't find similar issue, on the other hand it looks like a very common case so most likely my search query was not good enough

MadDoct, 2019-04-28:
Look at switchmode!... There's your solution! ;)

witek1308, 2019-04-28:
I have looked before asking but can't conclude which would this be. It looked to me like none will fit. Which one do you have in mind, 4?

MadDoct, 2019-04-28:
Exactly...

closed 2019-04-29 by ascillato2