arendst/Tasmota #5723: Stand-by LED

Nigcra, 2019-04-29:
```
00:32:15 RSL: STATUS = {"Status":{"Module":29,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,4,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:32:15 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:32:15","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":80,"SaveAddress":"F6000"}}
00:32:15 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0.8(sonoff)","BuildDateTime":"2019.04.15 17:01:53","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:32:15 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["censored","censored"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["8000E001","280500000100000000000000000000000000","00000000"]}}
00:32:15 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":510,"Free":492,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","0FCAC284","240783A0","23B617CE","00003BC0"]}}
00:32:15 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-4292","IPAddress":"192.168.1.209","Gateway":"192.168.1.10","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.10","Mac":"DC:4F:22:96:F0:C4","Webserver":2,"WifiConfig":4}}
00:32:15 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:32:15 1970","Local":"Thu Jan 01 00:32:15 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":99,"Sunrise":"07:43","Sunset":"16:03"}}
00:32:15 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:32:15"}}
00:32:15 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:32:15","Uptime":"0T00:32:15","Vcc":3.458,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"censored","BSSId":"censored","Channel":1,"RSSI":92,"LinkCount":2,"Downtime":"0T00:00:06"}}}
```

Hello guys,

I would like to completely deactivate the LEDs in standby for my Sonoff T1 2CH. Not only the WiFi LED, but also the Touch Button LEDs, which light up in stand-by mode.

Standby: https://www.dropbox.com/s/vj7i8sf0fxpzcz6/2019-04-29%2014.07.02.jpg?dl=1
Active: https://www.dropbox.com/s/278pxfbujl81ef5/2019-04-29%2014.07.12.jpg?dl=1

Of course I have already set the options
```
LedState0
LedPower0
```
 but this seems to have no effect on the standby light. Does anyone have an idea?

meingraham, 2019-04-29:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Nigcra, 2019-04-29:
Thank you very much, but I don't understand why this problem should be more suitable for chatting than the comparable problem from #1293 , which from my point of view is not yet really answered.

In addition, the chat is not searchable without login, a possible solution would disappear there. 

I was able to implement a full roller shutter control with the Sonoff Touch. It only fails because of the deactivation of the softly lit touch LEDs, which are too bright for a bedroom.

If there is no software-side or module-side solution and desoldering the LEDs is the only solution left, I can live with it, but then at least everyone who tries to get ahead with Google knows.

Translated with www.DeepL.com/Translator

meingraham, 2019-04-29:
Yes, the language barrier can be an issue. But I still believe that the forums I suggested are a better option as it relates to Tasmota. I think you will find the users on the forums to be very sensitive to this as many of them are not native English speakers themselves.

The issue that you reference is over one year old. If the issue is not resolved or the resolution provided in that topic does not work for you, it is likely far from applicable because the code have evolved tremendously since that date.

There is a possibility that what you are trying to do is a module or template configuration issue which is not a bug in the software. To explore this possibility, the chat mechanism is more interactive and can lead to a conclusion more quickly. If the chat dialog determines that your issue is best addressed as either a feature request or a bug report, then this GitHub issue can be revisited. However, first, it is best to troubleshoot the issue on Discord.

There is also a possibility that the function of the LEDs on your device (can vary by manufacturer and even hardware revision), is not possible to control with Tasmota. The LED could be hardwired (as you mention - requiring a physical modification to the device) to the output state. The LED could be controlled by an onboard MCU separate from the ESP SoC. Understanding these possibilities is something that a chat on Discord can more adequately address.

Yes, Discord is not searchable without login in. But it is searchable. Your point of having a thread which can be crawled by a search engine and made available for users in the future is valid. But the value of interactivity, we believe, outweighs this.

Regards.

Mike

closed 2019-04-29 by Nigcra