arendst/Tasmota #5727: SetDuration/RemainingDuration with Tasmota in ioBroker

stuermera, 2019-04-29:
Hello!

I am using two previous TUYA outlets, flashed with Tasmota 6.5.0 firmware.
The main reason was for not beeing in the cloud anymore.

The original TUYA plugin for ioBroker was able to let me set a Duration via "SetDuration" and let me read a Remaining time via "RemainingDuration". Together with Yahka (Homekit Plugin) I was able to set a duration for a specific power outlet to be switched off after a few minutes.

It seems the Tasmota 6.5.0 firmware is missing this feature - I only can find "Time" and "Uptime" values. But nothing happens when selecting these.

[image]

Any idea?
Thanks!

meingraham, 2019-04-29:
Duration of what?

stuermera, 2019-04-29:
A duration of how long the outlet is powered on.
I am using the outlet for my irrigation system in the garden. I set-up a "Valve" device in ioBroker with the power-switch set to "Active" and "InUse". So I can switch it on/off and have a nice colored irrigation-icon when powered on in Homekit on my iPhone :)

Setting a duration (how long the power is switched on) and reading the remaining duration isn't possible from what I've found out so far...

meingraham, 2019-04-29:
OK. Time is the time the device is set to. Uptime is how long the device has been powered.

Lots of ways to control output power durations in Tasmota. You can use [timers](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#timers). You can use [PulseTimer](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#pulsetimer).

In neither case can you query how much time is remaining in the duration though. You can reset a timer, cancel it, etc.

Please, close this issues and address any remaining questions to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota software feature requests and bug reporting. Troubleshooting is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

stuermera, 2019-04-29:
Ok thank you! :)

closed 2019-04-29 by stuermera