arendst/Tasmota #5728: NTP does not update

waldorffsmama, 2019-04-30:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
A clear and concise description of what the bug is: 

I have a couple of Sonoffs in the same network. None of them update the correct times from the NTP servers. They did initially after flashing but have stopped at some time. What I have tried so far:

1) changed NTP-servers, tested several, with URL and IP
2) changed configuration of DNS-servers
3) tested NTP-servers from my PC with a ping and would receive a reply
4) a) tried to update the Sonoffs with Tasmota 6.4.1 to 6.5.0: no error but it  remained the same version
b) tried to update uploading the sonoff.bin: File upload buffer miscompare

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

[x ] Searched the problem in issues and in the wiki
[x ] Hardware used : Sonoff
[x ] Development/Compiler/Upload tools used : ESPEasy, Tasmota 6.4.1 and 6.5.0
 Provide the output of commandstatus 0 :
00:08:07 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff Laser"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:08:07 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:06","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":19,"SaveCount":35,"SaveAddress":"F6000"}}
00:08:07 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sensors)","BuildDateTime":"2019-03-19T12:26:22","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:08:07 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["home",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
00:08:07 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":544,"Free":456,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDEE394","0007A3A4","B7FFBFCC","005ABBC0"]}}
00:08:07 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-Laser","IPAddress":"192.168.1.23","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:51:D6:6A","Webserver":2,"WifiConfig":4}}
00:08:07 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_51D66A","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:08:07 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:08:07 1970","Local":"Thu Jan 01 00:08:07 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:08:07 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:08:07"}}
00:08:07 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:08:07","Uptime":"0T00:08:06","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"home","BSSId":"E8:94:F6:E3:AB:46","Channel":9,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:03"}}}

[image]

Jason2866, 2019-04-30:
Duplicate. See #5283

waldorffsmama, 2019-04-30:
Thanks. Not really a duplicate. My issue was another one and I just managed to sort it out. I had the wrong gateway. For stranger reasons in my network, the gateway is on 192.168.1.2 instead of .1.

closed 2019-04-30 by waldorffsmama