arendst/Tasmota #5729: Fails to compile

Dazeddotdnc, 2019-04-30:
For the life of me, I can't get this to compile in Arduino IDE.
I found the libraries in the lib folder, copied them over per instructions.
I have no errors about missing libraries. 

The only warning/alert I get is:
>Multiple libraries were found for "ArduinoJson.h"
>Used: E:\Users\Dazed\Documents\Arduino\libraries\ArduinoJson-5.13.4
>Not used: E:\Users\Dazed\Documents\Arduino\libraries\Antares_ESP8266_MQTT
>Multiple libraries were found for "PubSubClient.h"
>Used: E:\Users\Dazed\Documents\Arduino\libraries\PubSubClient-EspEasy-2.6.09
>Not used: E:\Users\Dazed\Documents\Arduino\libraries\Antares_ESP8266_MQTT

I get a fair number of warnings that may stem from this?
Unused variables, some type mismatches.
Type mismatches in particular don't sound good. I probably have a conflict in the libraries.
I'm not sure how to find which one or what to do about it though.

It chugs along ok for a few moments anyway. And then it stops at:
>exit status 1
>Error compiling for board Generic ESP8266 Module.

I tried compiling  with board = NodeMCU 0.9 and NodeMCU 1.0 (my actual target is one of those two. Not sure what's different between them)
>exit status 1
>Error compiling for board ....

Since NewPing-1.9.1 is the last library reportedly used, I thought maybe this was the source of trouble. I found the sensor that was causing it to be included. I tried compiling it with that sensor excluded. All that changed is Adafruit_SGP30-1.0.0.13 became the last used library. It might be stumbling on an error in the library that comes after NewPing-1.9.1?
But, I don't know what that would be. The error is very non-helpful in pointing a direction.

I thought I modified something that broke it. So I tried compiling the stock source code.
>exit status 1
>Error compiling for board ....

Someone mentioned that they used libraries from the development branch and it fixed their issue. Lacking other things to try, I did the same.
>exit status 1
>Error compiling for board ....

I don't have a lot of clues to chase here.
Nothing I found on my own worked.
I'm out of ideas and this error message isn't proving to be very google-able.
Time to reach out for help.

Jason2866, 2019-04-30:
You have it installed over a normal Arduino Installation, this makes a working setup complicated.
Do a extra portable install as described in wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

Dazeddotdnc, 2019-04-30:
That was the missing step.

I remember seeing this trick mentioned somewhere along the way now. Guess I took it to be some oddity specific to his particular needs. It didn't register in my mind that adding the library in and of itself wasn't enough.

Thanks for the help.

closed 2019-04-30 by Dazeddotdnc