arendst/Tasmota #5730: Publish Domoticz IDX on telemetry

HobbytronicsPK, 2019-04-30:
Hi.,

Can we publish Domoticz IDX, in telemetry MQTT topic when connected to MQTT?

`Rule1 on Mqtt#Connected do Publish "Default Telemetry Topic" "Domoticz IDX"`

Is this possible?

Jason2866, 2019-05-01:
Yes, see rules example. You can adapt for your needs
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#5-use-of-variables-and-tele-

ascillato2, 2019-05-01:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-01 by ascillato2

HobbytronicsPK, 2019-05-02:
@Jason2866 you did not get me.

> Rule1 on Mqtt#Connected do Publish "Default Telemetry Topic" "Domoticz IDX"

I want to publish **Domoticz IDX number** assigned to a specific switch on default tele topic such as `tele/A01EEA/STATE`

Jason2866, 2019-05-02:
No, not possible without using rules.

HobbytronicsPK, 2019-05-05:
@Jason2866 Can we get the default **mqtt telemetry topic** and preset **Domoticz IDX number** in two variables such as var1 and var2? No matter with the use of rules?

Jason2866, 2019-05-05:
No, same reason as in #5522