arendst/Tasmota #5732: Easy method to add simple GPIO (pin high/low) 'sensors'

bikemike, 2019-05-01:
**Is your feature request related to a problem? Please describe.**
I have 5 switches set up to sense open/closed circuit. I do not want these switches to control the relays but there is no easy way (short of adding rules) to not control the relays. I have switchtopic set to 2. The relays are set to relay1, relay2, and relay3 and switches are numbered switch4, switch5, switch6, switch7, switch8. When wifi or the mqtt server go down, they start controlling relay1! This was very frustrating to debug and also very poorly documented.

**Describe the solution you'd like**
It would be great if there was an easy way to set up a gpio pin as a simple sensor (pin high/low)  through the "configure module" interface and not have it control a relay.

Another option would be to only allow a switch to control the relay of the same number (which is they way I thought it would work). It is very confusing when switch4 starts controlling relay1. 

Yet another option could be to add another switchtopic mode that doesn't fall back to switchtopic 0 when there is no mqtt connection. This is not ideal because switchtopic is already confusing enough and requires entering a console command.

**Describe alternatives you've considered**
After finally stumbling on several closed "not-fixed" bug reports I found that the only current solution is to define rules. 

Here are some of those related bugs:
https://github.com/arendst/Sonoff-Tasmota/issues/4299 
https://github.com/arendst/Sonoff-Tasmota/issues/4112 
https://github.com/arendst/Sonoff-Tasmota/issues/3520 
https://github.com/arendst/Sonoff-Tasmota/issues/1305 
https://github.com/arendst/Sonoff-Tasmota/issues/3264

blakadder, 2019-05-01:
The behaviour when mqtt goes down is documented in https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches.

Majority of Tasmota applications are for consumer devices with a set number of relays and buttons. 

Everything you described as a solution already exists in rules.
Rules are created for everyone going the DIY route. If you manage to connect 5 switches to an esp8266 module you should manage to read the rules article and program a few rules to manage all the switches and relays.

Jason2866, 2019-05-01:
@bikemike  As @blakadder  explained the behaviour from Tasmota is a design decision.
This behaviour will not be changed because it is NOT a bug and keeping backwards compatibility.
Feel free to fork and change it the way you want. If you find a solution which is small in code size and keeps backward compatibility a PR is highly welcome. Thanks for sharing your ideas.

digiblur, 2019-05-01:
I do agree that switchtopic can be confusing for some but now with the power of rules all of this can easily be done and still works if your wifi goes down.

bikemike, 2019-05-01:
I understand that it is a design decision. Perhaps the main issue then is the documentation. The fallback behavior is not stated clearly and prominently enough. It is at the very bottom of the page far from the initial definition of what switchtopic 2 does. I think the behavior should contained in a big yellow warning box right at the top of the switchtopic 2 description.

In addition to that, it is not stated anywhere (that I could find) that a switch other than switch1 will control relay1. Please point me to the documentation if it exists because I found it really odd that my switch4 started controlling relay1 when there was no mqtt connection.

Not many people test with and without an mqtt connection so it can be quite a surprise months/years later when wifi goes down and strange things start happening to garage doors (or whatever else is being controlled).

blakadder, 2019-05-01:
I put some warning signs as requested.

closed 2019-05-01 by ascillato2