arendst/Tasmota #5733: Support for more than one switch per Relay for Sonoff 4TCH

jromero2000, 2019-05-01:
I have 4 lights in a room controlled with 2 normal switches plus 2 double switches. In total I am using 6 GPIO inputs configured switch1-6 and connected to each physical switch. With current implementation it looks like switch 5-6 are associated to relay 1. So I modified the code so switches 5 and 6 works as second switch relay 1 and 2 respectively.

This the change I made in Sonoff file:
if (switchflag < 3) {
          //if (!SendKey(1, i +1, switchflag)) {  // Execute command via MQTT
          //  ExecuteCommandPower(i +1, switchflag, SRC_SWITCH);  // Execute command internally (if i < devices_present)
          if (!SendKey(1, i%(MAX_SWITCHES/2) +1, switchflag)) {  // Modification to control 1 relay with two inputs
            ExecuteCommandPower(i%(MAX_SWITCHES/2) +1, switchflag, SRC_SWITCH);  // Modification to control 1 relay with two inputs
          }
        }

I think this feature is quite interesting and also applicable to sonoff dual.

blakadder, 2019-05-01:
Everything described can be done by using rules

jziolkowski, 2019-05-01:
Why hardcode it at two? What if I want 3 switches per relay? I don't really see the point, and like @blakadder said, easily done using rules.

Jason2866, 2019-05-01:
Will not be changed because it is a special use case and this case can be realized via rules.
Please cose this feature request because your question is answered.
Thx!

closed 2019-05-01 by ascillato2