arendst/Tasmota #5735: Tasmota unexpectly restarts after certain switch plus button configuration

Key-Squaer, 2019-05-01:
### BUG DESCRIPTION
I have set up a Wemos D1 mini with Tasmota 6.5.0. It is configured having 1 switch and 1 button - see picture - connected to Wi-Fi.
TASMOTA resets & restart without any additional interaction after approx. 1 minute starting with a "fresh" configuration.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
Device used (i.e. Sonoff Basic) : Wemos D1 Mini
- [x] Tasmota binary firmware version number used : 6.5.0 / pre-compiled
- [x] Development IDE - Compiler / Upload tools used : n.a./OTA
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
Comment:
Perhaps not too helpful, as TASMOTA resets also the config data but nevertheless here it is :)

00:00:25 CMD: status 0
00:00:25 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:25 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:23","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":10,"SaveAddress":"FA000"}}
00:00:25 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019.03.19 12:30:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:00:25 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Overlord",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
00:00:25 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":508,"Free":492,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","0FDAE394","000783A0","23B617CE","00003BC0"]}}
00:00:25 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7002","IPAddress":"192.168.178.129","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"CC:50:E3:C7:5B:5A","Webserver":2,"WifiConfig":4}}
00:00:25 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C75B5A","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:25 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:25 1970","Local":"Thu Jan 01 00:00:25 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:00:25 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:25"}}
00:00:25 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:25","Uptime":"0T00:00:23","Vcc":2.737,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":96,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Overlord","BSSId":"34:31:C4:45:DA:75","Channel":13,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```

Complete serial log while setting up Wemos D1 mini with tasmota and reproduction of error see "### TO REPRODUCE"

CONSOLE OUTPUT HERE:
00:02:00 RSL: tele/sonoff/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:00"}
00:02:27 WIF: Hostname %s-%04d, SSId1 Overlord, SSId2 
00:02:29 APP: starte neu

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Projekt sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: verbinden mit AP1 Overlord in Modus 11N wie sonoff-7002...
00:00:06 WIF: verbunden
00:00:06 HTP: Web-Server aktiv bei sonoff-7002 mit IP-Adresse 192.168.178.129
00:00:13 CMD: weblog 4
00:00:13 RSL: stat/sonoff/RESULT = {"WebLog":4}
00:01:13 MOD: Generic Module
00:01:16 APP: starte neu

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Projekt sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: verbinden mit AP1 Overlord in Modus 11N wie sonoff-7002...
00:00:04 WIF: verbunden
00:00:04 HTP: Web-Server aktiv bei sonoff-7002 mit IP-Adresse 192.168.178.129
00:01:38 MOD: Generic Module, GPIO0 0, GPIO1 0, GPIO2 0, GPIO3 0, GPIO4 0, GPIO5 0, GPIO12 0, GPIO13 17, GPIO14 9, GPIO15 0, GPIO16 0
00:01:40 APP: starte neu

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Projekt sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: verbinden mit AP1 Overlord in Modus 11N wie sonoff-7002...
00:00:07 WIF: verbunden
00:00:07 HTP: Web-Server aktiv bei sonoff-7002 mit IP-Adresse 192.168.178.129
00:00:51 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:00:52 CFG: Standard verwenden
00:00:53 APP: starte neu

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Projekt sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: WifiManager aktiv für 3 Minuten
00:00:00 HTP: Web-Server aktiv bei sonoff-7002 mit IP-Adresse 192.168.4.1

### TO REPRODUCE
_Steps to reproduce the behavior:_

Connect to WEMOS D1 mini that is starting with a resetted TASMOTA, waiting to be set up and configure to connect to local Wi-Fi (via 192.168.4.1) - 
--> Save & reboot
--> Connect via Browser to website (129.168.178.129)
--> Console: Weblog 4
--> Main menu --> Set up --> Configure device --> Generic (18)
--> Save & reboot
--> Main menu --> Set up --> Configure device --> D7: Button1 (17) & d5 Switch1 (9)
--> Save & reboot
Wait approx 1min after restart
Unexpected restart & reset!

### EXPECTED BEHAVIOUR
Wrong behavior observed: Unexpected Restart&Reset after @00:00:51 (see above)
Expected behavior: Run stable & do "nothing"

### SCREENSHOTS
Device Setup:
![pic1](https://user-images.githubusercontent.com/50168664/57019298-5e68e300-6c26-11e9-9c15-76ac895003ac.JPG)

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-05-01:
Please update to the latest development branch firmware (6.5.0.9) to see if the problem persists. You can set your OTAURL to http://thehackbox.org/tasmota/020300/sonoff.bin and perform an OTA update.

Key-Squaer, 2019-05-01:
Hi Meingraham!
Unfortunately not working: Tasmota is showing that it is restarting but not update actually happened. Serial log states "00:00:53 MQT: stat/SporeCrawlerTest/UPGRADE = Failed Wrong HTTP code".

meingraham, 2019-05-01:
Language variant should not make a difference. Did you compile yourself or use a pre-compiled version? I'm not sure if pre-compiled development binaries are available for language variants (I just haven't paid attention). If you self-compiled, try the pre-compiled sonoff.bin (English) just for troubleshooting.

Did you originally flash your device using serial upload or OTA?
If serial, did you erase the flash before flashing Tasmota? If not, do this and flash the device again.

I recommend that you continue this troubleshooting process via the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota software feature requests and bug reporting. Troubleshooting is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Key-Squaer, 2019-05-01:
Thanks for the help so far.

Pre-compiled, there are many languages available :)
Original flashing (version 6.4.2?) was done by serial upload on an erased Wemos D1 mini. After that updates via OTA. 

I still think that this is a bug. Tasmota was working nicely on this device with a more complex configuration (1 relay, 3 switches attached, BME280, etc. Wi-Fi and MQTT set up) without any unexpected resets&restarts. Still I wanted to add one additional feature and for that I thought it was a good idea to change one switch to a button (so I can use switchtopic 1 and buttontopic 0 for different mqtt behaviour). That led to the first unexpected reset&restart. I tried to isolate that issue by removing 2 switches and BME289 just to make it simple and still this behavior shows up. I do fear that users would run in the issue. So I decided to report the bug.

Thanks!

meingraham, 2019-05-01:
OK. Ping me on Discord and we'll try to get you past this. The first thing I'd recommend is performing a serial flash to the latest development version (currently 6.5.0.9). You can download that from thehackbox. When you do flash it, remember to erase the flash first for good measure.

Key-Squaer, 2019-05-01:
I will try that. I am new to disord. Iam in the #issues channel of tasmato and a lot of users are there. How do I find you? What's your username?

closed 2019-05-01 by Key-Squaer

felixstorm, 2019-05-18:
Same problem here (pre-compiled sonoff.bin, no-name D1 mini as "generic" with 1 button and 2 relays connected), dev firmware from OTAURL posted above did not work at all, so I switched back to 6.4.1, which seems to work (but it has been running only for a few minutes yet)...

Serial log
```
00:00:00 Project sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 D27i in mode 11N as sonoff-7881...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-7881 with IP address 192.168.0.26
08:46:04 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
08:46:04 RSL: stat/sonoff/POWER1 = ON
08:46:05 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
08:46:05 RSL: stat/sonoff/POWER1 = OFF
08:46:06 RSL: stat/sonoff/RESULT = {"POWER2":"ON"}
08:46:06 RSL: stat/sonoff/POWER2 = ON
08:46:07 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
08:46:07 RSL: stat/sonoff/POWER2 = OFF
08:46:29 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
08:46:29 CFG: Use defaults
08:46:30 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
```

felixstorm, 2019-05-18:
Same problem again with 6.4.1 :-(

```
00:00:00 Project sonoff Sonoff Version 6.4.1(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 xxx in mode 11N as bewaesserung...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on bewaesserung.local with IP address 192.168.0.26
09:14:01 RSL: INFO1 = {"Module":"Generic","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_241EC9_fb/","GroupTopic":"sonoffs"}
09:14:01 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"bewaesserung","IPAddress":"192.168.0.26"}
09:14:01 RSL: INFO3 = {"RestartReason":"Software/System restart"}
09:14:01 RSL: RESULT = {"POWER1":"OFF"}
09:14:01 RSL: POWER1 = OFF
09:14:01 RSL: RESULT = {"POWER2":"OFF"}
09:14:01 RSL: POWER2 = OFF
09:14:09 RSL: STATE = {"Time":"2019-05-18T09:14:09","Uptime":"0T00:00:13","Vcc":2.912,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":6,"RSSI":100}}
09:14:09 RSL: RESULT = {"POWER1":"ON"}
09:14:09 RSL: POWER1 = ON
09:14:11 RSL: RESULT = {"POWER1":"OFF"}
09:14:11 RSL: POWER1 = OFF
09:14:12 RSL: RESULT = {"POWER2":"ON"}
09:14:12 RSL: POWER2 = ON
09:14:13 RSL: RESULT = {"POWER2":"OFF"}
09:14:13 RSL: POWER2 = OFF
...
09:14:23 RSL: RESULT = {"POWER1":"ON"}
09:14:23 RSL: POWER1 = ON
09:14:24 RSL: RESULT = {"POWER1":"OFF"}
09:14:24 RSL: POWER1 = OFF
09:14:37 RSL: RESULT = {"Reset":"Reset and Restarting"}
09:14:38 CFG: Use defaults
09:14:39 APP: Restarting
```

felixstorm, 2019-05-18:
Ok, found the issue after digging deeper and enabling more serial debug output: Obviously the one pin I was using for a button is somehow permanently being pulled low on my no-name D1 mini. This causes the reset after 40 seconds as if the button had been continuously pushed...