arendst/Tasmota #5737: Support for Wi-Fi Dimmer?

jsponz, 2019-05-02:
Is this Wi-Fi Dimmer supported?

https://www.aliexpress.com/item/Wifi-Smart-Dimmer-220V-240V-150W-Controller-Timer-Switch-light-Voice-Control-Works-With-Amazon-Alexa/32962631174.html?spm=a2g0s.9042311.0.0.27424c4dZ69TdG

I have used tuya-convert to flash Tasmota, and it works. I have it with Tasmota. 

I can't find any Info about this device. I have tried Tuya Dimmer but no success (no switching, no dimming control from the Tasmota interface). 

Thanks!

Jason2866, 2019-05-02:
No. Please [read](https://github.com/arendst/Sonoff-Tasmota#note) and this [CONTRIBUTING](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md)

ascillato2, 2019-05-02:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-02 by ascillato2

dr-apple, 2019-05-08:
Hello i have have the same device, i flash tasmota 6.5, but what is the pin occupancy for the serial connection and switch ? Have any a idea ?

keesschollaart81, 2019-05-08:
I also recently bought this device and I'm very intersted in Tasmota support. It's the cheapest wifi AC dimmer out there (~15$ price point), very interesting!

keesschollaart81, 2019-05-08:
Impressive work from @digiblur 💯, super cool to see these videos.

Some pictures from inside of the device:
https://ibb.co/pPhL3cL
https://ibb.co/7Ysv6gR
https://ibb.co/3pmzzVh
https://ibb.co/s2Gg5S3
https://ibb.co/jLbc7Yg

[Tuya Convert log](https://pastebin.pl/view/fa67dcd1)

[Original firmware binary](https://upload.cat/4f056b7c84354700)

After flashing Tasmota is was able to configure it via the web interface and I tried some different combinations but without success. Module set to 'Tuya Dimmer (54)'. After trying some different RX/TX configuration and some random guesses for the SetOption34 I was not able to see any RX messages in the console (with web log level 4). Also no incoming data when manually switching the device. 

What would be a next thing to try/work out? I don't think I'm able to solder/connect to the serial port myself (too small & no experience).

keesschollaart81, 2019-05-08:
Super cool to see the videos from @digiblur, I learned a lot.

Some pictures of the internals of the device:
https://ibb.co/pPhL3cL
https://ibb.co/7Ysv6gR
https://ibb.co/3pmzzVh
https://ibb.co/s2Gg5S3
https://ibb.co/jLbc7Yg

[The log of the Tuya Convert](https://pastebin.pl/view/fa67dcd1)

[The original firmware](https://upload.cat/4f056b7c84354700)

After flashing I was able to go to the Sonoff web page and try different configurations. I was not yet able to get it work work. I tried the 'Tuya Dimmer (54)' module with some different RT/TX configs. Also some random guesses with the SetOption34. When setting the web loglevel to 4, I did not see any RX messages coming in. Also when manually switching the device there was nothing in the console.

What would be a next step? I don't think I'm able to solder/connect to the serial port myself (very small/no experience). Any ideas?

keesschollaart81, 2019-05-08:
Super cool to see the videos from @digiblur, I learned a lot.

Some pictures of the internals of the device:
https://ibb.co/pPhL3cL
https://ibb.co/7Ysv6gR
https://ibb.co/3pmzzVh
https://ibb.co/s2Gg5S3
https://ibb.co/jLbc7Yg 

After flashing I was able to go to the Sonoff web page and try different configurations. I was not yet able to get it work work. I tried the 'Tuya Dimmer (54)' module with some different RT/TX configs. Also some random guesses with the SetOption34. When setting the web loglevel to 4, I did not see any RX messages coming in. Also when manually switching the device there was nothing in the console.

What would be a next step? I don't think I'm able to solder/connect to the serial port myself (very small/no experience). Any ideas?

keesschollaart81, 2019-05-08:
Super cool to see the videos from @digiblur, I learned a lot.

Some pictures of the internals of the device: 

After flashing I was able to go to the Sonoff web page and try different configurations. I was not yet able to get it work work. I tried the 'Tuya Dimmer (54)' module with some different RT/TX configs. Also some random guesses with the SetOption34. When setting the web loglevel to 4, I did not see any RX messages coming in. Also when manually switching the device there was nothing in the console.

What would be a next step? I don't think I'm able to solder/connect to the serial port myself (very small/no experience). Any ideas?

jsponz, 2019-05-09:
@keesschollaart81 Thanks for your analysis. I spent some time as well. One of my devices is broken: it does not boot. 

The other one is flashed with Tasmota, but nothing happens. Same result as you have.

dr-apple, 2019-05-09:
if I could find out the pin assignment, then I would try to read the protocol and recreate .... I have to lie on the couch for 2 weeks, have time ;-)

keesschollaart81, 2019-05-09:
Maybe @jsponz can disassemble his broken device and expose the chip so that we can see which Tuya device it compares with / which pins we have?

digiblur, 2019-05-09:
Which pins did you try for TX/RX?

jsponz, 2019-05-09:
@keesschollaart81 Of course!

It is already disassembled :-) 

Tomorrow I will send you pictures.

keesschollaart81, 2019-05-09:
@digiblur same as described [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer)

I also tried to point the Tuya Tx107 and Rx108 to the GPIO1 and 3. Also swapped 107 and 108 in both configs. Every test I tried to toggle the input switch but the logging is [always the same](http://prntscr.com/nmjab3)

jsponz, 2019-05-10:
@keesschollaart81 the same happens here with the original firmware. I have reported it and they have told me that it is a  production issue. A new device is on the way with this fix. I will confirm as soon as I have it...

However, after flashing one unit with Tasmota (as it is right now without support for this device), I see that the light is always off, so  I guess it is firmware related.

keesschollaart81, 2019-05-10:
After I set TuyaRx18 to GPIO16 it does not boot any more... :-/

dr-apple, 2019-05-10:
You must reset the esp with 3 x ac power on / off with 2 sec sleep

jsponz, 2019-05-10:
@dr-apple, @keesschollaart81 the same happened here and esp did not boot up anymore. 

I will try later, when taking pictures.

Jason2866, 2019-05-11:
Not all GPIOs can be used for all. See https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#restrictions

keesschollaart81, 2019-05-14:
@jsponz any update?

jsponz, 2019-05-14:
@keesschollaart81 Sorry, I have a personal issue. Tomorrow I will take the pictures.

jsponz, 2019-05-15:
As promises, please find attached the pictures. 

[image]
[image]
[image]
[image]
[image]
[image]

Jason2866, 2019-05-15:
It seems there is a sort of ESP module hooked on a PCB with a MCU.
The type of the MCU cant be identified on the photos. Next step is to identify type of MCU
and someone hast to hook up a protocol analyzer and sniff commands that needs to be sended
and are received from the ESP to activate on off and dimming. If this is done and providing this detailed information someone can implement this type of dimmer in Tasmota.
Tbh i dont believe this will happen because there are already supported easy to buy dimmers available.

CJS0815, 2019-05-17:
@Jason2866 
Please, can you say me what Types/Models of dimmers for in-wall? (only box without switch or push-button) is supported by tasmota? 
sorry for my english :-(

dr-apple, 2019-05-17:
The Model is :  QS-WIFI-D01-TRAIC

CJS0815, 2019-05-17:
You mean TRIAC?

Not available in Europe :-(
Not on aliexpress

Do you have a dealer for me?

dr-apple, 2019-05-17:
? show the first post here, it from ali and came frome china ?

CJS0815, 2019-05-17:
??? I dont understand !?! 

The part from the first post was actual not supported - or did I read complete wrong?
I asked for a supported module like this form the first post ...

Jason2866, 2019-05-17:
There is none.

dr-apple, 2019-05-17:
yes but you have asked : 
Not available in Europe :-(
Not on aliexpress

Do you have a dealer for me?

and i have mean, you mean this this device ;-)

CJS0815, 2019-05-17:
No , sorry (I tell you, my english is ....)

What type or models do you mean with "because there are already supported easy to buy dimmers available"?

dr-apple, 2019-05-17:
@CJS0815 from you: sorry (I tell you, my english is ....)

my too ;-) sorry

CJS0815, 2019-05-17:
Is this Wi-Fi Dimmer supported?
[image]
[image]
[image]
[image]

The product was [dealed with documentation in germany ](https://blog.ex-store.de/content/wifidimmer-v4)

Is thera a chance to get it together with Tasmota?

dr-apple, 2019-05-17:
Here the new version V5:
https://ex-store.de/2-Kanal-WiFi-WLan-Relay-V5-Blackline-fuer-Unterputzmontage

Is this tasmota ready ?

jsponz, 2019-05-22:
@keesschollaart81 Any update? :)

CJS0815, 2019-05-22:
@dr-apple 

Sorry, its not a new version, that is another version - the V5 is only a Switch-Relay
I speak from another product called V4 Dimmer-Modul
The pictures I've send above are from this V4 dimmer-Modul

keesschollaart81, 2019-05-23:
To be clear @CJS0815 I believe @jsponz and I are talking about the device linked in the opening post with pictures here https://github.com/arendst/Sonoff-Tasmota/issues/5737#issuecomment-490588738 and here https://github.com/arendst/Sonoff-Tasmota/issues/5737#issuecomment-492511979. This: https://github.com/arendst/Sonoff-Tasmota/issues/5737#issuecomment-493426380 is a completely (non dimmer) different device, let's not mess up this thread :-) which is about a very cheap Tuya based 220V AC dimmer.

As said, my first one got bricked. I now have a new one but I have not yet flashed it and I'm not sure I will do that, as long as there is no additional progress. 

Looking at the pictures of @jsponz the ESP chip looks quite different from the one @digiblur got to work in his video. I guess unless someone with more experience/tools spends time on it, we have to use it as it comes with Tuya's firmware...

jsponz, 2019-05-23:
@keesschollaart81 thanks for your clear summary. 

I had one issue with the first two units. When switching it off, there was a minimum level of light. That happened with the original firmware and hardware. They have kindly sent me two new units and they work as expected. When off, it is off :-) 

It is worth to make it work with Tasmota. It is cheap, robust and good quality. It works with LED lamps, incandescents, etc. It shouldn't be so complex, considering that the main effort is made for the other modules. 

Sorry but I can't do it. I don't have my debugging tools in my current location...

MortenVinding, 2019-05-26:
As you might have seen in #5859 I'm also trying to make this dimmer work with Tasmota.
Did try the Tuya convert road but did not manage to make it work. But glad you did, that probably means there is hope.
I will try the convert later then on a fresh installed Ubuntu or something.

Anyway I took one for the team and took mine to pieces, also removing the RF shield.
Did some measurements to find the pinout of this special module:
[image]

The MCU on the back is a STM8S003F3:
[https://www.st.com/resource/en/datasheet/stm8s003f3.pdf](url)
And TX and RX is connected to the ESP's TX and RX pins.

The same MCU seems to be used in dimmer they are talking about here:
[https://github.com/arendst/Sonoff-Tasmota/issues/469](url)
But it's hard to tell if the firmware is the same.

Hope this is of help for some...

dr-apple, 2019-05-26:
Thank you for your class support! Would not it be possible to connect to TX / RX in parallel and to 
record the commands sent by the ESP with original firmware to MCP?

I would test it myself, but unfortunately I already have tasmota on it :-(

Sorry for my very bad English :-(

MortenVinding, 2019-05-26:
Sorry just realized that my link to #469 did not work. It’s fixed now.
Have you tried to configure your like in that issue?

I can try to analyze the traffic on the RX and TX pins, if my module survived the dismantling yesterday (did require quite a bit of heat!)
And my logic analyzer and  me is still getting to know each other, so might not succeed.

dr-apple, 2019-05-26:
I once connected a USB 2 serial adapter, as described in the picture. But unfortunately nothing comes :-(
Have Tasmota still on it, there should be so by 3.3V, GND, U0RXD and U0TXD serial rauskommt 115200?

Have already turned U0TXD and U0RXD, unfortunately, no success :-(

The ESP goes but and I get on it by WLAN ;-)

MortenVinding, 2019-05-26:
> Have Tasmota still on it, there should be so by 3.3V, GND, U0RXD and U0TXD serial output at 115200?

From the other issue it seems it's using 9600 baud.
[https://github.com/arendst/Sonoff-Tasmota/issues/469#issuecomment-397851068](url)

Have you tried the firmware mentioned here:
[https://github.com/arendst/Sonoff-Tasmota/issues/469#issuecomment-398424806](url)
?

Unfortunately it seems that it is still not ready for daily use, but it should provide basic functionality.

MortenVinding, 2019-05-26:
Just realised it's basically the same ESP module used here:
https://groups.google.com/d/msg/sonoffusers/v0hEUKN5hpQ/xijNqbEhBAAJ[Link](url)

thxthx0, 2019-05-28:
or this:
[image]

DeanoXX, 2019-06-04:
I have a couple of the same dimmer. I had an original from about feb - that flashed OK with Tuya-convert and I have had Tasmota and ESPHome on - havent yet got it all working..

I have 2 more recently arrived. They wont flash with Tuya-convert. I have one running the orig firmware and can see the serial commands being sent on the ESP TX pin.

When it boots it has some text including...
"fireware info name:esp_QL_TD ,version:1.0.1 ,PRODECT_KEY:wnqaw70uBCsDqgZT"

When connected to the Smartlife app I can see the commands from the ESP to MCU. they dont appear to be std Tuya. At no point can I see anything come from the MCU in return....

These are typical bytes sent - each line is for a button press/slider move in the app
FF 55 00 05 DC 0A                                                       
FF 55 71 05 DC 0A
FF 55 FD 05 DC 0A
FF 55 FC 05 DC 0A
FF 55 FC 05 DC 0A 
FF 55 FE 05 DC 0A
FF 55 FF 05 DC 0A
FF 55 FF 05 DC 0A

The third byte is the dim level (00 = Off, FF = Full on, anywhere in between according to the dimmer slider)

But I cannot get the latest versions into a mode it will flash. Using the pinouts above and keeping GPIO 0 to ground doesnt seem to do anything different (it does work with the 1st Tuya converted module)

MortenVinding, 2019-06-04:
> or this:

image

Wow great find thxthx0!
Wonder if this dimmer is actually made by "Lingan"?
Seems they are making lots of wifi devices besides there ESP modules...

MortenVinding, 2019-06-04:
> 

I have a couple of the same dimmer. I had an original from about feb - that flashed OK with Tuya-convert and I have had Tasmota and ESPHome on - havent yet got it all working..I have 2 more recently arrived. They wont flash with Tuya-convert. I have one running the orig firmware and can see the serial commands being sent on the ESP TX pin.When it boots it has some text including..."fireware info name:esp_QL_TD ,version:1.0.1 ,PRODECT_KEY:wnqaw70uBCsDqgZT"
--

Nice info DeanoXX!
That might explain why I failed with Tuya-convert on my dimmer.

I did however also fail to get any sensible info from serial logging also. That's very intereresting, how did you see that?
What baud rate does it use? (and bits, stop bits etc).
Was the module connected to the dimmer when you logged it? (mine wasn't).

Still hase wires connected to my module, so will try to dump the firmware from that.

Whis Shelly would make a dimmer so we didn't have to mess with this...

jsponz, 2019-06-04:
@DeanoXX, Great news!! 

The same happened to me. On the first two units, I could use tuya-convert. On the second two units (the first ones were faulty and I could not switch off the light), I could not. In the next days I will take a look on this. 

With your info, code could be updated... 👍

DeanoXX, 2019-06-04:
Output is from the u0txd pin . 9600,n,8,1
Definitely works with the esp/mcu separated from the dimmer element. Pretty sure I have had it with them connected.

Pretty sure the switch input is GPIO13 but need to test more. I have one untouched. May see if that will flash via serial out of the box. If these would flash easily they would be a very good match for what I want.

Unfortunately I’m away for the next couple of days..so will be a while before I can get further.

thxthx0, 2019-06-04:
Hi, 👍

the dimmer is also listed @ amzon:
'TOOGOO Wifi Smart Dimmer Module 220V-240V 150W'

For flashing/reading ESP8266 maybe also set pin marked 'RE' (Reset?) to GND -
to disable ST MCU while flashing (same as for tuya-touch-dimmer).

Jason2866, 2019-06-05:
> These are typical bytes sent - each line is for a button press/slider move in the app
> FF 55 00 05 DC 0A
> FF 55 71 05 DC 0A
> FF 55 FD 05 DC 0A
> FF 55 FC 05 DC 0A
> FF 55 FC 05 DC 0A
> FF 55 FE 05 DC 0A
> FF [...]

With (s)serialsend you can test the commands like done for this [device](https://blakadder.github.io/templates/oil_diffuser_550ml.html)

HA-TB303, 2019-06-08:
I have exactly the same dimmer and I was able successfully to flash tasmota 6.5 to it using tuya-convert.
Unfortunately I did not know at the time this is a "special"  dimmer module using a seperate dimming chip which does seem to be supported in tasmota, but requires different tx and rx gpio's.

Before I fount that out, I flashed esphome. played around with that, but found there is no support for this special tuya dimming chip in esphome and tried to go back to tasmota. Flashed the tasmota minimal and then soft-bricked it somehow.

I now have to reflash by wire, but I'm not 100% sure how to proceed. Anyone having any luck flashing tasmota by wire and willing to share the wiring?

DeanoXX, 2019-06-08:
I have flashed 1 via Tuya (and via Serial) using the pins from the Lingan data sheet above.

From the edges of the Chip...
GPIO0 to GND to put it into flash mode
UTXD = TX
URXD = RX

Used the GND + VCC from the board itself as they're easier.

I desolder from the dimmer board first...

[image]

Cannot get my later 2 to flash at all. Via Serial or Tuya-Convert. very frustrating.

HA-TB303, 2019-06-08:
Tried, but mine won't flash either :(

thxthx0, 2019-06-08:
Did you try to set marked pin 'RE' (Reset?) to GND too, as mentioned above?

DeanoXX, 2019-06-08:
I have tried that yes with no success.

thxthx0, 2019-06-09:
Thanks for trying it out.
So we'll have to find other connections between ESP8266 and STM8S003F3 MCU - 
maybe cut traces.

Slofware, 2019-06-11:
So problems are not flashing the ESP, so you could use Arduino software. But the problem is how and what you send to the chip which controls the dimming?

james-fry, 2019-06-11:
Just ordered one from CN. Happy to be involved in testing when it arrives in a couple of weeks.

[235 more replies]