arendst/Tasmota #5740: Support TX20 Wind Sensor in Domoticz settings

javier6d2, 2019-05-03:
I would like to operate the following wind sensor on a Tasmota device connected to Raspberry PI and Domoticz:

https://www.amazon.de/Technoline-TX-20/dp/B01HXZ3KLA

The sensor has support on Tasmota but I would like to have the possibility to manage the Domoticz parameters for it.

At this moment I think is possible to write code to make it work with a script. This option would be a great addition to Tasmota. Unfortunately I can't (bad software developer) implement that by myself. I own the hardware and I would like to support an implementation with tests.

Thanks a lot!
Javalin

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-05-03:
Hi, 

This can be done using rules feature of Tasmota. You can check in other issues for examples and also in the wiki. 

If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-03 by ascillato2

javier6d2, 2019-05-05:
Please, may you say me how I can get data from my sonoff device?