arendst/Tasmota #5741: Combining RGB with W1 or W2 is not possible any more

mtmayr, 2019-05-04:
### BUG DESCRIPTION

Combining RGB with W1 W2 is not possible any more using the 5 channel H801 (20) hardware. You can see below, that RGB switches from a value to 0,0,0 when a value for W1 is added. Also some wrong values for W1 and W2 get stored: W1=ee, W2=dd got set, and the resulting color is W1=84 W2=7B. The actual color shown seems to suit the values in RESULT; I tried more modules to verify this.

This bug is probably related to issue  #5702 or  #5704, but I am not sure. Below is the same commands on February version, which is working fine. I changed the version from back then, because of permanent reboots.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : H801
- [x] Tasmota binary firmware version number used : 6.5.0.9(sonoff) / self-compiled
- [x] Development IDE - Compiler / Upload tools used : Arduino 1.8.9 IDE
- [x] Provide the output of command ``status 0`` :
```
02:30:19 MQT: stat/sonoff_6F95E6/STATUS = {"Status":{"Module":20,"FriendlyName":["Sonoff"],"Topic":"sonoff_6F95E6","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.100/firmware/sonoff.bin","RestartReason":"External System","Uptime":"2T03:47:14","StartupUTC":"2019-05-01T21:43:05","Sleep":50,"CfgHolder":4182,"BootCount":4,"SaveCount":33,"SaveAddress":"FB000"}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS2 = {"StatusFWR":{"Version":"6.5.0.9(sonoff)","BuildDateTime":"2019-05-01T23:30:12","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["A","B"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS4 = {"StatusMEM":{"ProgramSize":734,"Free":268,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","0FDAE394","000783A1","23B61FCE","01003BC0"]}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS5 = {"StatusNET":{"Hostname":"sonoff_6F95E6-5606","IPAddress":"192.168.1.101","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"60:01:94:6F:95:E6","Webserver":2,"WifiConfig":5}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%04X","MqttClient":"DVES_6F95E6","MqttUser":"socket","MqttCount":8,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS7 = {"StatusTIM":{"UTC":"Sat May 04 01:30:19 2019","Local":"Sat May 04 02:30:19 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:37","Sunset":"19:11"}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS10 = {"StatusSNS":{"Time":"2019-05-04T02:30:19"}}
02:30:19 MQT: stat/sonoff_6F95E6/STATUS11 = {"StatusSTS":{"Time":"2019-05-04T02:30:19","Uptime":"2T03:47:14","Vcc":3.438,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"000000847B","HSBColor":"14,87,100","Channel":[0,0,0,52,48],"CT":333,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"TTD","BSSId":"18:D6:C7:86:86:EE","Channel":6,"RSSI":94,"LinkCount":7,"Downtime":"0T00:00:22"}}}
```

- [x] Provide the output of console when you experience your issue if apply :
```
rogram Version	6.5.0.9(sonoff)
Build Date & Time	2019-05-01T23:30:12
Core/SDK Version	2_5_0/3.0.0-dev(c0f7b44)
01:36:43 CMD: Color ff5522
01:36:43 MQT: stat/sonoff_6F95E6/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF55220000","HSBColor":"14,87,100","Channel":[100,33,13,0,0],"CT":0}
01:36:55 CMD: Color ff552211
01:36:55 MQT: stat/sonoff_6F95E6/RESULT = {"POWER":"ON","Dimmer":7,"Color":"0000001100","HSBColor":"14,87,7","Channel":[0,0,0,7,0],"CT":500}
01:37:09 CMD: Color ff5522eedd
01:37:09 MQT: stat/sonoff_6F95E6/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000847B","HSBColor":"14,87,100","Channel":[0,0,0,52,48],"CT":333}
01:38:12 CMD: Color ff5522
01:38:12 MQT: stat/sonoff_6F95E6/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF55220000","HSBColor":"14,87,100","Channel":[100,33,13,0,0],"CT":0}
01:38:17 CMD: Channel4 100
01:38:17 MQT: stat/sonoff_6F95E6/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"14,87,100","Channel":[0,0,0,100,0],"CT":500}
```

### TO REPRODUCE
On H801, type to console:
Color ff5522
Color ff5522eedd
(value is 000000847B after that)

### EXPECTED BEHAVIOUR
Previous version let me set values for W1,W2 and still be RGB not 0,0,0. Same example from February build below.
```
Program Version	6.4.1.16(sensors)
Build Date & Time	2019-02-13T13:37:58
Core/SDK Version	2_5_0/3.0.0-dev(c0f7b44)
01:39:25 CMD: Color ff5522
01:39:25 MQT: stat/sonoff_6F94D8D8/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF55220000","HSBColor":"14,87,100","Channel":[100,33,13,0,0],"CT":499}
01:39:32 CMD: Color ff552211
01:39:32 MQT: stat/sonoff_6F94D8D8/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF55221100","HSBColor":"14,87,100","Channel":[100,33,13,6,0],"CT":476}
01:39:38 CMD: Color ff5522eedd
01:39:38 MQT: stat/sonoff_6F94D8D8/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF5522EEDD","HSBColor":"14,87,100","Channel":[100,33,13,93,86],"CT":454}
01:39:44 CMD: Color ff5522
01:39:44 MQT: stat/sonoff_6F94D8D8/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF55220000","HSBColor":"14,87,100","Channel":[100,33,13,0,0],"CT":499}
01:39:48 CMD: Channel4 100
01:39:48 MQT: stat/sonoff_6F94D8D8/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF5522FF00","HSBColor":"14,87,100","Channel":[100,33,13,100,0],"CT":153}
```
Greetings!

s-hadinger, 2019-05-04:
Thanks, I'm aware of this regression. It is definitely linked to #5702, I will hopefully work this week-end to relax the way colors are encoded and allow for 5 independant channels.

To my understanding, we normally want bulbs to be either RGB mode or CW white mode, except if SetOption37 is set to 128 or higher. If SetOption37 >=128 then we can combine both. Right?

So when you type : Color ff5522eedd
Option37=0, are we ok it should set 000000eedd (priority to white?)
Option37=128, it should set ff5522eeDD

Jason2866, 2019-05-04:
Your reboots may be caused because you are using core 2.5.0 with SDK pre 3.0.0.
The SDK pre 3.0.0 is known to have unresolved issues. If you want core 2.5.x use  latest feature/stage version. It uses SDK 2.2.x again.

mtmayr, 2019-05-04:
s-hadinger> Ok, then I hope for a good weekend :-)

It's the question of how it _should_ work: I have an 225 RGB LED strip (~2.3A each, on RGB) and 300 LEDs (3A, on W1) warm white, and 300 LEDs UV effect illumination (2A on W2). Automatically mixing parts of W1 to the RGB colors would be possible, but as the W1 strip is much brighter than the RGB, it is hard to adjust. The LEDs on W2 should not automatically be combined, as I turn them on separately. Normally I use mqtt-dash sending the RGB color choosing it from a rainbow palette, and afterwards I send a value for Channel4 to increase the lightness (which was set to 0 from the previous command). 

So there could be a connection between the RGBs and W1, but I am unsure how useful it is, as the configuration of connected LEDs might be versatile. mqtt-dash does not recognize rgbxy values anyway (despite I already mentioned it to the developer). Probably color input with color picker and two sliders for x,y with providing and understanding of rgbxy-hex values would be nice.

Jason2866> Thanks for your advice. My experience using this version combination is rather ambiguous anyway: One device that is connected to my computer' USB power supply has an uptime of 42 days at the moment without reboot. Another device that uses a power adapter rebooted 400 times and another that has a bad wifi connection (but an extenal antenna) rebooted 800 times in the same time span. That's quite annoying, because every reboot comes with a short light flash. But it could definitely be the fault of the power supplies.

s-hadinger, 2019-05-04:
Got it, thanks. Let's leave automatic white mixing from #5704 to a separate PR. 

My question was also: shall we allow manual mixing of RGB and White without Option37 above 128. I had the feeling from reading the code that the answer was no. It means that your use-case of setting RGB and adjusting only W1 should already work if Option37 is set to 128. 

The current limitation is that you cannot set W1 and W2 separately, they are always linked by color tone (ie combined sum of W1 and W2 never above 255). This is what I plan to change. 

I will get an older firmware and check the behavior :)

s-hadinger, 2019-05-04:
Also, I had the constant reboot issue once, I'm just worried it's linked to having 5 simultaneous PWM channel. I'll check if I still have the issue. 

Side question, you get an address in the logs, when there is a crash reboot. I don't know how to turn it into a function address.

mtmayr, 2019-05-04:
Where do I get the address/function from? It just says "Restart Reason | Exception" in Information and "Boot Count" gets increased.

If you see one device as (basically) one light source I understand your intent to merge RGB with W1 and W2. Although it is not possible to have have W1 and W2 be both on 100%. On the other hand, if you see them as 5 separate channels it also makes sense and has its use cases (e.g. for connecting 2 completely different light sources). Therefore it's probably best to have an option to separate the two operating methods. 

I tested "SetOption37 128" and it works already fine (with W1). I'll continue testing if you have any changes. Have a good weekend!

s-hadinger, 2019-05-04:
I just test with Tasmota v6.5.0.

Indeed you can set all 5 channels whatever Option37. I will get this behavior back.

Option37 > 128 only comes when you use CT or HSBColor commands. If Option37 > 128, you can set both separately, if not changing color will deactivate CT and vice-versa.

Jason2866, 2019-05-04:
Using 5 PWM stresses the ESP8266 a lot since PWM has to be done in software.
So the ESP power supply has to be a good one. Most Wemos mini clones only have a Vreg with 300mA this isnt enough for this use case and generates exceptions.

s-hadinger, 2019-05-04:
I have a working version with the new color-mode management in this branch:
https://github.com/s-hadinger/Sonoff-Tasmota/tree/fix/better_colormode

It works fine with RGB and RGBWC (3 and 5 channels), as well as with Alexa Hue Emulation.

I need to do more testing and will prepare the PR tomorrow.

Jason2866, 2019-05-05:
@s-hadinger  Tested your latest color-mode branch, it looks fine for me.
Since the power supply of my AiLight died with a Pof noise and some smoke i have now connected
the electronic part with the RGBW Leds to a labor power supply. 
I had some exceptions with the AiLight before the power supply died. I think the where related to powering issues. I will test it now until tomorrow if this exceptions are gone. If yes we can be sure that problems with exceptions are power issues.
My version is build from your branch with latest dev version from Arduino Core ESP8266
[image]

s-hadinger, 2019-05-05:
Thanks @Jason2866.
I submitted the PR #5748 with the changes.

@mtmayr can you please confirm it solves the issue?

ascillato2, 2019-05-05:
Closing this issue as the fix has been merged.

Thanks everyone for working on this :+1:

closed 2019-05-05 by ascillato2

mtmayr, 2019-05-05:
I can verify, that "SetOption27 128" solves my original problem - version 6.5.0.9(sonoff) for daily use.
But I cannot confirm, that I can go to 100% on both W1+W2 channels, as they are still linked. I tried updated "development" branch and "fix/better_colormode" branch from @s-hadinger . Probably i checked out the wrong branch or I missed an option.

```
00:12:43 CMD: Color ff0011ffee
00:12:43 MQT: stat/sonoff_6F9417/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000847B","HSBColor":"356,100,100","Channel":[0,0,0,52,48],"CT":332}
00:12:53 CMD: SetOption37 128
00:12:53 MQT: stat/sonoff_6F9417/RESULT = {"SetOption37":"128"}
00:12:54 CMD: Color ff0011ffee
00:12:54 MQT: stat/sonoff_6F9417/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF0011847B","HSBColor":"356,100,100","Channel":[100,0,7,52,48],"CT":332}
```
(...ffee -> ...52,48)

@Jason2866 Interesting to know the increased current drain for 5 PWM. I am testing on 7A/12V, don't know if internal power could be low some time. The current version did not reboot for 2 1/2 days :-)

Thanks and greetings!!

brainnex, 2019-05-16:
I don't get it...I have a RGBW (warm white LED strip) with Hue Emulation and Alexa app. With 4 PWM channels is working but i dont get WW in alexa app. So i set 5 PWM channels and now I get CT setting in Tasmota GUI and Alexa app. But control is getting weird. Also the control with alexa app is getting slow (maybe the ESp is getting slow. don't know). With 4 PWM Channels is working without issue.

Issue 1: Dimmer wont change his value if set under 54%. But channel color seems to change like it should. Dimmer values above 54 are setted right. 

Issue 2: Setting color Temperature results in strange behaviour and wont set WW or CW.

ex. Color is 0,0,0,0,0 and i set full WW in alexa app:
`20:32:26 CMD: color 0000000000
20:32:26 MQT: stat/electrodragon1/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"0000000000","HSBColor":"0,0,0","Channel":[0,0,0,0,0],"CT":383}
20:32:29 MQT: stat/electrodragon1/RESULT = {"POWER":"ON"}
20:32:29 MQT: stat/electrodragon1/POWER = ON
20:32:29 MQT: stat/electrodragon1/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFFFF0000","HSBColor":"0,0,100","Channel":[100,100,100,0,0],"CT":383}
20:32:30 MQT: stat/electrodragon1/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFFFF56A9","HSBColor":"0,0,100","Channel":[100,100,100,34,66],"CT":383}`

ex. Color is FF,0,0,0,0 and i set to full WW in alexa app:
`20:34:07 CMD: color FF00000000
20:34:07 MQT: stat/electrodragon1/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00000000","HSBColor":"0,100,100","Channel":[100,0,0,0,0],"CT":383}
20:34:14 MQT: stat/electrodragon1/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF000056A9","HSBColor":"0,100,100","Channel":[100,0,0,34,66],"CT":383}`

In both cases I expect color result to be 00 00 00 00 FF. As you can see color is setted twice.

Status 0 output:
`20:37:29 CMD: status 0
20:37:29 MQT: stat/electrodragon1/STATUS = {"Status":{"Module":0,"FriendlyName":["Wandbeleuchtung"],"Topic":"electrodragon1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:37:29 MQT: stat/electrodragon1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"2T23:40:40","StartupUTC":"2019-05-13T19:56:49","Sleep":50,"CfgHolder":4617,"BootCount":21,"SaveCount":501,"SaveAddress":"F4000"}}
20:37:29 MQT: stat/electrodragon1/STATUS2 = {"StatusFWR":{"Version":"6.5.0.10(1cdc046-sonoff)","BuildDateTime":"2019-05-13T17:04:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:37:29 MQT: stat/electrodragon1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Machete_braucht_kein_WLAN",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000180000000000000000000000000","00000000"]}}
20:37:29 MQT: stat/electrodragon1/STATUS4 = {"StatusMEM":{"ProgramSize":523,"Free":480,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16301C","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CC","01003BC0"]}}
20:37:29 MQT: stat/electrodragon1/STATUS5 = {"StatusNET":{"Hostname":"electrodragon1-3059","IPAddress":"192.168.178.44","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.76","Mac":"EC:FA:BC:1F:0B:F3","Webserver":2,"WifiConfig":4}}
20:37:29 MQT: stat/electrodragon1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.76","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1F0BF3","MqttUser":"DVES_USER","MqttCount":3,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:37:29 MQT: stat/electrodragon1/STATUS7 = {"StatusTIM":{"UTC":"Thu May 16 19:37:29 2019","Local":"Thu May 16 20:37:29 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:07","Sunset":"20:25"}}
20:37:29 MQT: stat/electrodragon1/STATUS10 = {"StatusSNS":{"Time":"2019-05-16T20:37:29","ANALOG":{"A0":71}}}
20:37:29 MQT: stat/electrodragon1/STATUS11 = {"StatusSTS":{"Time":"2019-05-16T20:37:29","Uptime":"2T23:40:40","SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"FF000056A9","HSBColor":"0,100,100","Channel":[100,0,0,34,66],"CT":383,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Machete_braucht_kein_WLAN","BSSId":"B8:27:EB:56:29:53","Channel":1,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:07"}}}
20:40:21 MQT: tele/electrodragon1/STATE = {"Time":"2019-05-16T20:40:21","Uptime":"2T23:43:32","SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"FF000056A9","HSBColor":"0,100,100","Channel":[100,0,0,34,66],"CT":383,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Machete_braucht_kein_WLAN","BSSId":"B8:27:EB:56:29:53","Channel":1,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:07"}}
20:40:21 MQT: tele/electrodragon1/SENSOR = {"Time":"2019-05-16T20:40:21","ANALOG":{"A0":69}}`

s-hadinger, 2019-05-16:
When setting 4 PMW, you should have both colors and white-colortone choices in Alexa app; although cold/warm settings won't change anything because you only have 1 white channel. You should not need to set 5 PMW.

Anyways, it's probably not a good idea to 'setOption27 128' with Alexa app, because Alexa does not support having both colors and white at the same time.

The issue you are facing is because the dimmer value returned by Tasmota is the max of RGB-dimmer and White-dimmer. In your case, it seems that one dimmer is at 54%, hence when you change the other you cannot go below 54%. That's also why it's slow: if you choose a dimmer below 54%, Alexa will still receive a 54% value and will consider it as a wrong value, then eventually display back 54%.

As a conclusion, don't set SetOption27 128 with Alexa+Hue Emulation, or face strange behavior.

brainnex, 2019-05-17:
It seams that I did something wrong at testing with the app and I thought it is not working and I have began trying 5 PWMs and SetOption37. But your are fully right. It works now. Many thanks for the new color mode. 
 Ps: you mean for sure SetOption37 not SetOption27.

s-hadinger, 2019-05-18:
Glad to see it worked, and yes I meant SetOption37. 

To answer your last question, Hue emulation only supports all RGB or all CT. What would you like to do?

brainnex, 2019-05-18:
I mean when there is no physical cold white channel the RGB channels could be used to make cold white. So the color temperature (ex from alexa app) is working even when there is no cold white channel. 

It's just an idea. No big issue.

Jason2866, 2019-05-18:
@brainnex  would be a can of worms. There are so many different RGB led typs. Furthermore RGB leds with different dimming values arent working linear. So you would get more or less white if you dim.