arendst/Tasmota #5743: Wrong SSID after manually activating Wi-Fi manager mode

DrJume, 2019-05-05:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
After I manually invoked the Wi-Fi manager mode by clicking the button 4 times on my Sonoff S26, the SSID was not "sonoff-XXXX" but appeared in the format of "ESP-XXXXXX".

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : _____
- [x] Tasmota binary firmware version number used : 6.5.0 / (pre-compiled)
- [x] Development IDE - Compiler / Upload tools used : ____ / ____
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
03:28:33 RSL: STATUS = {"Status":{"Module":8,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:28:33 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:38","StartupUTC":"2019-05-05T02:20:55","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":34,"SaveAddress":"FA000"}}
03:28:33 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
03:28:33 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["InternetzBox","SonoffAP"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","280500000100000000000000000000000000","00000000"]}}
03:28:33 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
03:28:33 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-0238","IPAddress":"192.168.178.41","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"84:F3:EB:3E:00:EE","Webserver":2,"WifiConfig":2}}
03:28:33 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun May 05 02:28:33 2019","Local":"Sun May 05 03:28:33 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:23","Sunset":"20:10"}}
03:28:33 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-05-05T03:28:33"}}
03:28:33 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-05-05T03:28:33","Uptime":"0T00:07:38","Vcc":3.138,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"InternetzBox","BSSId":"C8:0E:14:8F:22:7C","Channel":12,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

No acces to log while in wifi manager mode.
```

### TO REPRODUCE
1. Set the module to "Sonoff S2X"
2. Configure a Wi-Fi connection in the Web GUI. (AP1 with password, AP2 named "SonoffAP" without a password)
3. Press the hardware button 4 times.

### EXPECTED BEHAVIOUR
Access point appears with SSID formatted as "sonof-XXXX".

Jason2866, 2019-05-05:
This happens sometimes and is not under the control of Tasmota.
The Arduino SDK catches this request before Tasmota can act. Solution try again or power device off and on. If you search in issues you find a more detailed explanation too.

ascillato2, 2019-05-05:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-05 by ascillato2