arendst/Tasmota #5744: Feature Request: subscribe to additional toppics

Tommy-LSA, 2019-05-05:
I'm searching for a possibility to publish a message around and let different iot devices react on same message.

Examples:

I use a battery on my photovoltaic over night. At the point the battary is taking over in evening a message is published over MQTT /photovoltaic/battery/drain = 1 Now it would be fine if I could set some tasmota devices to listen to this message and react on with rules (e.g. locking switching on the device) Second example: The Inverter is limitted to feed only 70% into grid. If this limit is nearly reached a message is published over MQTT /photovoltaic/plant/cut70 = 1. This message, if received, would be fine to switch on different things like a pool pump or sprinkler system to avoid cutting power on inverter.
I have seen that espeasy has this possibility (subscribing any toppic) but all my iot devices are on Tasmota. So it would be bad to mix the iot devices only for that reason.

Thanks

Jason2866, 2019-05-05:
Tamota is not designed to and such features will not be implemented. This would need a complete redesign and rewriting alot of code which increases code size -> loosing features Tasmota now has.
 Such use cases can easily solved with a Home Automation system.

Tommy-LSA, 2019-05-05:
Well, no problem at all. If this is the strategy - good to know. I have to change to espeasy mid term. Wasn't aware that subscribing an additional toppic and making it usable within the rules will end in a complete re-design of Tasmota.

I know that things like that are also manageable by home automatization. I use FHEM since 10 years or so. FHEM is publishing this generic MQTT Messages. So I also know that's the less intelligent way, setting rules in Tasmota from outside while a tasmota rule could react on a generic message directly. How the existing rules can help me here without receiving the generic MQTT messages - would great to see an example. Same on how KNX can solve it.

meingraham, 2019-05-05:
Well, you can have "custom" topics of sorts. I suppose you could have rules that trigger on an event name and you can publish to that event topic. Again, just an idea.

You could further incorporate a specific group topic so you can publish to a set of devices at once.

I can only point you to the Tasmota rules wiki and let you play with those.

If you want to bounce off more ideas, I recommend you continue this dialog over on the Tasmota Discord chat. But, since it's likely not going to be added as a feature in the firmware, you can probably close this GitHub issue.

Cheers!

Mike

Tommy-LSA, 2019-05-05:
> Well, you can have "custom" topics of sorts. I suppose you could have rules that trigger on an event name and you can publish to that event topic. Again, just an idea.``

Don't understand. The device subscribe in MQTT setup on /myhome/garden/sprinklerpump. How can I subscribe additionally to /photovoltaic/battery/drain and use it in tasmota rules? In ESPEasy it's simply possible with module/device "MQTT Generic Import" which allows to subscribe to 4 additional toppics and handle it like sensordata. (use in rules, show on display etc)

I havn't found a rule example in wiki, which allows me to subscribe to a different toppic.

meingraham, 2019-05-05:
You're going to have to think about the solution differently.

You would publish to

`cmnd/<groupTopic>/event "<eventName>=<value>"`

And the rule on the devices with that <groupTopic> would be something like:
`ON event#battery=1 DO <action(s)> ENDON`

The rule actions can be any Tasmota command, e.g., `POWER` the relay on or off, etc. Commands could set the value of a variable and then, another rule can trigger on different values to perform different actions. It could even detect certain values and `PUBLISH` an MQTT message.

What do you mean by "locking switching on the device". Do you mean to disable the external input (switch or push button) or disable the switch from turning on? If the former, I'm not sure how you would do that with Tasmota settings.

So... now then, Discord it is for any further dialog. TTFN

Mike

Tommy-LSA, 2019-05-05:
Got a workaround via Discord. THX @meingraham

closed 2019-05-05 by Tommy-LSA

e-t-h, 2019-07-18:
Isn't this what you are looking for?
https://github.com/arendst/Sonoff-Tasmota/pull/5320