arendst/Tasmota #5745: software serial missing RX bytes

boaschti, 2019-05-05:
### BUG DESCRIPTION
I started to implement a Fingerprint Sensor which communicates via uart 57600baud. In Tasmota I selected SW serial on pin 4,5.
The Adafruit Fingerprint lib sends a ccommand to the sensor, the sensor answers and in Tasmota I get only the last byte. See screenshot last byte 0x1D and the print in Tasmota console below.

[image]

Thats the current code:
https://github.com/boaschti/Sonoff-Tasmota/tree/Add_AS608_Fingerprint/sonoff

The uart buffer is read in Adafruit_Fingerprint.cpp Line 309-377 per byte = mySerial->read();
I print the RX data in XSNS_43_AS608.ino with:
`    if (finger.verifyPassword()){
      snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_LOG " AS608x found"));
    }else{
      snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_LOG " AS608x not found %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X"), finger.mydata[0], finger.mydata[1], finger.mydata[2], finger.mydata[3], finger.mydata[4], finger.mydata[5], finger.mydata[6], finger.mydata[7], finger.mydata[8], finger.mydata[9], finger.mydata[10], finger.mydata[11]);
    }
    AddLog(LOG_LEVEL_INFO);`

Has anybody a idea what happens here? Is it a timing problem (sensor response too fast)?

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (i.e. Sonoff Basic) : wemos d1 mini
- [x ] Tasmota binary firmware version number used : 6.5.0 / (self-compiled)
- [x ] Development IDE - Compiler / Upload tools used : atom / ____
- [ x] Provide the output of command ``status 0`` :

Jason2866, 2019-05-05:
Please provide Status 0 output

boaschti, 2019-05-05:
Correct output?

Jason2866, 2019-05-05:
No, you have to open webfrontend and from there click Button "Console" now type in Console Command `Status 0` Post the output generated in conesole from this here. Please us function "Insert code" this can be reachec via the symbol "<>"

Jason2866, 2019-05-05:
Just take a look at your branch. You use are very old and outdated version. 
Please update to latest dev branch there are MANY issues fixed since that time.
Anyway if you use Core 2.3.0 you will have problems with software serial. See https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#arduino-core-differences
Switch to core 2.4.2 or latest core 2.5.x feature/stage

Jason2866, 2019-05-05:
You can take a look in driver https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_40_pn532.ino how software serial can be implemented in a custom driver. This driver works very reliable

ascillato2, 2019-05-05:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-05 by ascillato2

boaschti, 2019-05-06:
Newest dev branch works great! Much Thanks.