arendst/Tasmota #5746: SetOption13 variants for immediate response

leo-b, 2019-05-05:
I'd like to vote for two SetOption13 enhancements that could enable Button hold and/or multi-press functionality while still responding immediately.

Currently hold and multi-press functionality is disabled if SetOption13 is set to 1.

I am proposing two additional modes:

**SetOption13 = ``2``:** single-press and hold functionality could work together when triggering on button-release:
    - short press (time between button press and release lower than SetOption32): Immediately trigger button state 2 on button release.
    - long press: Immediately trigger button state 3 on button release.

**SetOption13 = ``3``:** multi-press could also be enabled with immediate actions by just triggering the single-press action immediately and additionally a the double-press action later once a double press has been detected.

My use case is an extended [stair-case timer](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#10-simple-staircase-timer) that uses long-press to increase the timer to e.g. 2 hours and uses double-press to prematurely turn off the light:

    Rule1 on button1#state=3 do backlog power1 1; RuleTimer1 20 endon on button1#state=2 do backlog power1 1; RuleTimer1 10 endon on Rules#Timer=1 do power1 off endon

(This would work even for my proposed **``SetOption13 3``** because by default double-press toggles the relay and since the above rule powers on for a single press, the second press would power off.)

Jason2866, 2019-05-05:
tbh, i dont no if this change is possible from design in Tasmota.
Someone has to do a PoC PR. Since you want the feature :-)

ascillato2, 2019-05-05:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-05 by ascillato2

leo-b, 2019-05-05:
Of course I did read the policy guidelines.
Besides #2186 which is incomplete and auto-closed because of inactivity, there is no issue that covers this topic.
If there is, please add a reference.

Jason2866, 2019-05-05:
Inactivity is a closing reason -> nobody does work on this.
A PoC PR is highly welcome from someone who is willing to invest time.