arendst/Tasmota #5749: RF Bridge, no codes displayed in console

fiechr, 2019-05-05:
I'm using a R2 type Sonoff RF Bridge with the default release binary 6.5.0. The web interface is working fine and I switched to module type "Sonoff Bridge (25)". I also disabled the serial logging, since as far as I understand, the serial connection is used for communication with the RF controller.

The issue is, that I don't get any RF codes or messages after entering "rfraw 177" in the web console.

I only get
`MQT: stat/sonoff-rfbridge/RESULT = {"RfRaw":"ON"}`
nothing more.

At first, I've tried with the default firmware and now with the Portisch firmware. In both cases the red LED was/is flashing, when I press some buttons one of my RF remotes. But I still can't see any new messages in the console. The RF controller itself seems to work fine.

The switch on the PCB is switched back to the on position.

ascillato2, 2019-05-05:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-05 by ascillato2

joaomoreira-obvio, 2021-09-24:
@fiechr I'm having the same issue, have you solved the problem? Thanks