arendst/Tasmota #5750: Domoticz add tempo ds18x20 multi sensors

roblad, 2019-05-05:
Could you add in xsns_05_ds18x20.ino

line 424
char data[70];

line 450
add

        if ((0 == tele_period)) {
         if (0 == i) {
            DomoticzSensor(DZ_TEMP, temperature);
         } else {

            snprintf_P(data, sizeof(data), DOMOTICZ_MESSAGE,int (Settings.domoticz_sensor_idx[DZ_TEMP]) + i, 0,temperature, DomoticzBatteryQuality(), DomoticzRssiQuality());
            MqttClient.publish(domoticz_in_topic, data,0);
            delayMicroseconds(500);
         }

ascillato2, 2019-05-05:
@roblad

Hi,

Please, make a Pull Request with your changes in order to be reviewed by Theo. 

Thanks

ascillato2, 2019-05-05:
If you need help on how making a PR, please, just ask here or in the Tasmota support chat. Thanks

closed 2019-05-05 by ascillato2