arendst/Tasmota #5755: Question around Reconnection to the MQTT server

nosaj66au, 2019-05-06:
Hi Guys,
   Hoping someone a answer a query I have around re connection to an MQTT broker and the time taken to do so. I have developed a method of using a floating IP between two RPI's running in hot /standby mode. Each Rpi has a physical IP Address and a single virtual IP which floats between them, where only the hot has the active floating IP. Install and running on each Rpi is an MQTT broker. The theory is, the Tasmota devices only talks to the MQTT broker via the floating IP and the brokers a bridged by the physical IP. When the active Rpi fails an automatic role to the standby Rpi will occur. 
 Now to my problem, on booting the tasmota devices all works well and MQTT is sent an received by the active Rpi on the floating IP. The connection is achieved almost instantly in one attempt. However if I kill the active Rpi causing a role to the standby Rpi. The floating IP will active within 5 sec and ready to accept MQTT client connections with 15-20sec. However Tasmota can take up to 3 minutes to connect even though is attempts ever 10 sec as expected. If I reboot the Tasmota device is connects first attempt which is quicker than waiting for MQTT alone to connect. Although this could be a solution it is a but messy.  Is the away speed up connection without a reboot?
Jason

Jason2866, 2019-05-06:
As you wrote correct Tasmota tries after 10 sec to get a recconect to the broker. The laggy behaviour is NOT under control of Tasmota it is Arduino ESP8266 and/or Espressif (closed source) related. 
You can try the different cores.

s-hadinger, 2019-05-06:
Can you give us more details about the way you are implementing the floating IP address on RPIs?
It looks like the IP did switch to RPI2 but the ARP table was not update on Tasmota device and/or on network switch. When you reboot Tasmota, ARP cache is empty - this makes me feel it's linked to ARP on Tasmota device.
I'm not an expert in ARP tables, but I guess the RPI2 should broadcast an ARP request to signal it is now the owner of the IP address. This is not under the control of Tasmota software though.

ascillato2, 2019-05-06:
You can try Tasmota with core 2.5.0 that is faster

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-06 by ascillato2

joba-1, 2019-05-06:
really looks like a missing gratuitous arp on the rpi taking over the
floating ip.

look here for a possible solution using send_arp:
https://wiki.wireshark.org/Gratuitous_ARP

lalo-uy, 2019-05-06:
Your problem is on ARP, not on IP.
The ARP in the ESP core has resolved the IP-> MAC address of the floating
IP to the first RPi.
It will  retry that resolving only after several time out .

richard-scott, 2019-06-06:
I'm also getting this issue. My use case is reproducible 😁

1. Setup Tasmota to talk to an mqtt server. 
2. Restart Tasmota and it obviously reconnects. 
3. Stop your mqtt server. 
4. Delete any persistent session data on your mqtt server, but leave your Tasmota powered up. 
5. Start your mqtt server with no session data. 
6 Watch as your Tasmota attempts to reconnect every 10 Sec and fails. 
7. Reboot your Tasmota and watch it nicely reconnect. 

There is no ARP issue with me, and I believe the lack of session data on the 'other' Rpi is why @nosaj66au  Tasmota device doesn't reconnect.

Jason2866, 2019-06-06:
Cant reproduce. If i shutdown PI (mqtt broker and Open Hab) and start it again all my Tasmota devices
reconnect to broker. In the time in between i cant notice a lag. Devices are reacting local and via web frontend.

richard-scott, 2019-06-06:
@Jason2866 did you delete your MQTT Session Data? If not the you'll never reproduce the issue.

arendst, 2019-06-06:
If you expect mqtt clean session to be cause of the problem, edit mqtt.ino and add a 0 to the mqtt connect cal.

CRCinAU, 2021-05-17:
> I'm also getting this issue. My use case is reproducible grin
> 
>     1. Setup Tasmota to talk to an mqtt server.
> 
>     2. Restart Tasmota and it obviously reconnects.
> 
>     3. Stop your mqtt server.
> 
>     4. Delete any persistent session data on your mqtt server, but leave your Tasmota powered up.
> 
>     5. Start your mqtt server with no session data.
> 
>     6. Watch as your Tasmota attempts to reconnect every 10 Sec and fails.
> 
>     7. Reboot your Tasmota and watch it nicely reconnect.
> 
> 
> There is no ARP issue with me, and I believe the lack of session data on the 'other' Rpi is why @nosaj66au Tasmota device doesn't reconnect.

I'm still seeing this with 9.4.0 - and can reproduce it every time via either rebooting my HomeAssistant VM - or doing a `systemctl restart mosquitto`...