arendst/Tasmota #5756: Support for Blitzwolf BW-SHP5

Cybolord, 2019-05-06:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2019-05-06:
It is not on the supported modules list and henceforth not supported. If you cannot map the gpio's yourself you'll have to wait for someone else with the device to do it. Keep an eye on the [template repo](http://blakadder.github.io/templates)

Cybolord, 2019-05-06:
so I would like to offer my help and not only waiting for "someone else" indeed but need a few hints ;)

[image]

blakadder, 2019-05-06:
It is appreciated but you have failed to read the message when opening an issue. This is not the place to configure your new device. Read the wiki on Templates and new device configuration or seek help in [discussion channels ](https://github.com/arendst/Sonoff-Tasmota/wiki#community)

meingraham, 2019-05-06:
@Cybolord - Hints available [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Templates) and [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices).

Cybolord, 2019-05-06:
> 
> 
> @Cybolord - Hints available [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Templates) and [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices).

...great, thanks a lot meingraham!

zewelor, 2019-05-06:
@Cybolord How did you flashed tasmota ? Using tuya-convert or flasher via pins ?

GoNzCiD, 2019-05-09:
> 
> 
> @Cybolord How did you flashed tasmota ? Using tuya-convert or flasher via pins ?

I can't by tuya-convert. I have flashed it via pins

GoNzCiD, 2019-05-09:
@Cybolord, can you made it work? I have just flashed one SHP5 with tasmota 6.5.0

zewelor, 2019-05-09:
@GoNzCiD DO you have photo which pins and how did you connect flasher ?

GoNzCiD, 2019-05-11:
[image]
IOD => IO0 short to gnd on boot to enter in flash mode. @zewelor 
And flashed with NodeMCU-PyFlasher-4.0

zewelor, 2019-05-12:
@GoNzCiD Thanks for info. Just to be sure, 5V is correct, not 3.3V ?

GoNzCiD, 2019-05-13:
> 
> 
> @GoNzCiD Thanks for info. Just to be sure, 5V is correct, not 3.3V ?

Yes, look at the pcb on the other side:
[image]

zewelor, 2019-05-13:
Right thanks !

waiet, 2019-05-14:
I adding more photos
[image]
[image]

thomasklingbeil, 2019-05-16:
[image]

Energy monitoring is done with an HLW8032 (https://datasheet.lcsc.com/szlcsc/Hiliwei-Tech-HLW8032_C128023.pdf)
I guess one of the outputs (probably PF) is sent through an photocoupler (EL357N) to one of the IO pins of the ESP8285.

thomasklingbeil, 2019-05-16:
Apart from the power monitoring, the device already seems to work: https://github.com/blakadder/templates/blob/master/_templates/blitzwolf_SHP5

To update my guess from earlier today: It seems as if PF is the only pin of the HLW8032 that is NOT connected. Thus, it seems more likely, that the serial connection is used. It should be UART with 4800bps, Start/Stop Bit and even parity. Every 50ms there should be 24 byte of data.

arendst, 2019-05-17:
Just ordered one to write a new nrg driver for serial HLW8032 support.

arendst, 2019-05-17:
While comparing the datasheets from HLW8032 (chinees) with CSE7766 (english) the resemblance is striking.

For those owning this device you might want to try the CSE7766 RX/TX config option in Tasmota to check if any energy measuring is shown.

kamkilt, 2019-05-20:
> Apart from the power monitoring, the device already seems to work: https://github.com/blakadder/templates/blob/master/_templates/blitzwolf_SHP5
> 

This module have also control of USB port on GPIO5.
My config is following: 
template: '{"NAME":"BlitzWolf SHP5","GPIO":[56,255,57,255,255,22,255,255,255,255,21,255,17],"FLAG":0,"BASE":18}'  - button contorling 230 V relay
 or 
template: '{"NAME":"BlitzWolf SHP5","GPIO":[56,255,57,255,255,22,255,255,255,255,21,255,18],"FLAG":0,"BASE":18}' - button contorling 5 V USB

Energy measuring using CSE7766 RX/TX on GPIO1 and GPIO3 is not working.

arendst, 2019-05-20:
Thx for testing CSE7766. When mine arrives I'll try to implement HLW8032

thomasklingbeil, 2019-05-20:
Does the regular Serial output of Sonoff Tasmota need to be disabled first?

arendst, 2019-05-20:
No, that's been handled by the CSE7766 driver. Did you try to swap RX/TX too?

thomasklingbeil, 2019-05-20:
I’ll give it a try soon. Unfortunately it seems that my first SHP5 has stopped working...

Side note: The ESP and the power monitor

kamkilt, 2019-05-20:
> No, that's been handled by the CSE7766 driver. Did you try to swap RX/TX too?
> 
> This works for CSE7766:
> 
> [image]

It started working, i have changed SerialLog to 0, Baudrate to 4800, but also trying to set CSE7766 on different GPIOs, so i'dont what finaly heleped, but it works on GPIOs that You send.

On start voltage was about 129 V, i calibrated it to my volteage 240 V.

I will check if it showing right values in the evening.

[image]

kamkilt, 2019-05-20:
So my final template are:
{"NAME":"SHP5","GPIO":[56,145,57,146,255,22,0,0,255,255,21,255,18],"FLAG":0,"BASE":18}

arendst, 2019-05-20:
Only the shown GPIO's are supported for CSE7766 serial connection so all others will fail.

Glad it works!

miquelbotanch, 2019-06-02:
Anyone has been able to flash it ?

I did it many times without issues with BW-SHP6 but with this the RX light of the programmer never turns on. (tried on 2 different SHP5)

Maybe I should use an external power supply? My programmer has 5<->3,3 power switch but doesnt work.

arendst, 2019-06-03:
Just did a serial upload as tuya-convert doesn't work anymore due to unsupported https.

[image]

Flashing went fine.

Make sure that once you activate the SHP5 template serial comms changes to 4800 bps even parity. That worked for me.

Jason2866, 2019-06-03:
Fancy soldering & wiring 😊

kamkilt, 2019-06-03:
I calibrated BW-SHP5 using 60 W bulb, with load energy measure is correct, but without anything connected it always shows about 3 W, it's not an USB output, because is not changing when something is connected to USB output, is there some way to calibrate energy measure to show 0 W when nothing is connected?

I have two modules and both shows same values.

[image]

arendst, 2019-06-04:
Probably not as I suspect the sensor is connected before the SHP5 logic so it will always measure the load of the SHP5.

arendst, 2019-06-04:
The final SHP5 template should be this:
```
{"NAME":"SHP5","GPIO":[57,145,56,146,0,22,0,0,0,0,21,0,18],"FLAG":0,"BASE":18}
```
where the blue led represents the link info and the red led follows any power setting. As other GPIO's cannot be used they need to be set to 0.

arendst, 2019-06-04:
Updated wiki and closing

closed 2019-06-04 by arendst

arendst, 2019-06-04:
@kamkilt if you do not want to measure loads below 5W you might want to set define CSE_MAX_INVALID_POWER in file xnrg_02_cse7766.ino from 128 to 1.

This will disable showing invalid power measurments usually below 5W.

arendst, 2019-06-04:
Use SetOption39 to control handling of invalid power measurments.

The HLW8032 reports invalid power often if values below 5W are found. During this situation it sometimes reports avalid load. By setting SetOption39 to 128 (default) it must read at least 128 invalid power readinfs before deciding there is no load. 

So to discard all loads below 6W simply set SetOption39 to 1 (0 will reset to default on next restart) so it will report no load below 6W.

chkbln, 2019-06-06:
Hello
How can I open the case? There is a screw on the bottom. I could solve that.
But how is it going now?
Can someone give me a hint.

Thank you

waiet, 2019-06-07:
> Hello
> How can I open the case? There is a screw on the bottom. I could solve that.
> But how is it going now?
> Can someone give me a hint.
> 
> Thank you

There are 1 screw and on the corners are clips. After screwing out the screw I opened case with old club card
[image]
[image]

waiet, 2019-06-07:
> I calibrated BW-SHP5 using 60 W bulb, with load energy measure is correct, but without anything connected it always shows about 3 W, it's not an USB output, because is not changing when something is [...]

I measured the energy consumption without a load with original fw. Power consumption  still was circa 4w on blitzwolf app and too on power meter after shp5. See image
[image]

meingraham, 2019-06-07:
Does it draw power with the relays off?

waiet, 2019-06-07:
On this img relays were  power on, but with power off is result the same

meingraham, 2019-06-07:
Turn on weblog 4 and see if you are getting any checksum errors.

What is Relay2 controlling?

kamkilt, 2019-06-08:
> What is Relay2 controlling?

USB Output.

arendst, 2019-06-08:
If you do not want to see measurements below 5W use the latest dev release and execute command ``setoption39 1`` as documents earlier in this thread.

kamkilt, 2019-06-08:
> If you do not want to see measurements below 5W use the latest dev release and execute command `setoption39 1` as documents earlier in this thread.

I did and it's working well. thanks.

waiet, 2019-06-08:
But shp 5 compared to the shp 6 has relatively high consumption without any load.
That's probably not a software issue. Tasmota is working perfectly.

kamkilt, 2019-06-09:
> But shp 5 compared to the shp 6 has relatively high consumption without any load.
> That's probably not a software issue. Tasmota is working perfectly.

I measured mine using Sonoff pow R2 and SHP5 gets around ~ 1 - 1.2 W, while SHP6 takes around 0.8 - 1 W, relays on both turned on.

Connecting SHP5 to SHP6, and reading measures from SHP6 gives similar results, so I don't see that high difference in consumption.

maschere, 2019-06-27:
It seems the Gosund SP112 (https://www.amazon.de/dp/B07PMW88L7) is the same as the SHP5, at least the template for the SHP5 works flawlessly for the Gosund as well. SO thanks a lot for that!

However I'm also getting strange power readings of the power consumption of the switch itsels. Without any load connected and both relays set to OFF, my power meter says 9 Watts (the same with both relays on). When I measure one SP112 with the other SP112 (after calibration) it says about 1 Watts. Can anyone with a previse power meter measure the SHP5 (or SP112) without any load?

hemy81, 2019-08-04:
Hey guys, thanks for this nice discussion so far. I have been trying to flash the SP112 without any success so far. I have the following items:
- Gosund SP112
- esptool (executable) for Mac
- USB adapter CP2102 (/dev/tty.SLAB_USBtoUART)

I think it cannot enter the flash mode. I shortened the IO0 and Gnd, upon plugin in the UART, but I only hear the relay "clicking" but the blue LED is still blinking. The only difference I hear is the relay that clicks once I remove the IO0 and Gnd cable. 

Any help is very welcome.

chkbln, 2019-08-04:
Hi
I connect GPO0 and GND together and then I plug the UART into the USB port.
The LED lights up red. Then I flash.
greeting

hemy81, 2019-08-05:
Hmm, that‘s what I did. I connected IO0 with GND and put the UART into USB port. The LED is not blinking red, but blue. I waited even for 10s with both GND and IO0 connected.

chkbln, 2019-08-05:
- GOSUND SP112 is broken?
- GPO0 is not at GND potential? Test the connection for correct contact.
I flashed 4 sockets on Tasmota without any problems

wvsbsp, 2019-08-19:
I just received my GOSUND SP112's and managed to open them without breaking any tabs. ;-)

Just one Idea: Has anybody thought about routing the unused GPIOs 4, 12 and 13 to the unused USB-datalines? This would be perfect for adding sensors via 1-wire, GPIO or I2C. 5V power supply included for free...
One Pin of the ESP-module is not clear to me. The pin between Vdd and GPIO14 - is it TOUT (ADC0) or IO15?

And some additions to the circuit itself: U3 (device-marking 2634 A833 - which is the date-code?) is a dual USB protection IC (such as TI TPS2513A or ST USBLC6-4) which protects plugged USB devices - good, but the pins have no other connection.
In the picture in article https://github.com/arendst/Sonoff-Tasmota/issues/5756#issuecomment-492991170
there are two ICs marked K0443I ans K0268I. This might be the production code. The ICs are one On-Bright OB2502TCPB and one OB2005VCPK. Both should be PWM regulators for the separated power-supplies. Unfortunately, i couldn't find the datasheets.

ascillato, 2019-08-19:
@wvsbsp

Please, NEVER CONNECT SENSORS TO A POWER MEASUREMENT DEVICE. 

YOU CAN BE ELECTROCUTED. 

the gnd pin is directly connected to Mains AC. 

For sensors please use a nodemcu instead. 

Devices with power measurement MUST BE USED AS IT COMES FROM FACTORY. 

DO NOT PUT YOU, YOUR FAMILY AND YOUR HOUSE IN DANGER 

Please, read the wiki! This warning is everywhere.

wvsbsp, 2019-08-20:
For shure it is not the case with th SP112. I plan to stick the sensors into the two external USB charging ports which are certainly isolated from mains.... 
Look at the device, the power measurement uses its own power-supply and the communication to the esp runs over an optocoupler.

Jason2866, 2019-08-20:
@ascillato2  please lock this thread
@wvsbsp  We do NOT want discussions with dangerous mods here.