arendst/Tasmota #5764: Possible bricking bug in 6.5.0?

badabing2005, 2019-05-08:
### BUG DESCRIPTION
Bricked two CE Smart Home LA WF3 Wifi Plug](https://github.com/arendst/Sonoff-Tasmota/wiki/CE-Smart-Home---LA-WF3-Wifi-Plug-(TYWE2S). during configuration changes that required restarts

The first one was tasmotized by opening the case and flashing using FTDI serial connection
The second one was flashed using Tuya-convert.

Although different flashing methods, both devices exhibited similar issues, hence why I'm leaning more towards Tasmota potential issue than Tuya-Convert bug.

In both cases I was able to get to the configuration and setup the devices properly, and they were working fine with MQTT off.

Then I needed to enable MQTT, and it needed to restart, the device never connected back to Wifi
Even though the Wifi configuration was previously working fine prior to the restart.
I was able to do 40s reset and start over
Again at the stage of enabling MQTT it stopped connecting to wifi, the blue LED would flash
I was persistent and it eventually worked, but then I needed another change (enable UI admin password) which required a restart and the problem recurred.

It is worth noting that this is not the first device I setup MQTT, I'm quite fluent and have too many Tasmotized devices properly working.
Also worth noting that at least in one case I was intentionally setting up to an MQTT address that was not reachable, as the plan was to configure the device ahead of time for a friend's Wifi / MQTT 

At times I was able to get into Wifi manager with 4 quick button presses instead of doing the full 40s reset.
Something curious about the 4 quick button method
When the device would be in WAP mode, the displayed SSID was not consistent
It would advertise as one of these
- ESP-<MAC-numbers>
- Sonoff-<MAC-numbers>
- Hostname (or was it the friendly name?)

I set a ping to 192.168.4.1
and setup my PC to reconnect automatically
I would see constant disconnects, the WAP would disappear, and then come back,
at times I would manage to type in my Wifi settings, but it would disconnect before it saved, at other times I'd manage to save and things would be working, again

But of course each time I wanted to make changes it would run into issues.
Eventually 4 quick presses wouldn't work anymore, and lastly even 40 seconds reset didn't work anymore.

No more flashing blue led
I cracked open both plugs and connected FTDI serial to flash and start clean.
I was no longer able to establish serial connection.

I believe both plugs are bricked.
I'm not sure if it was my imagination or not, at times I was unplugging, replugging it in hopes of reset and felt the device to be warmer than normal.

I reporting this here in case there is a possibility that Tasmota is causing the brick.
The majority of my Tasmota devices (and I have many) are all on 6.5.0

Are there any known issues with the latest stable release? should I move to dev channel?
I would have though the dev channel would be more bug prone than the stable version.

If you suspect a user error by all means please let me know and close this ticket.
Thanks

BB

 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
Close enough issue:
https://groups.google.com/forum/#!searchin/sonoffusers/bricked|sort:date/sonoffusers/UBcempVJkNQ/LTZTTwqvAwAJ
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : [CE Smart Home LA WF3 Wifi Plug](https://github.com/arendst/Sonoff-Tasmota/wiki/CE-Smart-Home---LA-WF3-Wifi-Plug-(TYWE2S)) 
- [x] Tasmota binary firmware version number used : 6.5.0 / official stable precompiled
- [x] Development IDE - Compiler / Upload tools used : ESPEasy / Tuya-convert
- [x] Provide the output of command ``status 0`` : 
```
STATUS 0 OUTPUT HERE:
bricked, can't grab it

```

### TO REPRODUCE
see above

### EXPECTED BEHAVIOUR
After saving the settings and restart, the device would work as normal

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Perhaps these are relevant
with 6.5.0 I used the following template
```
{"NAME":"CE_Plug","GPIO":[255,255,255,255,56,57,255,255,21,17,255,255,255],"FLAG":1,"BASE":18}
```
I set `switchmode1 7` and `pulsetime1 100` as I needed the plug to be momentarily on upon button press (although I expected this to be 10 seconds, it was never precise, at times it would take about 18-20 seconds) 

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2019-05-08:
Since you did not manage to erase the chip's flash completely and make a clean flash of Tasmota the cause of your issue might be a dirty flash by Tuya-Convert. Or it could be any of the following: a weak power supply/voltage regulator of the device, lack of memory, using settings that create a heavy load on the chip.

As far as bricking, did you try to put it into flash mode using the on chip gpio0 instead of the button? did you provide an external power source for the chip and not your serial-to-usb adapters?

In the future you can use Backlog to issue more than one command and eschew multiple restarts.
`Switchmode 7` creates a toggle action on pressing the button and a toggle action on depressing the button, i'm not sure thats what you wanted.
For instant button response there's `setoption 13`

badabing2005, 2019-05-08:
Thanks for the quick reply @blakadder 

> Since you did not manage to erase the chip's flash completely and make a clean flash of Tasmota the cause of your issue might be a dirty flash by Tuya-Convert.

I thought about that, but the second plug did not go through Tuya-Convert and still had the same issues.
Besides clean flashing Tuya-Convert would defeat it's purpose, If I'm not mistaken I'd have to crack open the plug and resort to Serial flashing afterwards no?

> Or it could be any of the following: a weak power supply/voltage regulator of the device, lack of memory, using settings that create a heavy load on the chip.

It's main's plug, I did not power the device, not much choice there, and my power for all my household devices seem ok. Perhaps the voltage converted / regulator inside the device has issues? Possibility, but that would mean lots of others would have the similar issues even with Tuya firmware.

As for settings, mine was quite simple.

> As far as bricking, did you try to put it into flash mode using the on chip gpio0 instead of the button? did you provide an external power source for the chip and not your serial-to-usb adapters?

Yes GPIO0 grounded
No I did not try external power source, I'll try that tonight, thanks for the suggestion, though I never had issues flashing directly from the serial-to-usb adapter all my other devices.

> In the future you can use Backlog to issue more than one command and eschew multiple restarts.
Right, and I do use Backlog commands often, just sometimes it's convenient to make quick changes through the UI. 

> Switchmode 7 creates a toggle action on pressing the button and a toggle action on depressing the button, i'm not sure thats what you wanted.
I didn't know that, thanks, would that also be the case if the toggle was done through the UI or API?
I guess that should double the duration?
1 time period for the press and another for the depress?

> For instant button response there's setoption 13
Wouldn't that be an issue if I want to use the multi-press button functions to put the device into wifi manager mode or reset the device?
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

With such plugs, I'd like to keep that option open.

Perhaps I should have used rules instead, could that have been the root issue? 
What basically I was trying to achieve is a timer like functionality.
Upon button press, the plug to be ON for x-seconds and then turn OFF by itself.

Again, thanks for taking the time to respond.
Have a good day

BB

meingraham, 2019-05-08:
@badabing2005 

When you flashed the device using the serial interface, did you erase_flash first? Since you are able to access the pins, I would recommend repeating the flash by booting into programming mode, erasing the flash, powering down your device, booting back into programming mode, and flashing the firmware.

... plus all the recommendations @blakadder made.

Regards.

Mike

Jason2866, 2019-05-08:
My 0,02 on this. Yes Tuya convert makes it easy to get Tasmota on Tuya devices.
BUT there is a big drawback, garbage data is left in flash. This can result in malfunctions like you describe.
Tasmota 6.5.0 runs on many devices without any issues, so it is quiet safe there is no device killer bug in there.

ascillato2, 2019-05-08:
Closing issue as there is no bricking bug.

Please, if you need further assistant on erasing and flashing, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-08 by ascillato2

badabing2005, 2019-05-09:
Thanks guys for the quick responses.

@meingraham 
Although I have access to the pins the serial interface is not working, I can't connect to the devices, both have the same issue.
And just to be sure that somehow my USB Serial or my PC would be the reason, I flashed another device totally fine with the same FTDI.
Also I don't recall when I flashed using the serial interface if I erased first or not, that was a while back, at a time when tuya-convert was not available

I appreciate your feedback, though I definitely have two bricked plugs.
No LED, no serial, dead.

For tuya-converted devices that have garbage data in the flash, is there a way of erasing while writing good tasmota data without losing connection and having to resort to serial flash?
I suspect not, but worth the ask

Thanks
BB

meingraham, 2019-05-09:
No ESP device is truly bricked if you can get to the serial pins. Yes, the device can be dead, but you really can't brick the ESP chip. If there are any signs of life, there's still hope.

Your only option is to try the serial interface. Even if you once were able to program the device with your FTDI adapter, there's a possibility the devices and even the adapter have changed over time. Try a different type of serial adapter (e.g., one based on the CH340) or powering the device through a separate power supply other than the serial adapter. Different devices have different power characteristics. Some can be powered by an FTDI adapter fine while others cannot. Proper power is a critical factor in achieving a successful flash via the serial interface.

Mike

badabing2005, 2019-05-09:
@meingraham 
It's good to know that no ESP device is truly bricked.
I forgot to mention that yesterday I did try powering up the device with external source and only used FTDI adapter for serial connection, no cigars.
I just ordered CH340 based serial adapter, we'll see if that can manage to salvage it.

Thanks for the feedback
Cheers
BB

badabing2005, 2019-05-09:
> My 0,02 on this. Yes Tuya convert makes it easy to get Tasmota on Tuya devices.
> BUT there is a big drawback, garbage data is left in flash. This can result in malfunctions like you describe.
> Tasmota [...]

I'm not saying that there is definitely a bug in Tasmota, but there is the possibility of it, under certain condition on some devices perhaps.
I already acknowledged that most of my devices are running 6.5.0 without issues, so I recognize that it's solid, but we can't lay it all on Tuya-convert, as one of the two plugs had never seen Tuya-convert and yet suffered the same fate. How do we explain that?

Thanks
BB

Jason2866, 2019-05-10:
Not erasing flash before flashing Tasmota and bad/cheap device(s). 
It is cheapest stuff (in every aspect) wrapped together...
Do expect the unexpected ;-)