arendst/Tasmota #5765: New PZEM-004T version (protocol changed)

martindb, 2019-05-08:
There is a new firmware version of the PZEM-004T sensor.
It actualy uses MODBUS RTU protocol. Also now reports power factor and frequency.
It should be great tasmota support it.

[modbus_PZEM-004T.txt](https://github.com/arendst/Sonoff-Tasmota/files/3155097/modbus_PZEM-004T.txt)

Giardo84, 2019-05-08:
I also have a new pzem-004T (v3.0).
I tried to make it work with Tasmota but it doesn't work.
I tried also with Arduino without any result.
The documentation is slightly different from the one posted by you.

curzon01, 2019-05-08:
> The documentation is slightly different from the one posted by you.
> In the documentation, the functional diagram shows a TTL interface instead of RS485
The dox desc are the same protocol, TTL and RS485 are only different physical layer.

Giardo84, 2019-05-08:
> > The documentation is slightly different from the one posted by you.
> > In the documentation, the functional diagram shows a TTL interface instead of RS485
> 
> The dox desc are the same protocol, TTL and RS485 are only different physical layer.

OK thanks.
So what modification should be made to make this library work with the pzem-004T (v3.0)?

curzon01, 2019-05-09:
> So what modification should be made to make this library work with the pzem-004T (v3.0)?

The name of the sensor may be confuse you - PZEM-004T protocol is implemented in module as `xnrg_05_pzem_ac.ino` and enabled as default in sonoff.bin/sonoff-sensors.bin (`#define USE_PZEM_AC`).

Have a look into [xnrg_05_pzem_ac.ino](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xnrg_05_pzem_ac.ino) comment at the beginning you read the following:
`Hardware Serial will be selected if GPIO1 = [98 PZEM016 Rx] and GPIO3 = [62 PZEM0XX Tx]` - pls try this or use `[98 PZEM016 Rx]`/`[62 PZEM0XX Tx]` on other GPIOs

But anyway  if you have problems to configure your device it's better to use some of the [community channels](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here).  Github issues are used for Tasmota software feature requests and bug reporting. Troubleshooting is more effective using an interactive forum.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-09 by ascillato2

SeByDocKy, 2019-05-09:
Hi,

I don't know if it's related or not but I am using also a PZEM-004T (I don't which version but probably the newest)... and I noticed that the informations are freezed after a couple of minutes. I have the last Tasmota build, timeperiod 10.

When it's freezing, after sometime 5min, measurements are back.... I noticed it's freezeing more for low power values.... don't know it's related or not.

Giardo84, 2019-05-09:
ok, it works but there is a little note to make.
In the sonoff configuration it is necessary to set GPIO1 = [62 PZEM0XX Tx] and GPIO3 = [98 PZEM016 Rx].
Physically connecting RX of an ESP01S with the PZEM-004T and TX of the ESP01S with the PZEM-004T RX.

SeByDocKy, 2019-05-09:
> ok, it works but there is a little note to make.
> In the sonoff configuration it is necessary to set GPIO1 = [62 PZEM0XX Tx] and GPIO3 = [98 PZEM016 Rx].
> Physically connecting RX of an ESP01S with the PZEM-004T and TX of the ESP01S with the PZEM-004T RX.

Yes .... THanks .... 

In fact I got a V2 and and V3..... Unfortunatly I killed my V2 this afternoom.... tried to fix why informations are freezed after a couple of Minutes. I did some test with the V3.... Seems to suffer from the same symptoms. I will do more test tmr. Thanks anyway

SeByDocKy, 2019-05-09:
Just tested the V3 and new hardware serial setting mentioned above and for me both current (and by consequence power) reading are wrong

Voltage | 238 V
-- | --
0.036 A
0 W
9 VA
9 VAr
0.05
50 Hz
2.565 kWh
11.427 kWh
18.933 kWh

I collected that on with the curent version

curzon01, 2019-05-09:
[Calibrated](https://github.com/arendst/Sonoff-Tasmota/wiki/power-monitoring-calibration)?

meingraham, 2019-05-09:
Can't calibrate a PZEM using Tasmota - https://github.com/arendst/Sonoff-Tasmota/issues/3208#issuecomment-405048466

azatikdinarik, 2024-11-30:
> Привет,
> 
> Не знаю, связано ли это или нет, но я также использую PZEM-004T (не знаю, какая версия, но, вероятно, самая новая)... и я заметил, что информация замораживается через пару минут. У меня последняя сборка Tasmota, timeperiod 10.
> 
> Когда он замерзает, примерно через 5 минут измерения возвращаются... Я заметил, что он замерзает больше при низких значениях мощности... не знаю, связано это или нет.
> 
> Если я сделаю программную перезагрузку моего NodeMCU V3, измерения вернутся

У меня такая ситуация показание то появляется то исчезает, причем время разное, не одинаково, и только если трехфазный интерфейс показывает на L2, а сеть у меня однофазный, прошивка  tasmota.bin, шаблон {"NAME":"HW-655 PZEM","GPIO":[0,1472,0,1536,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18} пробовал другие шаблоны безрезультатно, я не специалист только учусь,  не знаю как быть!?