arendst/Tasmota #5771: Configuration for TYWE3S module on Wlansmart RM Mini IR Hub

jjmuriel, 2019-05-09:
Hi.

Please, someone can tell me where should I solder the pins  for this device to connect via serial-usb?  This IR Hub has the same chip as the IR YTF. Or is it possible to load the firmware via OTA? Thanks.

https://www.amazon.com/gp/product/B07LG7WK2T/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

[image]
[image]
[image]

meingraham, 2019-05-09:
Have you tried [Tuya-Convert](https://github.com/ct-Open-Source/tuya-convert)?

Please, address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4) and close this issue on GitHub. The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota software feature requests and bug reporting. Troubleshooting is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

jjmuriel, 2019-05-09:
Ok @meingraham , thank you for your quick response.

closed 2019-05-09 by jjmuriel

jjmuriel, 2019-05-09:
Only for reference: 
[image]